esphome/issues #2498: is firefox supported

poohzaza166, 2021-10-01:
### The problem

when validate on firefox there is no text on screen at all
fix by moving to chrome

### Which version of ESPHome has the issue?

2021.9.2

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

esp01 wifi

### Anything in the logs that might be useful for us?

```txt
i had ublock orgins and privacy baggers
```

OttoWinter, 2021-10-03:
It _should_ work, I even use it myself.

Please:
- post your firefox version
- try without any addons
- try pressing ctrl+shift+r to reload the browser cache
- if the issue still persists, please post a screenshot of what it looks like

bkaufx, 2021-10-05:
Firefox works fine for me.  The only thing I've noticed is that it won't show progress in the logs when you upload a bin file whereas chrome does.

closed 2022-02-18 by github-actions[bot]