esphome/issues #2499: Component Light light.on_turn_off is only called after boot

dylan09, 2021-10-01:
### The problem

The trigger light.on_turn_off is only called directly after boot. When light is turned off from Home Assistant the trigger isn't executed.
I set default values for brightness and light temperature for my rgbww lights. This worked for version 1.20.4. But it is broken with version 2021.9.2

As you can see in log file, the trigger is called in an endless loop after boot (I call light.turn_off in on_boot). But after switching on the light from Home Assistant it is never called again.

As you can see in the log, `on_turn_off` is called in an endless loop until the first state switch to on. But it is never called for switching off light from HA.

### Which version of ESPHome has the issue?

2021.9.2

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.9.7

### What platform are you using?

ESP8266

### Board

esp_wroom_02

### Component causing the issue

light

### Example YAML snippet

```yaml
# esphome configuration WizLight E60 bulb living room

substitutions:
  node_name: xx-li-wiz2
  id_prefix: xx_li_wiz2
  name_prefix: xx_li_wiz2
  brightness_offset: "-20"

packages:
  wifi: !include common/wifi.yaml
  web_server: !include common/web_server.yaml
  mqtt:  !include common/mqtt.yaml
  time:  !include common/time.yaml
  sensor: !include common/def_sensor.yaml
  text_sensor: !include common/def_text_sensor.yaml

esphome:
  name: ${node_name}
  platform: ESP8266
  board: esp_wroom_02
  on_boot:
    priority: -10
    # ...
    then:
      - light.turn_off: ${id_prefix}

captive_portal:

logger:

mqtt:
  discovery: true

ota:
[... 241 lines cut ...]

[20:40:21][I][main:129]: Set brightness to 50.000000
[20:40:21][I][main:132]: Set temperature to 2800.000000
[20:40:21][D][light:346]: 'xx_li_wizzyS60' - Setting cold/warm white channels using white/color temperature values.
[20:40:21][D][light:035]: 'xx_li_wizzyS60' Setting:
[20:40:21][D][light:050]:   Brightness: 30%
[20:40:21][D][light:065]:   Color temperature: 357.1 mireds
[20:40:21][D][light:070]:   Cold white: 77%, warm white: 100%
[20:40:21][D][light:084]:   Transition length: 0.5s
[20:40:21][I][light:390]: 'xx_li_wizzyS60' - Keeping current color mode Cold/warm white for call without color mode.
[20:40:21][D][light:035]: 'xx_li_wizzyS60' Setting:
[20:40:21][D][light:046]:   State: ON
[20:40:21][D][light:084]:   Transition length: 0.5s
[20:40:23][D][light:035]: 'xx_li_wizzyS60' Setting:
[20:40:23][D][light:046]:   State: OFF
[20:40:23][D][light:084]:   Transition length: 0.5s
[20:40:30][D][sensor:121]: 'xx_li_wizzyS60 Uptime': Sending state 39.95700 s with 0 decimals of accuracy
[20:40:30][I][light:390]: 'xx_li_wizzyS60' - Keeping current color mode Cold/warm white for call without color mode.
[20:40:30][D][light:035]: 'xx_li_wizzyS60' Setting:
[20:40:30][D][light:046]:   State: ON
[20:40:30][D][light:084]:   Transition length: 0.5s
[20:40:32][D][light:035]: 'xx_li_wizzyS60' Setting:
[20:40:32][D][light:046]:   State: OFF
[20:40:32][D][light:084]:   Transition length: 0.5s
```

OttoWinter, 2021-10-03:
> `state: off`

So you're turning a light _off_ from the `on_turn_off` trigger.

Maybe it worked before, but that doesn't seem like the correct behavior anyway.

If a light is _already_ off it I don't think it _should_ call `on_turn_off` again.

dylan09, 2021-10-03:
I think that's a copy&paste error. I think the `state: off` statement is obsolete. But I will check if deleting the statement will fix the error.

The main purpose of the `on_turn_off` trigger is to set default values for `brightness` and `temperature`. And as could be seen in the log, when the light is turned off by Home Assistant the trigger isn't executed.
With version 1.20.4 it is working as expected.

OttoWinter, 2021-10-04:
Try using `light.turn_on` or add `state: on`.

What your code is doing right now is changing the color_temp etc of a light that is already off.

Again, if you control a light in `on_turn_off` I'd _expect_ it not to run `on_turn_off` again (because the light is _already_ off)

dylan09, 2021-10-04:
That is my intention. If the light is switched off, I set the brightness/temperature to default value. So if the light is turned on and no further parameters (brightness, color, ...) is send from HA it starts with this settings.

If I set the default values in the `on_turn_on` trigger, parameters (color, brightness, temperature, ...) send from HA are ignored and the light always turn on with my default values.

So it makes a big difference, if I set the default values using the `on_turn_off` or `on_turn_on` 

If I use the `on_turn_off` trigger, I am able to send different parameter from HA when light it turned on. If no parameters are given it starts with default.

If I use the `on_turn_on` trigger, the light always is using the default values, because the `on_turn_on` trigger is executed, after the light is turned on and the parameters send from HA are applied. 

Maybe there is another way to get the light working as I described above:
- Turn On with parameters from HA: Use these parameters
- Turn On without parameters from HA: Use default values, regardless which parameters was active, when the light was turned off.

Any hints are very welcome.

As a test, I removed all the code from the `on_turn_off` trigger. Only one logger output for debugging. Even without `light.control` the `on_turn_off` trigger is not executed when light is turned off.

dylan09, 2021-10-04:
I think my issue is related to issue #2458.
And if I remember correctly there was also an issue where transition length was a problem.

I have found, that changing the `default_transition_length` to 0ms changed the behaviour how state changes are detected.

I have implemented following yaml as test to see if light conditions are working:
```
interval:
  - interval: 2 sec
    then:
      - if:
          condition:
            light.is_on: ${id_prefix}
          then:
            - logger.log: 
                format: "Condition check: Light is on"
                tag: "intervall 2"
      - if:
          condition:
            light.is_off: ${id_prefix}
          then:
            - logger.log: 
                format: "Condition check: Light is off"
                tag: "intervall 2"
```

Log output, when`default_transition_length` is set to 500ms:
```
[22:58:18][D][intervall 2:272]: Condition check: Light is off
[22:58:19][I][light:390]: 'xx_li_wiz1' - Keeping current color mode Cold/warm white for call without color mode.
[22:58:19][D][light:035]: 'xx_li_wiz1' Setting:
[22:58:19][D][light:046]:   State: ON
[22:58:19][D][light:084]:   Transition length: 0.5s
[22:58:20][D][intervall 2:266]: Condition check: Light is on
[22:58:22][D][intervall 2:266]: Condition check: Light is on
[22:58:24][D][intervall 2:266]: Condition check: Light is on
[22:58:26][D][light:035]: 'xx_li_wiz1' Setting:
[22:58:26][D][light:046]:   State: OFF
[22:58:26][D][light:084]:   Transition length: 0.5s
[22:58:26][D][intervall 2:266]: Condition check: Light is on
[22:58:28][D][intervall 2:266]: Condition check: Light is on
```

Log output, when`default_transition_length` is set to 0ms:
```
[23:00:09][D][intervall 2:272]: Condition check: Light is off
[23:00:09][I][light:390]: 'xx_li_wiz1' - Keeping current color mode Cold/warm white for call without color mode.
[23:00:09][D][light:035]: 'xx_li_wiz1' Setting:
[23:00:09][D][light:046]:   State: ON
[23:00:11][D][intervall 2:266]: Condition check: Light is on
[23:00:13][D][intervall 2:266]: Condition check: Light is on
[23:00:15][D][intervall 2:266]: Condition check: Light is on
[23:00:17][D][intervall 2:266]: Condition check: Light is on
[23:00:17][D][light:035]: 'xx_li_wiz1' Setting:
[23:00:17][D][light:046]:   State: OFF
[23:00:19][D][intervall 2:272]: Condition check: Light is off
[23:00:21][D][intervall 2:272]: Condition check: Light is off
```

With transition lenght != 0ms also the condition check `light.is_off` isn't working. 

First log output shows, that after State: OFF, the condition for on is true.
In the second log output using 0ms, the check is working.

AZDane, 2021-10-13:
Could this issue potentially stem from this fix?
https://github.com/esphome/esphome/pull/2320

dylan09, 2021-10-13:
Thanks. Will test and check the fix when next ESPHome version is officially published.