esphome/issues #2500: BTLE Scanner takes 10 minutes to see device gone

tjharman, 2021-10-01:
### The problem

Firstly I'm sorry for creating an issue on this - I've searched both the HA docs and the ESPHome docs and I can't find a solution to my answer, so I ask here.  Please close immedietly if this was the wrong thing to do, I appreciate you have limited time and annoying "end users" asking questions isn't really what Github Issues are for.

I have created two BLE Trackers as such on my Sparkfun ESP32:

```
esp32_ble_tracker:
binary_sensor:
  - platform: ble_presence
    ibeacon_uuid: 'HEX'
    name: "Car"
    ibeacon_major: 'P'
    ibeacon_minor: 'Q'
  - platform: ble_presence
    ibeacon_uuid: 'HEX'
    name: "Tim's Keys"
    ibeacon_major: 'Y'
    ibeacon_minor: 'X'
```

It works great! (I've been waiting for ages to get iBeacon tracking working in ESPHome to replace my Happy Bubbles and I'm overjoyed it finally works)

As soon as I power up my iBeacon with a battery, the ESP32 sees it and the binary_sensor in Home Assistant trips to on.  Perfect.
However, when I pull the battery out, it takes a full 10 minutes before HA changes the binary_sensor back to Off.

What am I missing here?  What variable/thing must I adjust, if any, to make the timeout shorter?  I don't expect an immediate Off but I'd think no more than 30 seconds to maybe a minute?  10 minutes seems a very long time.

Thank you.

### Which version of ESPHome has the issue?

2021.9.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.7

### What platform are you using?

ESP32

### Board

Sparkfun ESP32

### Component causing the issue

ble_presence

tjharman, 2021-10-03:
I have found the answer to this - adjust "duration" to something much lower like 30seconds.
Thanks and sorry for time wasting.

closed 2021-10-03 by tjharman

sharpopotamus, 2021-12-05:
This helped me solve the same issue, I think the docs need to be updated. Right now, 'duration' is described as "The duration of each complete scan. This has no real impact on the device but can be used to debug the BLE stack. Defaults to 5min." Which is super misleading.