esphome/issues #2502: Enabling encryption causes warning message after firmware is OTA'd

bachya, 2021-10-01:
### The problem

When attempting to utilize encryption on a node flashed with 2021.9.2, the logs throw a warning and the node doesn't seem to provide any further logs.

### Which version of ESPHome has the issue?

2021.9.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.10.0b2

### What platform are you using?

ESP8266

### Board

esp8285

### Component causing the issue

api

### Example YAML snippet

```yaml
---
api:
  encryption:
    key: !secret api_encryption_key
  password: !secret api_password

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO4
      inverted: true
    name: Power Button
    on_press:
      - switch.toggle: relay

  - platform: status
    name: Smart Plug 1

esphome:
  name: smart-plug-1
  platform: ESP8266
  board: esp8285

logger:

ota:

status_led:
  pin:
    number: GPIO1
    inverted: true

switch:
  - platform: gpio
    id: led
    pin: GPIO13
    inverted: true

  - platform: gpio
    name: "${device_friendly_name}"
    pin: GPIO14
    id: relay
    on_turn_on:
      - switch.turn_on: led
    on_turn_off:
      - switch.turn_off: led

text_sensor:
  - platform: template
    name: "${device_friendly_name}: ESPHome Version (Short)"
    lambda: |-
      auto state = id(esphome_version).state;
      return state.substr(0, state.find(' '));

  - platform: version
    name: "${device_friendly_name}: ESPHome Version"
    id: esphome_version

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  output_power: 20dB

  manual_ip:
    static_ip: 192.168.1.100
    gateway: 192.168.1.1
    subnet: 255.255.255.0
```

### Anything in the logs that might be useful for us?

```txt
When I upload this firmware using the 2021.9.2 Docker image:

docker run --rm -v "$(PWD)/esphome":/config esphome/esphome:2021.9.2 run smart_plug_1.yaml

...it succeeds, but shortly thereafter, I get a warning in the logs (and no further logs are generated)::

```
INFO Successfully compiled program.
INFO Connecting to 192.168.1.100
INFO Uploading smart-plug-1/.pioenvs/smart-plug-1/firmware.bin (395392 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.1.100 using esphome API
WARNING Can't connect to ESPHome API for 172.16.20.141: Error while reading data: 0 bytes read on a total of 3 expected bytes
```

If I comment out the encryption lines and re-run the firmware, everything works as expected:

```
INFO Successfully compiled program.
INFO Connecting to 192.168.1.100
INFO Uploading smart-plug-1/.pioenvs/smart-plug-1/firmware.bin (374432 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.1.100 using esphome API
INFO Successfully connected to 192.168.1.100
[21:01:17][I][app:102]: ESPHome version 2021.9.2 compiled on Oct  1 2021, 21:00:25
[21:01:18][C][status_led:019]: Status LED:
[21:01:18][C][status_led:020]:   Pin: GPIO1 (Mode: OUTPUT, INVERTED)
[21:01:18][C][wifi:501]: WiFi:
[21:01:18][C][wifi:361]:   SSID: 'Phil?_IoT'
[21:01:18][C][wifi:362]:   IP Address: 192.168.1.100
[21:01:18][C][wifi:364]:   BSSID: 02:EC:DA:F1:1E:C1
[21:01:18][C][wifi:365]:   Hostname: 'smart-plug-1'
[21:01:18][C][wifi:369]:   Signal strength: -74 dB ▂▄▆█
[21:01:18][C][wifi:373]:   Channel: 1
[21:01:18][C][wifi:374]:   Subnet: 255.255.255.0
[21:01:18][C][wifi:375]:   Gateway: 192.168.1.1
[21:01:18][C][wifi:376]:   DNS1: (IP unset)
[21:01:18][C][wifi:377]:   DNS2: (IP unset)
[21:01:18][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Power Button'
[21:01:18][C][gpio.binary_sensor:016]:   Pin: GPIO4 (Mode: INPUT, INVERTED)
[21:01:18][C][switch.gpio:048]: GPIO Switch 'led'
[21:01:18][C][switch.gpio:048]:   Inverted: YES
[21:01:18][C][switch.gpio:049]:   Pin: GPIO13 (Mode: OUTPUT)
[21:01:18][C][switch.gpio:071]:   Restore Mode: Restore (Defaults to OFF)
[21:01:18][C][switch.gpio:048]: GPIO Switch 'Smart Plug 1'
[21:01:18][C][switch.gpio:049]:   Pin: GPIO14 (Mode: OUTPUT)
[21:01:18][C][switch.gpio:071]:   Restore Mode: Restore (Defaults to OFF)
[21:01:18][C][template.text_sensor:021]: Template Sensor 'Smart Plug 1: ESPHome Version (Short)'
[21:01:18][C][logger:193]: Logger:
[21:01:18][C][logger:194]:   Level: DEBUG
[21:01:18][C][logger:195]:   Log Baud Rate: 115200
[21:01:18][C][logger:196]:   Hardware UART: UART0
[21:01:18][C][status:034]: Status Binary Sensor 'Smart Plug 1 Status'
[21:01:18][C][status:034]:   Device Class: 'connectivity'
[21:01:18][C][version.text_sensor:021]: Version Text Sensor 'Smart Plug 1: ESPHome Version'
[21:01:18][C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[21:01:18][C][ota:029]: Over-The-Air Updates:
[21:01:18][C][ota:030]:   Address: 192.168.1.100:8266
[21:01:18][C][api:135]: API Server:
[21:01:18][C][api:136]:   Address: 192.168.1.100:6053
```
```

### Additional information

(FYI, I used the randomly-generated string from [the documentation](https://esphome.io/components/api.html?highlight=encryption#configuration-variables) for my encryption key.)

bachya, 2021-10-03:
This is also (perhaps?) being tracked in the Home Assistant repo: https://github.com/home-assistant/core/issues/56676

drthanwho, 2021-10-03:
> This is also (perhaps?) being tracked in the Home Assistant repo: [home-assistant/core#56676](https://github.com/home-assistant/core/issues/56676)

The difference on that issue is no one is actually trying to use encryption. This occurred simply after updating so just from whatever was changed on the integration's code.

OttoWinter, 2021-10-03:
> When attempting to utilize encryption on an esp8285 board flashed with 2021.9.2, the logs throw a warning and the node doesn't seem to provide any further logs.

Could you provide direct logs from the device while connecting?

It might be that the device is not capable enough of encryption (or the device is crashing somewhere). USB/Serial logs would help would hopefully contain what is going wrong

bachya, 2021-10-03:
@OttoWinter Here is the serial log of a freshly-flashed `d1_mini` (unfortunately, I can't get the serial logs of the `esp8285` boards because they're encased into their Tuya housings):

```
[10:03:50]rl[I][logger:174]: Log initialized
[10:03:50][I][app:029]: Running through setup()...
[10:03:50][D][binary_sensor:034]: 'Master Bathroom Climate Sensor Status': Sending initial state OFF
[10:03:50][C][bme280.sensor:081]: Setting up BME280...
[10:03:50][D][text_sensor:015]: 'Master Bathroom Climate Sensor: ESPHome Version': Sending state '2021.9.2 Oct  3 2021, 16:00:14'
[10:03:50][C][wifi:037]: Setting up WiFi...
[10:03:50][D][wifi:382]: Starting scan...
[10:03:50][D][text_sensor:015]: 'Master Bathroom Climate Sensor: ESPHome Version (Short)': Sending state '2021.9.2'
[10:03:50][D][bme280.sensor:197]: Got temperature=22.0°C pressure=846.0hPa humidity=40.6%
[10:03:50][D][sensor:121]: 'Master Bathroom Temperature': Sending state 21.98000 °C with 1 decimals of accuracy
[10:03:50][D][sensor:121]: 'Master Bathroom Pressure': Sending state 845.98468 hPa with 1 decimals of accuracy
[10:03:50][D][sensor:121]: 'Master Bathroom Humidity': Sending state 40.63770 % with 1 decimals of accuracy
[10:03:56][D][wifi:397]: Found networks:
[10:03:56][I][wifi:443]: - 'REDACTED' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:444]:     Channel: 6
[10:03:56][D][wifi:445]:     RSSI: -71 dB
[10:03:56][D][wifi:447]: - 'REDACTED' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - '' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - '' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - 'REDACTED' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - '' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - '' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - '' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - '' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - 'REDACTED' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - '' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - '' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - 'REDACTED' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - '' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - 'REDACTED' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - 'REDACTED' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - '' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - 'REDACTED' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - 'REDACTED' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - 'REDACTED' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - 'REDACTED' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - '' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - 'REDACTED' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - 'REDACTED' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - '' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - 'REDACTED' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - 'REDACTED' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - 'REDACTED' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - 'REDACTED' (REDACTED) ▂▄▆█
[10:03:56][D][wifi:447]: - 'REDACTED' (REDACTED) ▂▄▆█
[10:03:56][I][wifi:252]: WiFi Connecting to 'REDACTED'...
[10:03:58][I][wifi:515]: WiFi Connected!
[10:03:58][C][wifi:361]:   SSID: 'REDACTED'
[10:03:58][C][wifi:362]:   IP Address: 192.168.1.100
[10:03:58][C][wifi:364]:   BSSID: REDACTED
[10:03:58][C][wifi:365]:   Hostname: 'master-bathroom-climate-sensor'
[10:03:58][C][wifi:369]:   Signal strength: -72 dB ▂▄▆█
[10:03:58][C][wifi:373]:   Channel: 6
[10:03:58][C][wifi:374]:   Subnet: 255.255.255.0
[10:03:58][C][wifi:375]:   Gateway: 192.168.1.1
[10:03:58][C][wifi:376]:   DNS1: (IP unset)
[10:03:58][C][wifi:377]:   DNS2: (IP unset)
[10:03:58][C][ota:029]: Over-The-Air Updates:
[10:03:58][C][ota:030]:   Address: 192.168.1.100:8266
[10:03:58][C][api:023]: Setting up Home Assistant API server...
[10:03:58][I][app:060]: setup() finished successfully!
[10:03:58][I][app:102]: ESPHome version 2021.9.2 compiled on Oct  3 2021, 16:00:14
[10:03:58][C][wifi:501]: WiFi:
[10:03:58][C][wifi:361]:   SSID: 'Phil?_IoT'
[10:03:58][C][wifi:362]:   IP Address: 192.168.1.100
[10:03:58][C][wifi:364]:   BSSID: 02:EC:DA:32:AA:47
[10:03:58][C][wifi:365]:   Hostname: 'master-bathroom-climate-sensor'
[... 129 lines cut ...]
[10:07:01]
[10:07:01]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[10:07:01]
[10:07:01] ets Jan  8 2013,rst cause:2, boot mode:(3,7)
[10:07:01]
[10:07:01]load 0x4010f000, len 3584, room 16 
[10:07:01]tail 0
[10:07:01]chksum 0xb0
[10:07:01]csum 0xb0
[10:07:01]v2843a5ac
[10:07:01]~ld
```

Lastly, in case it's useful, the ESPHome Dashboard (2021.9.2) shows the node is offline:

[image]

...and attempting to view the logs over WiFi shows this:

```
INFO Reading configuration /config/master_bathroom_climate_sensor.yaml...
INFO Starting log output from 172.16.20.226 using esphome API
WARNING Can't connect to ESPHome API for 172.16.20.226: Error while reading data: [Errno 104] Connection reset by peer
INFO Trying to reconnect to 172.16.20.226 in the background
```

OttoWinter, 2021-10-03:
> I don't see the same erroneous log here, interestingly enough. When I try to add this node to a fresh HASS (2021.10.0b3), I am asked to input my encryption key

Side note: it looks like translation strings work now from your screenshot, can you confirm?

> ...and this shows up in the serial logs:

ok that seems to be the culprit; looks like for some reason the stack trace analyzer didn't attempt to decode that stack; I'll see if I can reproduce by using that board type

bachya, 2021-10-03:
> Side note: it looks like translation strings work now from your screenshot, can you confirm?

Yes, mostly, but it looks like something is missing at the end of the string in that screenshot (`for .`).

OttoWinter, 2021-10-05:
Ok I can replicate it with `esp01_1m` board type here, decoded stack trace:

```
[11:25:51]Exception (9):
WARNING Exception type: LoadStoreAlignmentCause: Load or store to an unaligned address
[11:25:51]epc1=0x40204e38 epc2=0x00000000 epc3=0x00000000 excvaddr=0x3ffe904b depc=0x00000000
WARNING Decoded 0x40204e38: mul at x25519.c
WARNING Decoded 0x40204eac: mul at x25519.c
WARNING Decoded 0x40203f45: poly1305_blocks at poly1305-donna.c
WARNING Decoded 0x40204ef0: mul1 at x25519.c
WARNING Decoded 0x4020503e: x25519
WARNING Decoded 0x402011e4: noise_cipherstate_decrypt_with_ad
WARNING Decoded 0x40204ca9: noise_curve25519_generate_keypair at dh-curve25519.c
WARNING Decoded 0x402049c1: noise_dhstate_generate_dependent_keypair
WARNING Decoded 0x402018e9: noise_handshakestate_write_message
WARNING Decoded 0x40201d77: noise_handshakestate_read_message
WARNING Decoded 0x40205a86: esphome::api::APIConnection::loop()
WARNING Decoded 0x40209142: esphome::api::APIServer::loop()
WARNING Decoded 0x40218d48: esphome::Component::call_loop()
WARNING Decoded 0x40218dc1: esphome::Component::call()  
WARNING Decoded 0x4020e371: esphome::Application::loop()
WARNING Decoded 0x4021034c: loop
WARNING Decoded 0x402169e8: loop_wrapper() at core_esp8266_main.cpp
WARNING Decoded 0x40100595: cont_wrapper
```

The method that produces this issue is here:

https://github.com/esphome/noise-c/blob/bc74e61f6cd37f0145be92366ea6c7864ed2589c/src/crypto/x25519/x25519.c#L136

but the bug _doesn't_ appear in pio `debug` build types (which is probably why I never noticed it haha), that's going to make JTAG debugging harder - fun times :laughing:

OttoWinter, 2021-10-05:
For reference: here's the disassembled binary, the exception happens on address `0x40204e38`

```
40204df4 <mul>:
40204df4:	80c112               	addi	a1, a1, -128
40204df7:	1d61d2               	s32i	a13, a1, 116
40204dfa:	1c61e2               	s32i	a14, a1, 112
40204dfd:	1b61f2               	s32i	a15, a1, 108
40204e00:	04dd                	mov.n	a13, a4
40204e02:	02ed                	mov.n	a14, a2
40204e04:	03fd                	mov.n	a15, a3
40204e06:	012d                	mov.n	a2, a1
40204e08:	030c                	movi.n	a3, 0
40204e0a:	044c                	movi.n	a4, 64
40204e0c:	1e61c2               	s32i	a12, a1, 120
40204e0f:	1f6102               	s32i	a0, a1, 124
40204e12:	05cd                	mov.n	a12, a5
40204e14:	f40701               	l32r	a0, 40201e30 <noise_handshakestate_split+0x50>
40204e17:	0000c0               	callx0	a0
40204e1a:	016d                	mov.n	a6, a1
40204e1c:	a0acd0               	addx4	a10, a12, a13
40204e1f:	001906               	j	40204e87 <mul+0x93>
40204e22:	0c0c                	movi.n	a12, 0
40204e24:	0db8                	l32i.n	a11, a13, 0
40204e26:	207cc0               	or	a7, a12, a12
40204e29:	86ca                	add.n	a8, a6, a12
40204e2b:	0828                	l32i.n	a2, a8, 0
40204e2d:	190c                	movi.n	a9, 1
40204e2f:	727a                	add.n	a7, a2, a7
40204e31:	013727               	bltu	a7, a2, 40204e36 <mul+0x42>
40204e34:	090c                	movi.n	a9, 0
40204e36:	2fca                	add.n	a2, a15, a12
*** 40204e38:	0228                	l32i.n	a2, a2, 0
40204e3a:	030c                	movi.n	a3, 0
40204e3c:	0b4d                	mov.n	a4, a11
40204e3e:	205330               	or	a5, a3, a3
40204e41:	126162               	s32i	a6, a1, 72
40204e44:	156172               	s32i	a7, a1, 84
40204e47:	116182               	s32i	a8, a1, 68
40204e4a:	146192               	s32i	a9, a1, 80
40204e4d:	1361a2               	s32i	a10, a1, 76
40204e50:	1061b2               	s32i	a11, a1, 64
40204e53:	fb4701               	l32r	a0, 40203b70 <chacha_encrypt_bytes+0x740>
40204e56:	0000c0               	callx0	a0
40204e59:	152172               	l32i	a7, a1, 84
40204e5c:	140c                	movi.n	a4, 1
40204e5e:	272a                	add.n	a2, a7, a2
40204e60:	122162               	l32i	a6, a1, 72
40204e63:	112182               	l32i	a8, a1, 68
40204e66:	142192               	l32i	a9, a1, 80
40204e69:	1321a2               	l32i	a10, a1, 76
40204e6c:	1021b2               	l32i	a11, a1, 64
40204e6f:	013277               	bltu	a2, a7, 40204e74 <mul+0x80>
40204e72:	040c                	movi.n	a4, 0
40204e74:	993a                	add.n	a9, a9, a3
40204e76:	949a                	add.n	a9, a4, a9
40204e78:	0829                	s32i.n	a2, a8, 0
40204e7a:	cc4b                	addi.n	a12, a12, 4
40204e7c:	097d                	mov.n	a7, a9
40204e7e:	a7cc66               	bnei	a12, 32, 40204e29 <mul+0x35>
40204e81:	8699                	s32i.n	a9, a6, 32
40204e83:	dd4b                	addi.n	a13, a13, 4
40204e85:	664b                	addi.n	a6, a6, 4
40204e87:	979da7               	bne	a13, a10, 40204e22 <mul+0x2e>
40204e8a:	0c0c                	movi.n	a12, 0
40204e8c:	0cdd                	mov.n	a13, a12
40204e8e:	21ca                	add.n	a2, a1, a12
40204e90:	0238                	l32i.n	a3, a2, 0
40204e92:	1f0c                	movi.n	a15, 1
40204e94:	d3da                	add.n	a13, a3, a13
[... 87 lines cut ...]
40204edf:	07cd                	mov.n	a12, a7
40204ee1:	51ca                	add.n	a5, a1, a12
40204ee3:	0538                	l32i.n	a3, a5, 0
40204ee5:	8558                	l32i.n	a5, a5, 32
40204ee7:	40c122               	addi	a2, a1, 64
40204eea:	642c                	movi.n	a4, 38
40204eec:	ffe3c5               	call0	40204d2c <umaal>
40204eef:	3dca                	add.n	a3, a13, a12
40204ef1:	0329                	s32i.n	a2, a3, 0
40204ef3:	cc4b                	addi.n	a12, a12, 4
40204ef5:	e8cc66               	bnei	a12, 32, 40204ee1 <mul+0x8d>
40204ef8:	102132               	l32i	a3, a1, 64
40204efb:	0d2d                	mov.n	a2, a13
40204efd:	ffe745               	call0	40204d74 <propagate>
*** 40204f00:	232102               	l32i	a0, a1, 140
40204f03:	90a092               	movi	a9, 144
40204f06:	2221c2               	l32i	a12, a1, 136
40204f09:	2121d2               	l32i	a13, a1, 132
40204f0c:	2021e2               	l32i	a14, a1, 128
40204f0f:	1f21f2               	l32i	a15, a1, 124
40204f12:	119a                	add.n	a1, a1, a9
40204f14:	f00d                	ret.n
```

If someone with more xtensa/assembly experience could look into this I'd be grateful :slightly_smiling_face: I don't even know where to start diagnising this, at least the c code ([here](https://github.com/esphome/noise-c/blob/bc74e61f6cd37f0145be92366ea6c7864ed2589c/src/crypto/x25519/x25519.c#L136)) looks fine to me