esphome/issues #2503: ruuvi_ble and xiaomi_ble should not register without yaml opt-in

spbrogan, 2021-10-01:
### The problem

sorry...this walks the line between bug and feature but given the user experience I went with bug.
When i enable 
``` yaml
esp32_ble_tracker:
```
in my yaml config the [ruuvi_ble](https://github.com/esphome/esphome/tree/dev/esphome/components/ruuvi_ble) and [xiaomi ble](https://github.com/esphome/esphome/tree/dev/esphome/components/xiaomi_ble) listeners get registered. 

To a user this is unexpected and given the potential growth for many different BLE devices, memory foot print and perf concerns it would make sense that these listeners only get added when the yaml author opts in.  

### Which version of ESPHome has the issue?

2012.9.2 

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Component causing the issue

ruuvi_ble and xiaomi_ble 

### Example YAML snippet

```yaml
yaml
esp32_ble_tracker:
```
```

### Additional information

in main.cpp you can see
``` cpp
 // xiaomi_ble:
  //   id: xiaomi_ble_xiaomilistener
  //   esp32_ble_id: esp32_ble_tracker_esp32bletracker
  xiaomi_ble_xiaomilistener = new xiaomi_ble::XiaomiListener();
  esp32_ble_tracker_esp32bletracker->register_listener(xiaomi_ble_xiaomilistener);
  // ruuvi_ble:
  //   id: ruuvi_ble_ruuvilistener
  //   esp32_ble_id: esp32_ble_tracker_esp32bletracker
  ruuvi_ble_ruuvilistener = new ruuvi_ble::RuuviListener();
  esp32_ble_tracker_esp32bletracker->register_listener(ruuvi_ble_ruuvilistener);
  // =========== AUTO GENERATED CODE END ============
  // ========= YOU CAN EDIT AFTER THIS LINE =========
```

spbrogan, 2021-10-01:
A few more details thanks to @ssieb
Issue is with https://github.com/esphome/esphome/blob/d0dfc94a61cd6d3758e4284156d5ce707a7803b2/esphome/components/esp32_ble_tracker/__init__.py#L21

ssieb, 2021-10-01:
My question is why is the ble tracker pulling in all the listeners instead of them getting pulled in by the components that need them?

spbrogan, 2021-10-02:
For my component I used the case where a sensor is added with MAC of 00:00:00:00:00:00 as a way to do the scanner.  This way there is an easy way for yaml to include the integration and it doesn't have to be auto loaded in all code that is using the ble tracker.   You can see here: https://github.com/spbrogan/esphome/blob/spbrogan/add_mopeka_pro_check_v1/esphome/components/mopeka_pro_check/mopeka_pro_check.cpp#L66

interested to hear feedback from any of the BLE integration owners.

OttoWinter, 2021-10-03:
So this is due to historic reasons. First only really `xiaomi_ble` existed for `esp32_ble_tracker`, and to get the easiest config setup I just enabled `xiaomi_ble` listening when the tracker is enabled.

> To a user this is unexpected and given the potential growth for many different BLE devices, memory foot print and perf concerns it would make sense that these listeners only get added when the yaml author opts in.

Compared to what BLE uses itself in terms of memory/code size, I _think_ xiaomi/ruuvi should be almost nothing, no? Especially on ESP32 we usually don't have to worry about resources _that_ much.

spbrogan, 2021-10-04:
yeah.  Makes sense.   Looks like for my build it was 3.5KB of flash space.  How do you feel about changing it to require user to put something in yaml to opt in for each integration? Personally I just don't like integrations that I am not using getting loaded.