esphome/issues #2504: SM2135 component locking up when attempting to control RGB

skandiah94, 2021-10-02:
### The problem

I've got a Mirabella Genio RGBCT LED downlight which I've flashed to ESPHome. I'm able to control the temperature and brightness of the light, however when I swap to colour control the light will change to the first colour I chose and then "freeze". While it is frozen I am unable to adjust brightness, colour, mode, or toggle the light on or off. 

Looking at logs via ESPHome Dashboard or the web UI I can see message being sent and the device responding appropriately, but the actual light stays the same. The only way to fix this is to power cycle the light from the mains.

I have confirmed that the issue is with ESPHome by flashing Tasmota to the device and controlling it using the appropriate template (https://templates.blakadder.com/mirabella_genio_I002741.html). With Tasmota I am able to control the light as expected (choose colour, brightness, swap to warm/cold white etc)

Happy to be a guinea pig for any code changes :) 

### Which version of ESPHome has the issue?

2021.9.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.9.7

### What platform are you using?

ESP8266

### Board

esp8285

### Component causing the issue

sm2135

### Example YAML snippet

```yaml
esphome:
  name: ensuite-downlight1
  platform: ESP8266
  board: esp8285

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: ""

wifi:
  ssid: "IoT Devices"
  password: !secret wifi_password
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Ensuite-Downlight1"
    password: ""

captive_portal:

web_server:

sm2135:
  clock_pin: GPIO12
  data_pin: GPIO14

# Individual outputs
output:
  - platform: sm2135
    id: output_red
    channel: 1
  - platform: sm2135
    id: output_green
    channel: 0
  - platform: sm2135
    id: output_blue
    channel: 2
  - platform: sm2135
    id: output_white
    channel: 3
  - platform: sm2135
    id: output_warmwhite
    channel: 4

light:
  - platform: rgbww
    name: "Ensuite Downlight 1"
    red: output_red 
    green: output_green 
    blue: output_blue 
    cold_white: output_white 
    warm_white: output_warmwhite 
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2700 K
    color_interlock: true
    restore_mode: ALWAYS_ON
```

BoukeHaarsma23, 2021-10-03:
Have you tried only using only the colors and use it as an RGB light? Does it also freeze then? I cannot reproduce this on my lights. So it'll be some trial and error I'm afraid.

skandiah94, 2021-10-04:
I've just flashed the config to RGB with the sm2135 driver. The light turns on and is frozen straight away, no response to change in colour, or toggle on/off via HA. Logs via ESPHome dashboard/Light's Web UI are still responsive and show the correct RGB values being set.

It seems as if like once the sm2135 starts driving the RGB channels it locks up.. I'll have a look to see if I can verify the light uses sm2135 (I've got a broken one near that I can pull apart) but based off the Tasmota template I'm assuming it's correct

Excuse the lazy editing via comments, but here's the config I used to test RGB only
```
sm2135:
  clock_pin: GPIO12
  data_pin: GPIO14

# Individual outputs
output:
  - platform: sm2135
    id: output_red
    channel: 1
  - platform: sm2135
    id: output_green
    channel: 0
  - platform: sm2135
    id: output_blue
    channel: 2
  #- platform: sm2135
  #  id: output_white
  #  channel: 3
  #- platform: sm2135
  #  id: output_warmwhite
  #  channel: 4

light:
  - platform: rgb #ww
    name: "${friendly_name}"
    red: output_red 
    green: output_green 
    blue: output_blue 
    #cold_white: output_white 
    #warm_white: output_warmwhite 
    #cold_white_color_temperature: 6500 K
    #warm_white_color_temperature: 2700 K
    #color_interlock: true
    # Ensure the light turns on by default if the physical switch is actuated.
    restore_mode: ALWAYS_ON
    gamma_correct: false
```

dvisser, 2021-12-01:
Any updates on this one please?

skandiah94, 2021-12-29:
Hey @BoukeHaarsma23 have you had a chance to look into this any further?

BoukeHaarsma23, 2021-12-29:
I have been prototyping a lot of stuff lately. But I'm having issues with
my own lights now using the newer implementation. I'll keep on it.

skandiah94, 2022-02-23:
Sorry to be the squeaky wheel, but any updates on this?

dvisser, 2022-03-07:
@BoukeHaarsma23 i know you are busy like everyone is, but just curious if you have had any luck on your own lights with this one?

BoukeHaarsma23, 2022-03-17:
I have not been able to fix my branch. However, I see a new PR open https://github.com/esphome/esphome/pull/3311 that claims to fix this issue.

skandiah94, 2022-05-31:
I've found a workaround solution, limiting the power of the R, G, and B channels to 0.2 seems to stop the device from locking up during use. Credits to this [post](https://community.home-assistant.io/t/mirabella-genio-smart-home/85816/465?u=sk94)

closed 2024-03-20 by jesserockz