esphome/issues #2506: INA226 component on ESP32 return only Voltage (2021.10.0.dev0)

reaper7, 2021-10-03:
### The problem

INA226 component return only VOLTAGE value, POWER and CURRENT are always 0.
The same yaml compiled on earlier versions 2021.8.2,  2021.9.0, 2021.9.1, 2021.9.2) work correctly and all values are returned.
Something has probably changed in ina226 or i2c component in the time between the release of version 2021.9.2(where it worked well) and the current version of dev.

### Which version of ESPHome has the issue?

dev.zip 2021.10.02 16:02

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.9.7

### What platform are you using?

ESP32

### Board

mhetesp32minikit

### Component causing the issue

INA226

### Example YAML snippet

```yaml
# Substitutions
substitutions:
  # Device Names
  device_name: "bathroom141"
  friendly_name: "Bathroom"
  sub_name: "Controller"
  # Icon
  # https://materialdesignicons.com/
  # https://www.esphome.app/docs/icons
  # main_icon: "dishwasher"
  main_icon: "mdi:paper-roll-outline"
  # Default Relay State
  # Aka: `restore_mode` in documentation
  # Options: `RESTORE_DEFAULT_OFF`, `RESTORE_DEFAULT_ON`, `ALWAYS_ON` & `ALWAYS_OFF`
  default_state: "ALWAYS_OFF"
  #
  # https://esphome.io/components/light/index.html
  transition_time: 1000ms

# CORE CONFIG
# https://esphome.io/components/esphome.html
esphome:
  name: "${device_name}"
  platform: ESP32
  board: mhetesp32minikit

# WIFI
# https://esphome.io/components/wifi.html
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  power_save_mode: LIGHT
  reboot_timeout: !secret reboot_timeout_wifi
  fast_connect: !secret fast_connect_wifi
  manual_ip:
    static_ip: !secret bathroom141
    gateway: !secret gateway
[... 727 lines cut ...]
[12:01:29][D][sensor:121]: 'Bathroom Water Shower Cold Total': Sending state 0.00000 L/day with 2 decimals of accuracy
[12:01:29][D][pulse_counter:160]: 'Bathroom Water Sink Warm': Retrieved counter: 0.00 pulses/min
[12:01:29][D][sensor:121]: 'Bathroom Water Sink Warm': Sending state 0.00000 L/min with 2 decimals of accuracy
[12:01:29][D][sensor:121]: 'Bathroom Water Sink Warm Total': Sending state 0.00000 L/day with 2 decimals of accuracy
[12:01:29][D][sensor:121]: 'Bathroom INA226 Bus Voltage': Sending state 12.05250 V with 1 decimals of accuracy
[12:01:29][D][sensor:121]: 'Bathroom INA226 Shunt Voltage': Sending state 0.00000 V with 2 decimals of accuracy
[12:01:29][D][sensor:121]: 'Bathroom INA226 Current': Sending state 0.00000 A with 2 decimals of accuracy
[12:01:29][D][sensor:121]: 'Bathroom INA226 Power': Sending state 0.00000 W with 1 decimals of accuracy
[12:01:30][D][pulse_counter:160]: 'Bathroom Water Sink Cold': Retrieved counter: 0.00 pulses/min
[12:01:30][D][sensor:121]: 'Bathroom Water Sink Cold': Sending state 0.00000 L/min with 2 decimals of accuracy
[12:01:30][D][sensor:121]: 'Bathroom Water Sink Cold Total': Sending state 0.00000 L/day with 2 decimals of accuracy
[12:01:31][D][pulse_counter:160]: 'Bathroom Water Shower Warm': Retrieved counter: 0.00 pulses/min
[12:01:31][D][sensor:121]: 'Bathroom Water Shower Warm': Sending state 0.00000 L/min with 2 decimals of accuracy
[12:01:31][D][sensor:121]: 'Bathroom Water Shower Warm Total': Sending state 0.00000 L/day with 2 decimals of accuracy
[12:01:34][D][pulse_counter:160]: 'Bathroom Water Shower Cold': Retrieved counter: 0.00 pulses/min
[12:01:34][D][sensor:121]: 'Bathroom Water Shower Cold': Sending state 0.00000 L/min with 2 decimals of accuracy
[12:01:34][D][sensor:121]: 'Bathroom Water Shower Cold Total': Sending state 0.00000 L/day with 2 decimals of accuracy
[12:01:34][D][pulse_counter:160]: 'Bathroom Water Sink Warm': Retrieved counter: 0.00 pulses/min
[12:01:34][D][sensor:121]: 'Bathroom Water Sink Warm': Sending state 0.00000 L/min with 2 decimals of accuracy
[12:01:34][D][sensor:121]: 'Bathroom Water Sink Warm Total': Sending state 0.00000 L/day with 2 decimals of accuracy
[12:01:34][D][sensor:121]: 'Bathroom INA226 Bus Voltage': Sending state 12.05250 V with 1 decimals of accuracy
[12:01:34][D][sensor:121]: 'Bathroom INA226 Shunt Voltage': Sending state 0.00000 V with 2 decimals of accuracy
[12:01:34][D][sensor:121]: 'Bathroom INA226 Current': Sending state 0.00000 A with 2 decimals of accuracy
[12:01:34][D][sensor:121]: 'Bathroom INA226 Power': Sending state 0.00000 W with 1 decimals of accuracy
```

OttoWinter, 2021-10-03:
Can you check if the following fixes it?

```yaml
# put this in the YAML (only temporarily until fixed!)
external_components:
  - source: github://esphome/esphome@fix-ina226-idf
    components: [ina226]
```

thanks!

reaper7, 2021-10-03:
@OttoWinter - unfortunately, it doesn't change anything,
direct file replacement (ina226.cpp, ina226.h) too :(

OttoWinter, 2021-10-04:
> maybe there should be some delay between the readings of subsequent parameters?

Maybe, but there wasn't one before either, so I wouldn't think so.

reaper7, 2021-10-04:
I also tested the latest i2c patch https://github.com/esphome/esphome/commit/5c06cd8eb32ccbde3d3c9605ba70d302a7ae18b1,
together two fixes (ina226 and i2c) and each separately. Still nothing ...

during compilation i noticed some warning regarding i2c:
```
$ esphome run bathroom141.yaml
INFO Reading configuration bathroom141.yaml...
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d bathroom141
Processing bathroom141 (board: mhetesp32minikit; framework: arduino; platform: espressif32 @ 3.3.2)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif32 3.10006.210326 (1.0.6)
 - tool-esptoolpy 1.30100.210531 (3.1.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF Modes: Finder ~ off, Compatibility ~ soft
Library Manager: Installing Hash
Library Manager: Already installed, built-in library
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 1.3.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <Wire> 1.0.1
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <ESPmDNS> 1.0
Compiling .pioenvs\bathroom141\src\esphome\components\adc\adc_sensor.cpp.o
Compiling .pioenvs\bathroom141\src\esphome\components\api\api_connection.cpp.o
.
.
.
Compiling .pioenvs\bathroom141\FrameworkArduino\cbuf.cpp.o
Compiling .pioenvs\bathroom141\FrameworkArduino\esp32-hal-adc.c.o
Compiling .pioenvs\bathroom141\FrameworkArduino\esp32-hal-bt.c.o
Compiling .pioenvs\bathroom141\FrameworkArduino\esp32-hal-cpu.c.o
Compiling .pioenvs\bathroom141\FrameworkArduino\esp32-hal-dac.c.o
Compiling .pioenvs\bathroom141\FrameworkArduino\esp32-hal-gpio.c.o
Compiling .pioenvs\bathroom141\FrameworkArduino\esp32-hal-i2c.c.o
C:\Users\reaper7\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-i2c.c: In function 'i2cDumpFifo':
C:\Users\reaper7\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-i2c.c:415:35: warning: ' ' flag used with '%X' gnu_printf format [-Wformat=]
       } else k += sprintf(&buf[k],"% 4X ",fifoBuffer[i]);
                                   ^
C:\Users\reaper7\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-i2c.c:415:35: warning: ' ' flag used with '%X' gnu_printf format [-Wformat=]
Compiling .pioenvs\bathroom141\FrameworkArduino\esp32-hal-ledc.c.o
Compiling .pioenvs\bathroom141\FrameworkArduino\esp32-hal-log.c.o
Compiling .pioenvs\bathroom141\FrameworkArduino\esp32-hal-matrix.c.o
Compiling .pioenvs\bathroom141\FrameworkArduino\esp32-hal-misc.c.o
Compiling .pioenvs\bathroom141\FrameworkArduino\esp32-hal-psram.c.o
Compiling .pioenvs\bathroom141\FrameworkArduino\esp32-hal-rmt.c.o
Compiling .pioenvs\bathroom141\FrameworkArduino\esp32-hal-sigmadelta.c.o
Compiling .pioenvs\bathroom141\FrameworkArduino\esp32-hal-spi.c.o
Compiling .pioenvs\bathroom141\FrameworkArduino\esp32-hal-time.c.o
C:\Users\reaper7\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c: In function 'spiTransferBytesNL':
C:\Users\reaper7\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
C:\Users\reaper7\.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^
Compiling .pioenvs\bathroom141\FrameworkArduino\esp32-hal-timer.c.o
Compiling .pioenvs\bathroom141\FrameworkArduino\esp32-hal-touch.c.o
Compiling .pioenvs\bathroom141\FrameworkArduino\esp32-hal-uart.c.o
Compiling .pioenvs\bathroom141\FrameworkArduino\libb64\cdecode.c.o
Compiling .pioenvs\bathroom141\FrameworkArduino\libb64\cencode.c.o
Compiling .pioenvs\bathroom141\FrameworkArduino\main.cpp.o
Compiling .pioenvs\bathroom141\FrameworkArduino\stdlib_noniso.c.o
Compiling .pioenvs\bathroom141\FrameworkArduino\wiring_pulse.c.o
Compiling .pioenvs\bathroom141\FrameworkArduino\wiring_shift.c.o
Archiving .pioenvs\bathroom141\libFrameworkArduino.a
Linking .pioenvs\bathroom141\firmware.elf
Retrieving maximum program size .pioenvs\bathroom141\firmware.elf
Checking size .pioenvs\bathroom141\firmware.elf
RAM:   [=         ]  14.0% (used 46028 bytes from 327680 bytes)
Flash: [======    ]  59.0% (used 1082062 bytes from 1835008 bytes)
Building .pioenvs\bathroom141\firmware.bin
Merged 1 ELF section
========================= [SUCCESS] Took 46.61 seconds =========================
```

reaper7, 2021-10-05:
very verbose log from 2021.9.2:
```
[15:34:36][I][app:102]: ESPHome version 2021.9.2 compiled on Oct  5 2021, 15:33:55
[15:34:36][C][status_led:019]: Status LED:
[15:34:36][C][status_led:020]:   Pin: GPIO2 (Mode: OUTPUT)
[15:34:36][C][i2c:028]: I2C Bus:
[15:34:36][C][i2c:029]:   SDA Pin: GPIO26
[15:34:36][C][i2c:030]:   SCL Pin: GPIO5
[15:34:36][C][i2c:031]:   Frequency: 50000 Hz
[15:34:36][C][uptime.sensor:030]: Uptime Sensor 'Bathroom Controller Uptime Sensor'
[15:34:36][C][uptime.sensor:030]:   State Class: 'total_increasing'
[15:34:36][C][uptime.sensor:030]:   Unit of Measurement: 's'
[15:34:36][C][uptime.sensor:030]:   Accuracy Decimals: 0
[15:34:36][C][uptime.sensor:030]:   Icon: 'mdi:timer-outline'
[15:34:36][V][uptime.sensor:030]:   Unique ID: 'ac67b23687fc-uptime'
[15:34:36][C][ledc.output:032]: LEDC Output:
[15:34:36][C][ledc.output:033]:   Pin GPIO18 (Mode: OUTPUT)
[15:34:36][C][ledc.output:034]:   LEDC Channel: 0
[15:34:36][C][ledc.output:035]:   Frequency: 2441.0 Hz
[15:34:36][C][logger:193]: Logger:
[15:34:36][C][logger:194]:   Level: VERY_VERBOSE
[15:34:36][C][logger:195]:   Log Baud Rate: 0
[15:34:36][C][logger:196]:   Hardware UART: UART0
[15:34:36][C][light:097]: Light 'Bathroom Light Strip Shower'
[15:34:36][C][light:099]:   Default Transition Length: 1.0s
[15:34:36][C][light:100]:   Gamma Correct: 0.00
[15:34:36][C][ina226:079]: INA226:
[15:34:36][C][ina226:080]:   Address: 0x40
[15:34:36][C][ina226:086]:   Update Interval: 5.0s
[15:34:36][C][ina226:088]:   Bus Voltage 'Bathroom INA226 Bus Voltage'
[15:34:36][C][ina226:088]:     Device Class: 'voltage'
[15:34:36][C][ina226:088]:     State Class: 'measurement'
[15:34:36][C][ina226:088]:     Unit of Measurement: 'V'
[15:34:36][C][ina226:088]:     Accuracy Decimals: 1
[15:34:36][C][ina226:089]:   Shunt Voltage 'Bathroom INA226 Shunt Voltage'
[15:34:36][C][ina226:089]:     Device Class: 'voltage'
[15:34:36][C][ina226:089]:     State Class: 'measurement'
[15:34:36][C][ina226:089]:     Unit of Measurement: 'V'
[15:34:36][C][ina226:089]:     Accuracy Decimals: 2
[15:34:36][C][ina226:090]:   Current 'Bathroom INA226 Current'
[15:34:36][C][ina226:090]:     Device Class: 'current'
[15:34:36][C][ina226:090]:     State Class: 'measurement'
[15:34:36][C][ina226:090]:     Unit of Measurement: 'A'
[15:34:36][C][ina226:090]:     Accuracy Decimals: 2
[15:34:36][C][ina226:091]:   Power 'Bathroom INA226 Power'
[15:34:36][C][ina226:091]:     Device Class: 'power'
[15:34:36][C][ina226:091]:     State Class: 'measurement'
[15:34:36][C][ina226:091]:     Unit of Measurement: 'W'
[15:34:36][C][ina226:091]:     Accuracy Decimals: 1
[15:34:36][C][status:034]: Status Binary Sensor 'Bathroom Controller Status'
[15:34:36][C][status:034]:   Device Class: 'connectivity'
[15:34:36][C][restart:021]: Restart Switch 'Bathroom Controller Restart'
[15:34:36][C][restart:021]:   Icon: 'mdi:restart-alert'
[15:34:36][C][captive_portal:148]: Captive Portal:
[15:34:36][C][web_server:152]: Web Server:
[15:34:36][C][web_server:153]:   Address: 192.168.0.141:80
[15:34:36][C][ota:029]: Over-The-Air Updates:
[15:34:36][C][ota:030]:   Address: 192.168.0.141:3232
[15:34:36][C][api:135]: API Server:
[15:34:36][C][api:136]:   Address: 192.168.0.141:6053
[15:34:36][C][wifi_signal.sensor:009]: WiFi Signal 'Bathroom Controller WiFi Status'
[15:34:36][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[15:34:36][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[15:34:36][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[15:34:36][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[15:34:36][V][wifi_signal.sensor:009]:   Unique ID: 'ac67b23687fc-wifisignal'
[15:34:38][VV][scheduler:151]: Running interval '' with interval=10000 last_execution=1028 (now=11035)
[15:34:40][VV][scheduler:151]: Running interval 'update' with interval=5000 last_execution=8412 (now=13413)
[15:34:40][VV][i2c:056]: Beginning Transmission to 0x40:
[15:34:40][VV][i2c:093]:     Writing 0b00000010 (0x02)
[... 632 lines cut ...]
[15:43:01][V][sensor:070]: 'Bathroom INA226 Shunt Voltage': Received new state 0.000023
[15:43:01][D][sensor:121]: 'Bathroom INA226 Shunt Voltage': Sending state 0.00002 V with 2 decimals of accuracy
[15:43:01][VV][i2c.arduino:121]: 0x40 TX 04
[15:43:01][VV][i2c.arduino:097]: 0x40 RX 0000
[15:43:01][V][sensor:070]: 'Bathroom INA226 Current': Received new state 0.000000
[15:43:01][VV][sensor.filter:013]: Filter(0x3ffb8564)::input(0.000000)
[15:43:01][VV][sensor.filter:020]: Filter(0x3ffb8564)::output(0.000000) -> SENSOR
[15:43:01][D][sensor:121]: 'Bathroom INA226 Current': Sending state 0.00000 A with 2 decimals of accuracy
[15:43:01][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 4101048820
  state: 0
  missing_state: NO
}
[15:43:02][VV][i2c.arduino:121]: 0x40 TX 03
[15:43:02][VV][i2c.arduino:097]: 0x40 RX 0000
[15:43:02][V][sensor:070]: 'Bathroom INA226 Power': Received new state 0.000000
[15:43:02][VV][sensor.filter:013]: Filter(0x3ffb8fc0)::input(0.000000)
[15:43:02][VV][sensor.filter:020]: Filter(0x3ffb8fc0)::output(0.000000) -> SENSOR
[15:43:02][D][sensor:121]: 'Bathroom INA226 Power': Sending state 0.00000 W with 1 decimals of accuracy
[15:43:02][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 1445225900
  state: 0
  missing_state: NO
}
```

reaper7, 2021-10-05:
i also tried to remove the voltage and shunt voltage, leaving only current and power, but still gets 0 value

OttoWinter, 2021-10-06:
That is really strange :/

I looked over the code over and over again but I can't see anything different... All I see is that the sensor just outputs a different value on the second request

The only thing that got added in _recent_ dev was i2c bus restoring.

If possible, try again with _latest_ dev (includes the i2c better logging), cusotm ina226 and remove the following line:

https://github.com/esphome/esphome/blob/22e3bc7cfe75501004148826c666f80fa3123016/esphome/components/i2c/i2c_bus_arduino.cpp#L14

reaper7, 2021-10-06:
@OttoWinter - removing recovery from i2c setup nothing changes ...
(except that I have additional information in the log:  _[i2c.arduino:041]:   Recovery: failed, SCL is held low on the bus_  )

This is problem with **INA226Component::setup()** or maybe better in **write_byte_16**
looks like 16 bit values for config and calibration have hi/lo bytes swapped.

I added an additional, "second setup" in INA226Component::dump_config()
```
  // 0b0000001000000000;
  // 0b0000000100000000;
  // 0b0000000000100000;
  // 0b0000000000000111;
  // -------------------
  // 0b0000001100100111; => 0x0327
  //this->write_byte_16(INA226_REGISTER_CONFIG, 0x0327);
  this->write_byte_16(INA226_REGISTER_CONFIG, 0x2703);

  uint16_t tmp_read = 0x0000;
  if (this->read_ina226_reg_(INA226_REGISTER_CONFIG, &tmp_read)) {
    ESP_LOGVV(TAG, "0x%02X RX 0x%02X", INA226_REGISTER_CONFIG, tmp_read);
  }

  //this->write_byte_16(INA226_REGISTER_CALIBRATION, 0x0800);
  this->write_byte_16(INA226_REGISTER_CALIBRATION, 0x0008);

  if (this->read_ina226_reg_(INA226_REGISTER_CALIBRATION, &tmp_read)) {
    ESP_LOGVV(TAG, "0x%02X RX 0x%02X", INA226_REGISTER_CALIBRATION, tmp_read);
  }
```

and now I get valid values for current and power:
```
[17:54:43][C][ina226:080]: INA226:
[17:54:43][C][ina226:081]:   Address: 0x40
[17:54:43][C][ina226:087]:   Update Interval: 5.0s
[17:54:43][C][ina226:089]:   Bus Voltage 'Bathroom INA226 Bus Voltage'
[17:54:43][C][ina226:089]:     Device Class: 'voltage'
[17:54:43][C][ina226:089]:     State Class: 'measurement'
[17:54:43][C][ina226:089]:     Unit of Measurement: 'V'
[17:54:43][C][ina226:089]:     Accuracy Decimals: 1
[17:54:43][C][ina226:090]:   Shunt Voltage 'Bathroom INA226 Shunt Voltage'
[17:54:43][C][ina226:090]:     Device Class: 'voltage'
[17:54:43][C][ina226:090]:     State Class: 'measurement'
[17:54:43][C][ina226:090]:     Unit of Measurement: 'V'
[17:54:43][C][ina226:090]:     Accuracy Decimals: 2
[17:54:43][C][ina226:091]:   Current 'Bathroom INA226 Current'
[17:54:43][C][ina226:091]:     Device Class: 'current'
[17:54:43][C][ina226:091]:     State Class: 'measurement'
[17:54:43][C][ina226:091]:     Unit of Measurement: 'A'
[17:54:43][C][ina226:091]:     Accuracy Decimals: 2
[17:54:43][C][ina226:092]:   Power 'Bathroom INA226 Power'
[17:54:43][C][ina226:092]:     Device Class: 'power'
[17:54:43][C][ina226:092]:     State Class: 'measurement'
[17:54:43][C][ina226:092]:     Unit of Measurement: 'W'
[17:54:43][C][ina226:092]:     Accuracy Decimals: 1
[17:54:43][VV][i2c.arduino:121]: 0x40 TX 000080
[17:54:43][VV][i2c.arduino:121]: 0x40 TX 000327
[17:54:43][VV][i2c.arduino:121]: 0x40 TX 00
[17:54:43][VV][i2c.arduino:097]: 0x40 RX 4327
[17:54:43][VV][ina226:107]: 0x00 RX 0x4327
[17:54:43][VV][i2c.arduino:121]: 0x40 TX 050800
[17:54:43][VV][i2c.arduino:121]: 0x40 TX 05
[17:54:43][VV][i2c.arduino:097]: 0x40 RX 0800
[17:54:43][VV][ina226:114]: 0x05 RX 0x800
[17:54:43][VV][scheduler:152]: Running interval 'update' with interval=5000 last_execution=4200 (now=9367)
[17:54:43][VV][i2c.arduino:121]: 0x40 TX 02
[17:54:43][VV][i2c.arduino:097]: 0x40 RX 254F
[17:54:43][V][sensor:070]: 'Bathroom INA226 Bus Voltage': Received new state 11.938749
[17:54:43][D][sensor:121]: 'Bathroom INA226 Bus Voltage': Sending state 11.93875 V with 1 decimals of accuracy
[17:54:43][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 3669810102
  state: 11.9387
  missing_state: NO
}
[17:54:43][VV][i2c.arduino:121]: 0x40 TX 01
[17:54:43][VV][i2c.arduino:097]: 0x40 RX 2F39
[17:54:43][V][sensor:070]: 'Bathroom INA226 Shunt Voltage': Received new state 0.030222
[17:54:43][D][sensor:121]: 'Bathroom INA226 Shunt Voltage': Sending state 0.03022 V with 2 decimals of accuracy
[17:54:43][VV][i2c.arduino:121]: 0x40 TX 04
[17:54:43][VV][i2c.arduino:097]: 0x40 RX 2F3A
[17:54:43][V][sensor:070]: 'Bathroom INA226 Current': Received new state 3.022500
[17:54:43][VV][sensor.filter:013]: Filter(0x3ffb867c)::input(3.022500)
[17:54:43][VV][sensor.filter:020]: Filter(0x3ffb867c)::output(3.022500) -> SENSOR
[17:54:43][D][sensor:121]: 'Bathroom INA226 Current': Sending state 3.02250 A with 2 decimals of accuracy
[17:54:43][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 4101048820
  state: 3.0225
  missing_state: NO
}
[17:54:43][VV][i2c.arduino:121]: 0x40 TX 03
[17:54:43][VV][i2c.arduino:097]: 0x40 RX 168E
[17:54:43][V][sensor:070]: 'Bathroom INA226 Power': Received new state 36.087502
[17:54:43][VV][sensor.filter:013]: Filter(0x3ffb92b4)::input(36.087502)
[17:54:43][VV][sensor.filter:020]: Filter(0x3ffb92b4)::output(36.087502) -> SENSOR
[17:54:43][D][sensor:121]: 'Bathroom INA226 Power': Sending state 36.08750 W with 1 decimals of accuracy
[17:54:43][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 1445225900
  state: 36.0875
  missing_state: NO
}
```

OttoWinter, 2021-10-07:
Closing via https://github.com/esphome/esphome/pull/2460