esphome/issues #2508: Globals error on 2021.10.0-dev

markusg1234, 2021-10-04:
### The problem

For some reason the yaml below compiles with 2021.9.2 but on 2021.10.0-dev brings up an 
error: ISO C++ forbids comparison between pointer and integer [-fpermissive]

Regards,

Markus

### Which version of ESPHome has the issue?

v2021.10.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.10.0b3

### What platform are you using?

ESP32

### Board

nodemcu-32s

### Component causing the issue

globals

### Example YAML snippet

```yaml
esphome:
  name: computer-room-blind
  platform: ESP32
  board: nodemcu-32s 

  on_boot:
    then:
        - stepper.report_position:
            id: computer_room_stepper
            position: !lambda |-
              if (id(computer_room_stepper_global) < 1) {
                return 0;
              } else {
                return 950;
              }
        - stepper.set_target:
            id: computer_room_stepper
            target: !lambda |-
              if (id(computer_room_stepper_global) < 1) {
                return 0;
              } else {
                return 950;
              }

        - lambda: |-
           id(computer_room_cover).publish_state(computer_room_cover_global);  
#          id(computer_room_stepper).report_position(computer_room_stepper_global);  
# Enable logging
logger:

api:
  password: "*"
ota:
  password: "*"
[... 488 lines cut ...]
/root/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c: In function 'spiTransferBytesNL':
/root/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
/root/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^
Compiling /data/computer-room-blind/.pioenvs/computer-room-blind/FrameworkArduino/esp32-hal-timer.c.o
Compiling /data/computer-room-blind/.pioenvs/computer-room-blind/FrameworkArduino/esp32-hal-touch.c.o
Compiling /data/computer-room-blind/.pioenvs/computer-room-blind/FrameworkArduino/esp32-hal-uart.c.o
Compiling /data/computer-room-blind/.pioenvs/computer-room-blind/FrameworkArduino/libb64/cdecode.c.o
Compiling /data/computer-room-blind/.pioenvs/computer-room-blind/FrameworkArduino/libb64/cencode.c.o
Compiling /data/computer-room-blind/.pioenvs/computer-room-blind/FrameworkArduino/main.cpp.o
Compiling /data/computer-room-blind/.pioenvs/computer-room-blind/FrameworkArduino/stdlib_noniso.c.o
Compiling /data/computer-room-blind/.pioenvs/computer-room-blind/FrameworkArduino/wiring_pulse.c.o
Compiling /data/computer-room-blind/.pioenvs/computer-room-blind/FrameworkArduino/wiring_shift.c.o
Archiving /data/computer-room-blind/.pioenvs/computer-room-blind/libFrameworkArduino.a
Linking /data/computer-room-blind/.pioenvs/computer-room-blind/firmware.elf
Retrieving maximum program size /data/computer-room-blind/.pioenvs/computer-room-blind/firmware.elf
Checking size /data/computer-room-blind/.pioenvs/computer-room-blind/firmware.elf
RAM:   [==        ]  16.4% (used 53748 bytes from 327680 bytes)
Flash: [========  ]  83.5% (used 1532350 bytes from 1835008 bytes)
Building /data/computer-room-blind/.pioenvs/computer-room-blind/firmware.bin
========================= [SUCCESS] Took 37.06 seconds =========================
INFO Successfully compiled program.

peterforeman, 2021-10-04:
I have this same issue in my "energy" (dsmr) esp32. No compile issues with latest version, but with dev:

```
globals: 
  - id: power_max
    type: int
    restore_value: true
    initial_value: "0"
```

```
    power_delivered:
      id: power_delivered
      name: "Power Delivered"
      filters:
        - multiply: 1000
      unit_of_measurement: "W"
      accuracy_decimals: 0
      on_value:
        then:
          - lambda: |-
              if ( x > id(power_max)) {
                id(power_max) = x;
                id(actual_delivery_today_max).update();
              }
```

Gives compile errors:

```
/config/energy.yaml:116:16: error: invalid operands of types 'float' and 'esphome::globals::RestoringGlobalsComponent<int>*' to binary 'operator>'
               if ( x > id(power_max)) {
                ^
/config/energy.yaml:117:19: error: cannot convert 'float' to 'esphome::globals::RestoringGlobalsComponent<int>*' in assignment
                 id(power_max) = x;
```

OttoWinter, 2021-10-04:
Fixed by https://github.com/esphome/esphome/pull/2442

closed 2021-10-04 by OttoWinter