esphome/issues #2509: Nightly build 2021-10-0, missing WiFiType.h

SeByDocKy, 2021-10-04:
### The problem

Compilation Error linked with a missing file

### Which version of ESPHome has the issue?

2021.10.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.7

### What platform are you using?

ESP32

### Board

WROOM32

### Component causing the issue

Wifi

### Example YAML snippet

```yaml
esphome:
  name: station_meteo
  platform: ESP32
  board: esp32dev

  includes:
    - veml6070_custom_sensor.h
    - tx23_custom_sensor.h
  libraries:
    - "https://github.com/adafruit/Adafruit_VEML6070"
    - "https://github.com/egzumer/Arduino-LaCrosse-TX23-Library"
    - SPI
    
  on_boot:
    priority: -100
    then:
      - switch.turn_on: reset_GPIO02
      - delay: 1000ms
      - switch.turn_off: reset_GPIO02  
 
wifi:
  ssid: !secret esphome_ssid
  password: !secret esphome_password
  
# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:

ota:

time:

  - platform: homeassistant
    id: homeassistant_time  
    
    on_time:
[... 234 lines cut ...]
Compiling /data/station_meteo/.pioenvs/station_meteo/src/esphome/components/network/util.cpp.o
Compiling /data/station_meteo/.pioenvs/station_meteo/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
Compiling /data/station_meteo/.pioenvs/station_meteo/src/esphome/components/ota/ota_backend_arduino_esp8266.cpp.o
Compiling /data/station_meteo/.pioenvs/station_meteo/src/esphome/components/ota/ota_backend_esp_idf.cpp.o
Compiling /data/station_meteo/.pioenvs/station_meteo/src/esphome/components/ota/ota_component.cpp.o
Compiling /data/station_meteo/.pioenvs/station_meteo/src/esphome/components/pulse_counter/pulse_counter_sensor.cpp.o
In file included from src/esphome/components/network/util.cpp:5:0:
src/esphome/components/wifi/wifi_component.h:13:22: fatal error: WiFiType.h: No such file or directory

******************************************************************
* Looking for WiFiType.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFiType.h"
* Web  > https://platformio.org/lib/search?query=header:WiFiType.h
*
******************************************************************

compilation terminated.
*** [/data/station_meteo/.pioenvs/station_meteo/src/esphome/components/network/util.cpp.o] Error 1
========================= [FAILED] Took 12.31 seconds =========================
```

### Additional information

i) Compile and work fine with the stable version

OttoWinter, 2021-10-04:
Fixed by https://github.com/esphome/esphome/pull/2441

closed 2021-10-04 by OttoWinter

SeByDocKy, 2021-10-04:
@OttoWinter 

I just installed the las dev with this hotfix .... but still have error :(

INFO Reading configuration /config/esphome/station_meteo.yaml...
WARNING 'station_meteo': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
INFO Detected timezone 'Europe/Paris'
INFO Generating C++ source...
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/station_meteo
Processing station_meteo (board: esp32dev; framework: arduino; platform: espressif32 @ 3.3.2)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
LDF Modes: Finder ~ off, Compatibility ~ soft
Library Manager: Installing Hash
Library Manager: Already installed, built-in library
Dependency Graph
|-- <Adafruit VEML6070 Library> 1.0.6+sha.ecd2a5e
|-- <La Crosse RX23 Library> 1.0.0+sha.92ce726
|-- <SPI> 1.0
|-- <Update> 1.0
|-- <Wire> 1.0.1
|-- <ESPmDNS> 1.0
Compiling /data/station_meteo/.pioenvs/station_meteo/src/esphome/components/network/util.cpp.o
In file included from src/esphome/components/network/util.cpp:5:0:
src/esphome/components/wifi/wifi_component.h:13:22: fatal error: WiFiType.h: No such file or directory

******************************************************************
* Looking for WiFiType.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:WiFiType.h"
* Web  > https://platformio.org/lib/search?query=header:WiFiType.h
*
******************************************************************

compilation terminated.
Compiling /data/station_meteo/.pioenvs/station_meteo/src/esphome/components/template/sensor/template_sensor.cpp.o
Compiling /data/station_meteo/.pioenvs/station_meteo/src/esphome/components/template/text_sensor/template_text_sensor.cpp.o
*** [/data/station_meteo/.pioenvs/station_meteo/src/esphome/components/network/util.cpp.o] Error 1
Compiling /data/station_meteo/.pioenvs/station_meteo/src/esphome/components/text_sensor/filter.cpp.o
========================= [FAILED] Took 10.57 seconds =========================

OttoWinter, 2021-10-04:
How did you install the latest dev?

If docker, note that the dev docker image is built once a day, so it can take up to 24h

SeByDocKy, 2021-10-04:
> How did you install the latest dev?
> 
> If docker, note that the dev docker image is built once a day, so it can take up to 24h

via the HASSIO & the rebuild button. Not the right way ?