esphome/issues #2510: Compilation error in a custom sensor using "millis()"

SeByDocKy, 2021-10-04:
### The problem

Compilation error .... Compile fine with the stable release

### Which version of ESPHome has the issue?

2021.10.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.7

### What platform are you using?

ESP32

### Board

WROOM32

### Example YAML snippet

```yaml
esphome:
  name: antbms
  platform: ESP32
  board: esp32dev
  includes:
    - antbms_custom_sensor.h

wifi:
  ssid:  !secret esphome_ssid  #HOME
  password: !secret esphome_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

switch:
  - platform: restart
    name: "ANTBMS_restart"
    
sensor:
  - platform: custom
    id: ANTBMS_id
    lambda: |-
     auto ANTBMS = new ANTBMSCustomSensor();
     App.register_component(ANTBMS);
     return {ANTBMS->ANTBMS_nbr_cell , ANTBMS->ANTBMS_soc , ANTBMS->ANTBMS_cell_av , ANTBMS->ANTBMS_cell_min ,  ANTBMS->ANTBMS_cell_max ,  ANTBMS->ANTBMS_cell_1 , ANTBMS->ANTBMS_cell_2 , ANTBMS->ANTBMS_cell_3 , ANTBMS->ANTBMS_cell_4 , ANTBMS->ANTBMS_cell_5 , ANTBMS->ANTBMS_cell_6 , ANTBMS->ANTBMS_cell_7 , ANTBMS->ANTBMS_cell_8 , ANTBMS->ANTBMS_cell_9 , ANTBMS->ANTBMS_cell_10 , ANTBMS->ANTBMS_cell_11 , ANTBMS->ANTBMS_cell_12 , ANTBMS->ANTBMS_cell_13 , ANTBMS->ANTBMS_cell_14 , ANTBMS->ANTBMS_cell_15 , ANTBMS->ANTBMS_cell_16 , ANTBMS->ANTBMS_t_bal , ANTBMS->ANTBMS_t_fet , ANTBMS->ANTBMS_t_1 , ANTBMS->ANTBMS_t_2 , ANTBMS->ANTBMS_t_3 , ANTBMS->ANTBMS_t_4 , ANTBMS->ANTBMS_puiss , ANTBMS->ANTBMS_volt , ANTBMS->ANTBMS_amp , ANTBMS->ANTBMS_Ah_install , ANTBMS->ANTBMS_Ah_rest , ANTBMS->ANTBMS_Ah_Total , ANTBMS->ANTBMS_charge_idx ,  ANTBMS->ANTBMS_discharge_idx , ANTBMS->ANTBMS_balance_idx , ANTBMS->ANTBMS_cells_balanced_idx};

    sensors:  
    - name: "ANTBMS_nbr_cell"  #1
      accuracy_decimals: 1
      icon: mdi:flash
      
    - name: "ANTBMS_soc"   #2
      unit_of_measurement: "%"
      accuracy_decimals: 1
[... 559 lines cut ...]
		cell_2 = ((((uint8_t)incomingByte[8])<< 8) + (uint8_t)incomingByte[9]); 
		cell_3 = ((((uint8_t)incomingByte[10])<< 8) + (uint8_t)incomingByte[11]); 
		cell_4 = ((((uint8_t)incomingByte[12])<< 8) + (uint8_t)incomingByte[13]); 
		cell_5 = ((((uint8_t)incomingByte[14])<< 8) + (uint8_t)incomingByte[15]); 
		cell_6 = ((((uint8_t)incomingByte[16])<< 8) + (uint8_t)incomingByte[17]); 
		cell_7 = ((((uint8_t)incomingByte[18])<< 8) + (uint8_t)incomingByte[19]); 
		cell_8 = ((((uint8_t)incomingByte[20])<< 8) + (uint8_t)incomingByte[21]); 
		cell_9 = ((((uint8_t)incomingByte[22])<< 8) + (uint8_t)incomingByte[23]); 
		cell_10 = ((((uint8_t)incomingByte[24])<< 8) + (uint8_t)incomingByte[25]); 
		cell_11 = ((((uint8_t)incomingByte[26])<< 8) + (uint8_t)incomingByte[27]); 
		cell_12 = ((((uint8_t)incomingByte[28])<< 8) + (uint8_t)incomingByte[29]); 
		cell_13 = ((((uint8_t)incomingByte[30])<< 8) + (uint8_t)incomingByte[31]); 
		cell_14 = ((((uint8_t)incomingByte[32])<< 8) + (uint8_t)incomingByte[33]);
        cell_15 = ((((uint8_t)incomingByte[34])<< 8) + (uint8_t)incomingByte[35]);		
		cell_16 = ((((uint8_t)incomingByte[36])<< 8) + (uint8_t)incomingByte[37]);
		Ah_install = ((((uint8_t)incomingByte[75])<< 24) + (((uint8_t)incomingByte[76])<< 16)+ (((uint8_t)incomingByte[77])<< 8)+ ((uint8_t)incomingByte[78]));
		Ah_rest = ((((uint8_t)incomingByte[79])<< 24) + (((uint8_t)incomingByte[80])<< 16)+ (((uint8_t)incomingByte[81])<< 8)+ ((uint8_t)incomingByte[82]));
		Ah_Total = ((((uint8_t)incomingByte[83])<< 24) + (((uint8_t)incomingByte[84])<< 16)+ (((uint8_t)incomingByte[85])<< 8)+ ((uint8_t)incomingByte[86]));
		
		charge_idx = (uint8_t)incomingByte[103];
		discharge_idx = (uint8_t)incomingByte[104];
		balance_idx = (uint8_t)incomingByte[105];
		cells_balanced_idx = ((((uint8_t)incomingByte[132])<< 8) + (uint8_t)incomingByte[135]);
	}
};```

jesserockz, 2021-10-04:
You need to `#include "esphome/core/hal.h"`

SeByDocKy, 2021-10-04:
> You need to `#include "esphome/core/hal.h"`

still the same error with this #include :(

SeByDocKy, 2021-10-04:
Ok .... I solved the problem ....

Need to use esphome::milli()

jesserockz, 2021-10-04:
or you should have your custom component inside the `esphome` namespace

SeByDocKy, 2021-10-05:
> or you should have your custom component inside the `esphome` namespace

How to proceed ?

OttoWinter, 2021-10-05:
Yeah ideally we wouldn't break user setups (for reference, the problem is that we do `using namespace esphome` globally in main.cpp so that custom components don't need to write `esphome::` everywhere, but then `delay` is ambiguous with `esphome::delay` vs arduino's top-level `delay`).

Maybe we can do a work around (not a nice one, but should probably work (injected in `main.cpp` before the custom includes)

```cpp
#define delay(x) esphome::delay(x)
#define delayMicroseconds(x) esphome::delayMicroseconds(x)
#define millis() esphome::millis()
```

I _think_ that should work for most cases, and at the same time wouldn't interfere with normal esphome codebase.

SeByDocKy, 2021-10-05:
> Yeah ideally we wouldn't break user setups (for reference, the problem is that we do `using namespace esphome` globally in main.cpp so that custom components don't need to write `esphome::` everywhere, but then `delay` is ambiguous with `esphome::delay` vs arduino's top-level `delay`).
> 
> Maybe we can do a work around (not a nice one, but should probably work (injected in `main.cpp` before the custom includes)
> 
> ```c++
> #define delay(x) esphome::delay(x)
> #define delayMicroseconds(x) esphome::delayMicroseconds(x)
> #define millis() esphome::millis()
> ```
> 
> I _think_ that should work for most cases, and at the same time wouldn't interfere with normal esphome codebase.

I did a trick like this by checking the ESPhome version ....

closed 2021-10-21 by OttoWinter