esphome/issues #2511: API constantly disconnecting from HA

SmarterHomeLife, 2021-10-05:
### The problem

I have lighting control input boards which each have 2 x MCP23017 chips providing 32 inputs, if I connect 1 board to the I2C bus it works and successfully connects to HA.
 
When connecting multiple MCP23017 boards the esphome API keeps connecting and disconnecting from HA but functionality of esphome device works as in I get no I2C errors, all binary_sensor inputs show correctly in the esphome logs and led functions work no all boards.

With 2 or more MCP23017 boards the esphome logs shows this constantly;

[12:55:56][D][api.connection:153]: Home Assistant 2021.10.0b5 (10.0.4.10) requested disconnected
[12:55:56][D][api:095]: Accepted 10.0.4.10
[12:55:56][V][api.connection:711]: Hello from client: 'Home Assistant 2021.10.0b5 (10.0.4.10)'
[12:55:56][V][api:107]: Removing connection to Home Assistant 2021.10.0b5 (10.0.4.10)
[12:55:56][D][api.connection:727]: Home Assistant 2021.10.0b5 (10.0.4.10): Connected successfully

And HA logs show;

2021-10-05 12:35:12 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.4.11: Timeout waiting for response for
2021-10-05 12:35:18 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.4.11: Timeout waiting for response for
2021-10-05 12:35:24 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.4.11: Timeout waiting for response for

### Which version of ESPHome has the issue?

2021.9.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.10.0b5 and 2021.9.7

### What platform are you using?

ESP32

### Board

D1 Mini ESP32 and QuinLED ESP32 are both the same

### Component causing the issue

API/I2C

### Example YAML snippet

```yaml
substitutions:
  devicename: lighting-control-d32-2b24
  name: Lighting Control 2B24
  static_ip: 10.0.4.11

esphome:
  name: $devicename
  platform: ESP32
  board: wemos_d1_mini32

  on_boot:
    priority: -10
    then:
      - light.turn_on:
          id: led
          brightness: 100%
          red: 100%
          blue: 100%
          green: 100%
          effect: random

wifi:
  ssid: !secret ssid
[... 522 lines cut ...]

2021-10-05 12:35:12 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.4.11: Timeout waiting for response for
2021-10-05 12:35:18 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.4.11: Timeout waiting for response for
2021-10-05 12:35:24 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.4.11: Timeout waiting for response for
2021-10-05 12:35:31 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.4.11: Timeout waiting for response for
2021-10-05 12:35:36 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.4.11: Timeout waiting for response for
2021-10-05 12:35:42 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.4.11: Timeout waiting for response for
2021-10-05 12:35:48 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.4.11: Timeout waiting for response for
2021-10-05 12:35:53 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.4.11: Timeout waiting for response for
2021-10-05 12:36:00 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.4.11: Timeout waiting for response for
2021-10-05 12:36:05 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.4.11: Timeout waiting for response for
2021-10-05 12:36:11 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.4.11: Timeout waiting for response for
2021-10-05 12:36:18 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.4.11: Timeout waiting for response for
2021-10-05 12:36:25 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.4.11: Timeout waiting for response for
2021-10-05 12:36:31 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.4.11: Timeout waiting for response for
2021-10-05 12:36:37 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.4.11: Timeout waiting for response for
2021-10-05 12:36:43 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.4.11: Timeout waiting for response for
2021-10-05 12:36:50 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.4.11: Timeout waiting for response for
2021-10-05 12:36:55 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.4.11: Timeout waiting for response for
2021-10-05 12:37:02 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 10.0.4.11: Timeout waiting for response for
```

### Additional information

I've tried multiple different esp32 boards as well as multiple different MCP23017 boards in combination and the results are always the same, I've tried all different frequencies on the I2C bus and also removing the web_server and other components to try to resolve this but nothing has worked so far, any help would really be appreciated.

OttoWinter, 2021-10-05:
Can you enable the following in your _HA_ `configuration.yaml` and report the result? Thanks!

```yaml
logger:
  logs:
    aioesphomeapi.connection: debug
```

SmarterHomeLife, 2021-10-05:
Hi @OttoWinter 

I've tried to grab relevant lines from the HA log below, is there a way I can narrow the search further?

Thanks Brad

2021-10-05 23:17:40 DEBUG (MainThread) [zeroconf] Received from 'fe80::1c6d:bcdb:6956:a76e':5353 [socket 30 (('::', 5353, 0, 0))]: <DNSIncoming:{id=0, flags=0, truncated=False, n_q=1, n_ans=0, n_auth=0, n_add=1, questions=[ptr[question,QM,in,_meshcop._udp.local.]], answers=[]}> (66 bytes) as [b'\x00\x00\x00\x00\x00\x01\x00\x00\x00\x00\x00\x01\x08_meshcop\x04_udp\x05local\x00\x00\x0c\x00\x01\x00\x00)\x05\xa0\x00\x00\x11\x94\x00\x12\x00\x04\x00\x0e\x00\xcb6Bb\x8e=\xf44Bb\x8e=\xf4']
2021-10-05 23:17:40 DEBUG (MainThread) [zeroconf] Received from '10.0.4.11':5353 [socket 24 (('10.0.4.10', 5353))]: <DNSIncoming:{id=0, flags=33792, truncated=False, n_q=0, n_ans=3, n_auth=0, n_add=1, questions=[], answers=[record[ptr,in,_esphomelib._tcp.local.]=4500/4499,lighting-control-d32-2b24._esphomelib._tcp.local., record[srv,in-unique,lighting-control-d32-2b24._esphomelib._tcp.local.]=120/119,lighting-control-d32-2b24.local.:6053, record[txt,in-unique,lighting-control-d32-2b24._esphomelib._tcp.local.]=4500/4499,b'\x15board='..., record[a,in-unique,lighting-control-d32-2b24.local.]=120/119,10.0.4.11]}> (237 bytes) as [b'\x00\x00\x84\x00\x00\x00\x00\x03\x00\x00\x00\x01\x0b_esphomelib\x04_tcp\x05local\x00\x00\x0c\x00\x01\x00\x00\x11\x94\x00\x1c\x19lighting-control-d32-2b24\xc0\x0c\xc0.\x00!\x80\x01\x00\x00\x00x\x00"\x00\x00\x00\x00\x17\xa5\x19lighting-control-d32-2b24\xc0\x1d\xc0.\x00\x10\x80\x01\x00\x00\x11\x94\x00Y\x15board=wemos_d1_mini32\x0eplatform=ESP32\x10mac=ac67b22d2b24\x11address=10.0.4.11\x10version=2021.9.2\xc0\\\x00\x01\x80\x01\x00\x00\x00x\x00\x04\n\x00\x04\x0b']
2021-10-05 23:17:40 DEBUG (MainThread) [homeassistant.components.zeroconf] service_update: type=_esphomelib._tcp.local. name=lighting-control-d32-2b24._esphomelib._tcp.local. state_change=ServiceStateChange.Added
2021-10-05 23:17:43 DEBUG (MainThread) [aioesphomeapi.connection] 10.0.4.11: Got message of type <class 'api_pb2.HelloResponse'>: api_version_major: 1
api_version_minor: 6
server_info: "lighting-control-d32-2b24 (esphome v2021.9.2)"
2021-10-05 23:17:43 DEBUG (MainThread) [aioesphomeapi.connection] 10.0.4.11: Successfully connected ('lighting-control-d32-2b24 (esphome v2021.9.2)' API=1.6)
2021-10-05 23:17:43 DEBUG (MainThread) [aioesphomeapi.connection] 10.0.4.11: Sending <class 'api_pb2.ConnectRequest'>: password: "1nn0v810N!"
2021-10-05 23:17:43 DEBUG (MainThread) [aioesphomeapi.connection] Sending frame 000c030a0a316e6e30763831304e21
2021-10-05 23:17:45 DEBUG (MainThread) [aioesphomeapi.connection] 10.0.4.11: Got message of type <class 'api_pb2.DeviceInfoResponse'>: uses_password: true
name: "lighting-control-d32-2b24"
mac_address: "AC:67:B2:2D:2B:24"
esphome_version: "2021.9.2"
compilation_time: "Oct 5 2021, 12:42:59"
model: "wemos_d1_mini32"
2021-10-05 23:17:45 DEBUG (MainThread) [aioesphomeapi.connection] 10.0.4.11: Sending <class 'api_pb2.ListEntitiesRequest'>:
2021-10-05 23:17:45 DEBUG (MainThread) [aioesphomeapi.connection] Sending frame 00000b
2021-10-05 23:17:45 DEBUG (MainThread) [aioesphomeapi.connection] lighting-control-d32-2b24 @ 10.0.4.11: Got message of type <class 'api_pb2.ListEntitiesBinarySensorResponse'>: object_id: "lighting_control_2b24_status"
key: 2856265233
name: "Lighting Control 2B24 Status"
unique_id: "lighting-control-d32-2b24binary_sensorlighting_control_2b24_status"
device_class: "connectivity"
is_status_binary_sensor: true
2021-10-05 23:17:45 DEBUG (MainThread) [aioesphomeapi.connection] lighting-control-d32-2b24 @ 10.0.4.11: Got message of type <class 'api_pb2.ListEntitiesBinarySensorResponse'>: object_id: "button_1"
key: 2661066113
name: "Button 1"
unique_id: "lighting-control-d32-2b24binary_sensorbutton_1"
2021-10-05 23:17:45 DEBUG (MainThread) [aioesphomeapi.connection] lighting-control-d32-2b24 @ 10.0.4.11: Got message of type <class 'api_pb2.ListEntitiesBinarySensorResponse'>: object_id: "button_2"
key: 2661066114
name: "Button 2"
unique_id: "lighting-control-d32-2b24binary_sensorbutton_2"
2021-10-05 23:17:45 DEBUG (MainThread) [aioesphomeapi.connection] lighting-control-d32-2b24 @ 10.0.4.11: Got message of type <class 'api_pb2.ListEntitiesBinarySensorResponse'>: object_id: "button_3"
key: 2661066115
name: "Button 3"
unique_id: "lighting-control-d32-2b24binary_sensorbutton_3"
2021-10-05 23:17:46 DEBUG (MainThread) [aioesphomeapi.connection] lighting-control-d32-2b24 @ 10.0.4.11: Got message of type <class 'api_pb2.ListEntitiesBinarySensorResponse'>: object_id: "button_6"
key: 2661066118
name: "Button 6"
unique_id: "lighting-control-d32-2b24binary_sensorbutton_6"
2021-10-05 23:17:46 DEBUG (MainThread) [aioesphomeapi.connection] lighting-control-d32-2b24 @ 10.0.4.11: Got message of type <class 'api_pb2.ListEntitiesBinarySensorResponse'>: object_id: "button_7"
key: 2661066119
name: "Button 7"
unique_id: "lighting-control-d32-2b24binary_sensorbutton_7"
2021-10-05 23:17:46 DEBUG (MainThread) [aioesphomeapi.connection] lighting-control-d32-2b24 @ 10.0.4.11: Got message of type <class 'api_pb2.ListEntitiesBinarySensorResponse'>: object_id: "button_8"
key: 2661066120
name: "Button 8"
unique_id: "lighting-control-d32-2b24binary_sensorbutton_8"
2021-10-05 23:17:46 DEBUG (MainThread) [aioesphomeapi.connection] lighting-control-d32-2b24 @ 10.0.4.11: Got message of type <class 'api_pb2.ListEntitiesBinarySensorResponse'>: object_id: "button_9"
key: 2661066121
name: "Button 9"
[... cut ...]
key: 949523758
name: "Button 86"
unique_id: "lighting-control-d32-2b24binary_sensorbutton_86"
2021-10-05 23:18:02 DEBUG (MainThread) [aioesphomeapi.connection] lighting-control-d32-2b24 @ 10.0.4.11: Got message of type <class 'api_pb2.DisconnectResponse'>:
2021-10-05 23:18:02 INFO (MainThread) [homeassistant.components.esphome] Disconnected from ESPHome API for 10.0.4.11
2021-10-05 23:18:02 DEBUG (MainThread) [aioesphomeapi.connection] lighting-control-d32-2b24 @ 10.0.4.11: Socket closed, stopping read loop

pinsdorf, 2021-12-04:
Hi, I have a similar issue with an 8266 (Wemos D1mini) operating an RC522 RFid reader over I2C. I'm running version 2021.11.5 on HA and version 2021.11.4 on EspHome.

I noticed the similar error message like OP in my logs: 
```
2021-12-03 17:21:32 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for 192.168.29.67 @ 192.168.29.67: Hello timed out
2021-12-03 17:25:02 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 192.168.29.67: Timeout waiting for response for <class 'api_pb2.ListEntitiesRequest'>
2021-12-03 17:25:21 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 192.168.29.67: Timeout waiting for response for <class 'api_pb2.ListEntitiesRequest'>

[... skipping similar entries, total 120 occurrences ...]

2021-12-03 18:00:19 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 192.168.29.67: Timeout waiting for response for <class 'api_pb2.ListEntitiesRequest'>
2021-12-03 18:00:38 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 192.168.29.67: Timeout waiting for response for <class 'api_pb2.ListEntitiesRequest'>
2021-12-03 18:00:56 WARNING (MainThread) [homeassistant.components.esphome] Error getting initial data for 192.168.29.67: Timeout waiting for response for <class 'api_pb2.ListEntitiesRequest'>
2021-12-03 18:01:19 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for 192.168.29.67 @ 192.168.29.67: Hello timed out
2021-12-03 18:08:36 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for 192.168.29.67 @ 192.168.29.67: Hello timed out
2021-12-03 18:36:11 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for 192.168.29.67 @ 192.168.29.67: Hello timed out
2021-12-03 18:39:54 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for 192.168.29.67 @ 192.168.29.67: Hello timed out
2021-12-03 18:49:44 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for 192.168.29.67 @ 192.168.29.67: Hello timed out
2021-12-03 21:02:13 WARNING (MainThread) [aioesphomeapi.reconnect_logic] Can't connect to ESPHome API for 192.168.29.61 @ 192.168.29.61: Error connecting to ('192.168.29.61', 6053): [Errno 113] Connect call failed ('192.168.29.61', 6053)
```

I see this happening frequently, maybe once a day, but don't know how to reproduce or avoid it. In my use case I usually take manual action and power cycle the device when I see that the esphome node does not report its connection state anymore.

I've followed a similar issue #1194 from 2020 and improved wifi connectivity without success.

I'm replying to this thread as it seems the most recent addressing this issue. 

@OttoWinter Is there any update? How can I help?