esphome/issues #2512: Last dev not connecting to I2C devices or returning bad or 0 values

SeByDocKy, 2021-10-05:
### The problem

I have at least two models with ESP32 + I2C sensors. WIth the last dev (2021-10-0), I2C sensors either is not scanned or returns 0 for my BH1750). For a ESP01m with ADS1115, voltage reading are approximatively divided by 4. Of course everything works flowless with the stable version.
For the BH1750, I can't even see the returned values... not even sure the sensors reading is done currectly (see log)

### Which version of ESPHome has the issue?

2021.10.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.7

### What platform are you using?

ESP32

### Board

WROOM32

### Component causing the issue

I2C BH1750, I2C ADS1115

### Example YAML snippet

```yaml
esphome:
  name: potager
  platform: ESP32
  board: esp32dev
  includes:
    - tinovi_esp32_custom_sensor.h

wifi:
  ssid: !secret esphome_ssid
  password: !secret esphome_password

# Enable logging

logger:
  #baud_rate: 0 #need this to free up UART pins
  #level: debug

# Enable Home Assistant API
api:

ota:

time:
  - platform: homeassistant
    id: homeassistant_time  
    
    on_time:
      - seconds: 55
        minutes: 59
        hours: 23
        then:  
          - sensor.template.publish:
              id: template_potager_veille
              state: !lambda return id(daily_potager).state;
              
      - seconds: /10
        then:
          - sensor.template.publish:
[... 715 lines cut ...]
                return ( float( id(mosquitto_battery).state )  > float(id(mosquitto_tension_seuil_haut).state) );
         then:
           - switch.turn_on: mosquitto_relay
           
    - if:
         condition:
            and:
             - lambda: |-
                return (  int( id(mosquitto_regulation).state)  == 2);
             - lambda: |-
                return (  int( id(mosquitto_scheduller).state ) == 1 );
         then:
           - switch.turn_on: mosquitto_relay       
     
    - if:
         condition:
            and:
             - lambda: |-
                return ( int( id(mosquitto_regulation).state)  == 2);
             - lambda: |-
                return ( int( id(mosquitto_scheduller).state ) == 0 );
         then:
           - switch.turn_off: mosquitto_relay   
     
     ```

reaper7, 2021-10-05:
This is probably the same problem like in my case https://github.com/esphome/issues/issues/2506
I see that @OttoWinter is preparing in-depth logging of the i2c bus, maybe this will explain the reasons

OttoWinter, 2021-10-05:
Please try

1. stable 2021.9.x version with `very_verbose` log level

```yaml
logger:
  level: very_verbose
```

2. dev with custom i2c and very_verbose level

```yaml
logger:
  level: very_verbose
external_components:
  - source: github://esphome/esphome@i2c-data-logging
    components: [i2c]
```

and report the results, thanks!

SeByDocKy, 2021-10-05:
1. stable 2021.9.x version with very_verbose log level
```INFO Reading configuration /config/esphome/potager.yaml...
INFO Detected timezone 'CET' with UTC offset 1 and daylight saving time from 27 March 02:00:00 to 30 October 03:00:00
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Deleting /data/potager/.pioenvs
INFO Deleting /data/potager/.piolibdeps
INFO Compiling app...
INFO Running:  platformio run -d /config/esphome/potager
Processing potager (board: esp32dev; framework: arduino; platform: platformio/espressif32@3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif32 3.10006.210326 (1.0.6) 
 - tool-esptoolpy 1.30000.201119 (3.0.0) 
 - toolchain-xtensa32 2.50200.97 (5.2.0)
Library Manager: Installing Hash
Library Manager: Already installed, built-in library
Dependency Graph
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <Wire> 1.0.1
|-- <noise-c> 0.1.1
|-- <Update> 1.0
|-- <WiFi> 1.0
Compiling /data/potager/.pioenvs/potager/src/esphome/components/api/api_connection.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/api/api_server.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/api/list_entities.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/api/proto.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/api/user_services.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/api/util.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/bh1750/bh1750.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/binary_sensor/automation.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/binary_sensor/filter.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/cover/cover.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/esp32_camera/esp32_camera.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/gpio/binary_sensor/gpio_binary_sensor.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/gpio/output/gpio_binary_output.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/gpio/switch/gpio_switch.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/homeassistant/time/homeassistant_time.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/i2c/i2c.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/light/addressable_light.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/light/automation.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/light/esp_color_correction.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/light/esp_hsv_color.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/light/esp_range_view.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/light/light_call.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/light/light_json_schema.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/light/light_output.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/light/light_state.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/logger/logger.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/ota/ota_component.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/output/automation.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/output/float_output.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/pulse_counter/pulse_counter_sensor.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/restart/restart_switch.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/sensor/automation.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/sensor/filter.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/sensor/sensor.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/socket/bsd_sockets_impl.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/socket/lwip_raw_tcp_impl.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/status/status_binary_sensor.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/switch/automation.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/switch/switch.cpp.o
Compiling /data/potager/.pioenvs/potager/src/esphome/components/template/sensor/template_sensor.cpp.o
[... 2906 lines cut ...]
[18:02:25][V][component:206]: Component pulse_counter.sensor took a long time for an operation (0.08 s).
[18:02:25][V][component:207]: Components should block for at most 20-30ms.
[18:02:26][VV][scheduler:152]: Running interval 'update' with interval=1000 last_execution=80196 (now=81197)
[... cut ...]
[18:02:27][D][pulse_counter:160]: 'water_flow_meter_potager': Retrieved counter: 0.00 pulses/min
[18:02:27][V][sensor:070]: 'water_flow_meter_potager': Received new state 0.000000
[18:02:27][VV][sensor.filter:013]: Filter(0x3ffb97c0)::input(0.000000)
[18:02:27][VV][sensor.filter:197]: LambdaFilter(0x3ffb97c0)::new_value(0.000000) -> 0.000000
[18:02:27][VV][sensor.filter:020]: Filter(0x3ffb97c0)::output(0.000000) -> SENSOR
[18:02:27][D][sensor:121]: 'water_flow_meter_potager': Sending state 0.00000 l with 2 decimals o

mmakaay, 2021-10-07:
Update on my testing: I have setup a circuit with an ESP-01 and a bme680, and I could get I2C to work using the latest dev.

SeByDocKy, 2021-10-07:
> Update on my testing: I have setup a circuit with an ESP-01 and a bme680, and I could get I2C to work using the latest dev.

Ok .... for the ES01m, my problem comes from a ADS115 where voltage reading are wrong (~ true voltage/4).

reaper7, 2021-10-07:
@SeByDocKy, @mmakaay - https://github.com/esphome/issues/issues/2506#issuecomment-936572626
in my opinion, everything what uses _write_byte_16_ (write_bytes_16) to set the configuration will not work properly,
because hi / lo bytes are in the wrong position (not swapped)

mmakaay, 2021-10-07:
Yeah, looking at the `write_byte_16()` method, that makes sense.
Work is being done in there to create a pretty reversed byte order, but then the original data is sent.
Ah, and that is also the conclusion from the thread you mentioned.
@OttoWinter : I assume you will pick up that fix?

martgras, 2021-10-07:
thx - should try to write code on the phone :)

reaper7, 2021-10-07:
checked, works for ina226

mmakaay, 2021-10-08:
Looking at the bh1750 component code, the read method looks good to me. Two bytes are read into a `uint16_t`, and the bytes are reversed to get to the correct host ordering. The method that was fixed (`write_bytes_16(...)`) has nothing to do with that bit of code, so it makes sense that the fix didn't result in a working sensor for you.

I don't have a bh1750, so I'm unable to debug the issue unfortunately.
I'll order one, so I can test things myself, but that will take a few days to arrive.

martgras, 2021-10-08:
Very verbose logging with all other components might help.
I also ordered a bh1750 today 🙂

mmakaay, 2021-10-08:
I submitted a PR for the invalid return value check. 
It's not likely that this fixes the issue, but it is at least a step in the right direction.

mmakaay, 2021-10-11:
That is a actually great idea!
It would definitely have triggered the sensor.