esphome/issues #2514: ERROR Error binary size: Unknown error from ESP

tungmeister, 2021-10-05:
### The problem

I test the dev channel quite frequently on of my esp8266 driven spotlights however after a recent update I'm now getting
```
ERROR Error binary size: Unknown error from ESP
```
at the end of the log when I try to update. The binary compiles fine however fails to flash. I'm also now seeing this if I try to update the bulb from the stable release. I also had this issue with a nodemcu with a recent update from the dev branch but thought it was the board acting up so flashed it directly via USB. Unfortunately that's not an option an would much rather not have to try and pull it apart to flash it (it came with tasmota from the factory)

### Which version of ESPHome has the issue?

2021.10.X Dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9

### What platform are you using?

ESP8266

OttoWinter, 2021-10-06:
Yes, almost certainly related to https://github.com/esphome/esphome/pull/2432

If you have captive portal/web server enabled, you can also upload an update via that. Otherwise no there is no way to apply the update if you used that dev commit before

tungmeister, 2021-10-06:
@mmakaay I was afraid you were going to say that  - is it that the firmware will fail no matter what is done from the server or that the server doesn't understand the response?

tungmeister, 2021-10-07:
No worries, thought that might be the case, I'll have a go at a manual flash on it if I can get it apart without breaking it, the peril's of flashing Dev builds!

closed 2021-10-07 by tungmeister