esphome/issues #2515: button/link to open web_server is missing on dashboard

poldim, 2021-10-05:
### The problem

There used to be a button, now it's gone.
@jesserockz confirmed he is seeing the same thing on his installation.

[image]

### Which version of ESPHome has the issue?

v2021.10.0-dev

### What type of installation are you using?

Docker

### What platform are you using?

ESP8266

### Component causing the issue

dashboard

### Additional information

@jesserockz mentioned it might have to do with #2374

closed 2021-10-05 by jesserockz

jesserockz, 2021-10-15:
Can you paste the output from `http://<dashboard>:6052/info?configuration=esphome_dining_room_switch.yaml` ?

poldim, 2021-10-15:
> Can you paste the output from `http://<dashboard>:6052/info?configuration=esphome_dining_room_switch.yaml` ?

So I first went to this link and got an error.  At this point it’s only showing the “update available” icon.  I then updated the firmware on that device.  When I came back to the dashboard, the “update available” icon was replaced with the web server icon.

I don’t think this was ever the case before.  After a new version was released, the not-yet-updated devices still displayed their web server link buttons.

[image]

jesserockz, 2021-10-17:
Both icons can be visible at the same time. ESPHome caches the used integrations (including `web_sever`), and the dashboard uses this cache to decide to show or not. The cache is updated when you install to your device.

poldim, 2021-10-17:
> The cache is updated when you install to your device.

Is the integration caching based on actual install to the device or just the compilation process?

> ESPHome caches the used integrations (including `web_sever`), and the dashboard uses this cache to decide to show or not.   

None of my 40+ devices were showing them simultaneously.  I went through and updated all of my devices. They now all show the webserver link correctly.  

> Both icons can be visible at the same time.

If needed I can confirm whether the next dev update rolls out whether both icons can be visible at the same time