esphome/issues #2519: Wifi breaks on m5stack-core-esp32 with esp32_ble_tracker enabled

brettp, 2021-10-06:
### The problem

After upgrading to 2021.9.2, all of my M5 Atom Lite devices have problems connecting to wifi with "Auth Expired" and "Association Expired" expired messages.

This persists for all board types I've tried, including m5stack-core-esp32 and node32s.

### Which version of ESPHome has the issue?

2021.9.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.10.0

### What platform are you using?

ESP32

### Board

m5stack-core-esp32

### Component causing the issue

esp32_ble_tracker, node32s

### Example YAML snippet

```yaml
esp32_ble_tracker:
#   scan_parameters:
#     interval: 320ms
#     window: 30ms

esphome:
  name: test-test-test
  platform: ESP32
  board: m5stack-core-esp32

wifi:
  networks:
  - ssid: "MySSID"
    password: "MyPassword"
  # # power_save_mode: "none"
  #   manual_ip:
  #     static_ip: 10.22.123.230
  #     gateway: 10.22.123.1
  #     subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test Hotspot"
    password: "Gq8EcQNp8gbZ"

captive_portal:

# Enable logging
logger:
```

### Anything in the logs that might be useful for us?

```txt
[16:27:32]\xff[I][logger:174]: Log initialized
[16:27:32][I][app:029]: Running through setup()...
[16:27:33][D][esp32_ble_tracker:180]: Starting scan...
[16:27:33][C][wifi:037]: Setting up WiFi...
[16:27:33][D][wifi:382]: Starting scan...
[16:27:33][D][esp32_ble_tracker:628]: Found device 5B:A8:5C:B9:36:C1 RSSI=-91
[16:27:33][D][esp32_ble_tracker:649]:   Address Type: RANDOM
[16:27:33][D][esp32_ble_tracker:653]:   TX Power: 2
[16:27:34][D][esp32_ble_tracker:628]: Found device 57:A9:CF:9E:8F:1D RSSI=-50
[16:27:34][D][esp32_ble_tracker:649]:   Address Type: RANDOM
[16:27:36][D][wifi:397]: Found networks:
[16:27:36][I][wifi:443]: - 'MySSID' (AB:CD:EF:C8:3C:6E) ▂▄▆█
[16:27:36][D][wifi:444]:     Channel: 1
[16:27:36][D][wifi:445]:     RSSI: -46 dB
[16:27:36][I][wifi:443]: - 'MySSID' (AB:CD:EF:C8:49:B5) ▂▄▆█
[16:27:36][D][wifi:444]:     Channel: 6
[16:27:36][D][wifi:445]:     RSSI: -62 dB
[16:27:36][I][wifi:443]: - 'MySSID' (AB:CD:EF:CF:31:DF) ▂▄▆█
[16:27:36][D][wifi:444]:     Channel: 6
[16:27:36][D][wifi:445]:     RSSI: -67 dB

[snip other wifi networks]]

[16:27:36][I][wifi:252]: WiFi Connecting to 'MySSID'...
[16:27:36][D][esp32_ble_tracker:628]: Found device 3C:D5:E5:9D:B1:A4 RSSI=-98
[16:27:36][D][esp32_ble_tracker:649]:   Address Type: RANDOM
[16:27:37][D][esp32_ble_tracker:628]: Found device 6A:69:95:C5:42:25 RSSI=-85
[16:27:37][D][esp32_ble_tracker:649]:   Address Type: RANDOM
[16:27:40][D][esp32_ble_tracker:628]: Found device 6D:0B:1A:80:10:81 RSSI=-88
[16:27:40][D][esp32_ble_tracker:649]:   Address Type: RANDOM
[16:27:40][D][esp32_ble_tracker:653]:   TX Power: 2
[16:27:44][D][esp32_ble_tracker:628]: Found device FF:FF:80:01:D4:BE RSSI=-91
[16:27:44][D][esp32_ble_tracker:649]:   Address Type: PUBLIC
[16:27:44][D][esp32_ble_tracker:651]:   Name: 'Device'
[16:27:45][D][esp32_ble_tracker:628]: Found device 6E:C8:B5:18:05:8C RSSI=-35
[16:27:45][D][esp32_ble_tracker:649]:   Address Type: RANDOM
[16:27:47][D][esp32_ble_tracker:628]: Found device 78:AE:7D:93:83:11 RSSI=-61
[16:27:47][D][esp32_ble_tracker:649]:   Address Type: RANDOM
[16:27:47][W][wifi_esp32:447]: Event: Disconnected ssid='MySSID' bssid=AB:CD:EF:C8:3C:6E reason='Handshake Failed'
[16:27:47][W][wifi:549]: Error while connecting to network.
[16:27:47][W][wifi:587]: Restarting WiFi adapter...
[16:27:47][I][wifi:252]: WiFi Connecting to 'MySSID'...
[16:27:47][W][wifi:573]: WiFi Unknown connection status 255
[16:27:47][W][wifi:573]: WiFi Unknown connection status 255
[16:27:47][W][wifi:573]: WiFi Unknown connection status 255
[16:27:48][W][wifi_esp32:447]: Event: Disconnected ssid='MySSID' bssid=AB:CD:EF:C8:3C:6E reason='Auth Expired'
[16:27:48][W][wifi:549]: Error while connecting to network.
[16:27:48][W][wifi:587]: Restarting WiFi adapter...
```

### Additional information

As indicated by the commented out lines, I've tried changing the BLE scan parameters, setting the power save mode, and specifying a static IP (both in the config and on my router) with no changes.

OttoWinter, 2021-10-07:
1. What version did you upgrade from?
 2. Try removing the device from power for a minute or so, then try again (with the original config), also try that with the router (I've seen this error before spuriously appearing with certain routers)

brettp, 2021-12-12:
I was able to resolve this on my Atom Lite by flashing an updated (but unofficial) CH552 firmware, as outlined here by @sohtamei: https://github.com/espressif/arduino-esp32/issues/2144#issuecomment-976084652

This requires creating a custom USB cable to put the CH552 into download mode, and seems to unfortunately break the ability to automatically switch the ESP32 into bootloader mode (meaning on the Atom Lite you have to push the side button twice, about 1 second apart when trying to flash), but I've had solid Wifi and BLE performance since.

NB: When flashing with WCHISPTool, be sure to keep the default pin for download mode as 3.6. The alternative pin (1.5) is not exposed via the cable on the Atom Lite and is a bit of a pain to use jumpers to pull to ground manually when powering on.

Davst, 2023-08-25:
I'm having this same issue... I've read other forums suggesting moving to PlatformIO IDE but that hardly is possible when using esphome.. .any idea if this is fixable without creating custom cables etc.?

joe-sydney, 2023-08-25:
Ever since I changed my yaml to the following, I've not had any issues:

esp32:
  board: m5stack-atom #or whatever you're using
  framework:
    type: esp-idf #arduino