esphome/issues #2522: m5stack ENVII sensor SHT30 not working with SHT3xd component

lhoracek, 2021-10-07:
### The problem

Trying to get values from m5stack EnvII sensor that contains bmp280 and sht30 sensors. Having i2c set up with scan enabled. It can see device with address 0x44. But when sht3xd tries to initialize, it marks itself as failed.

Tested it with this Arduino snippet and then the sensor provides values fine:
```
#include "Wire.h"
#include "SHT31.h"

uint32_t start;
uint32_t stop;

SHT31 sht;

void setup()
{
  Serial.begin(115200);
  Wire.begin();

  sht.begin(0x44);    //Sensor I2C Address

  Wire.setClock(100000);
  uint16_t stat = sht.readStatus();
  Serial.print(stat, HEX);
  Serial.println();
}

void loop()
{
  sht.read();

  Serial.print("Temperature:");
  Serial.print(sht.getTemperature(), 1);
  Serial.print("\t");
  Serial.print("Humidity:");
  Serial.println(sht.getHumidity(), 1);
  delay(50);
}
```

### Which version of ESPHome has the issue?

2021.10.0-dev

### What type of installation are you using?

pip

### What platform are you using?

ESP32

### Board

m5stack core

### Component causing the issue

sensor.sht3xd

### Example YAML snippet

```yaml
esphome:
  name: heart-rate-display
  platform: ESP32
  board: m5stack-core-esp32
  
# Enable logging
logger:
[... 137 lines cut ...]
[11:38:32][C][sgp30:228]: SGP30:
[11:38:32][C][sgp30:229]:   Address: 0x58
[11:38:32][C][sgp30:249]:   Serial number: 23847698
[11:38:32][C][sgp30:255]:   Baseline: No baseline configured
[11:38:32][C][sgp30:257]:   Warm up time: 43200s
[11:38:32][C][sgp30:259]:   Update Interval: 1.0s
[11:38:32][C][sgp30:260]:   eCO2 sensor 'Workshop eCO2'
[11:38:32][C][sgp30:260]:     Device Class: 'carbon_dioxide'
[11:38:32][C][sgp30:260]:     State Class: 'measurement'
[11:38:32][C][sgp30:260]:     Unit of Measurement: 'ppm'
[11:38:32][C][sgp30:260]:     Accuracy Decimals: 1
[11:38:32][C][sgp30:260]:     Icon: 'mdi:molecule-co2'
[11:38:32][C][sgp30:261]:   TVOC sensor 'Workshop TVOC'
[11:38:32][C][sgp30:261]:     Device Class: 'volatile_organic_compounds'
[11:38:32][C][sgp30:261]:     State Class: 'measurement'
[11:38:32][C][sgp30:261]:     Unit of Measurement: 'ppb'
[11:38:32][C][sgp30:261]:     Accuracy Decimals: 1
[11:38:32][C][sgp30:261]:     Icon: 'mdi:radiator'
[11:38:32][C][sgp30:264]: Store baseline: YES
[11:38:32][C][sgp30:270]:   Compensation: No source configured
```

### Additional information

Tried stable, beta and dev releases

anatoly-savchenkov, 2021-11-06:
I have somewhat similar problem. sht31 + sgp30 + barometric pressure + luminance sensor all attached on the single I2C bus don't work with the same symptoms. My suspect is SGP30, although error is reported by sht3x like in your case.

**I was able to solve this by moving sgp30 to a separate I2C bus, see example below. I'm curious if this would solve problems on your side.** 

sgp30 does not work for me even if attached to a dedicated I2C bus (reports very inaccurate numbers). But that's a different story which I'll try to solve once time allows.

```
i2c:
 - id: bus_a
   sda: 13
   scl: 15
   scan: true
 - id: bus_b
   sda: 21
   scl: 22
   scan: true
sensor:
  - platform: sht3xd
    temperature:
      name: Temperature
      id: temperature
      device_class: temperature
      state_class: measurement
      unit_of_measurement: "°C"
      accuracy_decimals: 1
    humidity:
      name: Humidity
      id: humidity
      device_class: humidity
      state_class: measurement
      unit_of_measurement: "%"
      accuracy_decimals: 1
    address: 0x44
    update_interval: 1s
    i2c_id: bus_a

  - platform: max44009
    name: Luminance
    id: luminance
    update_interval: 1s
    i2c_id: bus_a

  - platform: ms5611
    temperature:
      name: Temperature
      id: ms5611_temperature
      unit_of_measurement: "°C"
      device_class: "temperature"
      state_class: "measurement"
      accuracy_decimals: 1      
    pressure:
      name: Pressure
      id: pressure
      filters:
        - multiply: 0.75006157584566
      device_class: pressure
      state_class: measurement
      accuracy_decimals: 1
      unit_of_measurement: "mmHg"
    address: 0x77
    update_interval: 1s
    i2c_id: bus_a

  - platform: sgp30
    eco2:
      name: eCO2
      accuracy_decimals: 1
      id: office_eco2
    tvoc:
      name: TVOC
      accuracy_decimals: 1
      id: office_tvoc
    store_baseline: yes
    address: 0x58
    update_interval: 1s
    i2c_id: bus_b
    baseline:
      eco2_baseline: 0x90E8
      tvoc_baseline: 0x942C
    compensation:
      temperature_source: temperature
      humidity_source: humidity
```

lhoracek, 2022-01-11:
Even just having the ENVII sensor with BMP280 and SHT30 alone connected, I get the error:
```
22:06:34][C][bmp280.sensor:056]: Setting up BMP280...
[22:06:34][C][sht3xd:019]: Setting up SHT3xD...
[22:06:34][E][component:112]: Component sht3xd.sensor was marked as failed.
```
And BMP280 continues to report values correctly. Not able to split the sensors in the m5stack module :-/

randybb, 2022-01-11:
no issues with ENVIII (SHT30 + QMP6988, which is not supported) and BME280:
```
[22:39:17][D][sensor:125]: 'DEV 1 BME280 Temperature': Sending state 24.58000 °C with 1 decimals of accuracy
[22:39:17][D][sensor:125]: 'DEV 1 BME280 Pressure': Sending state 1018.23511 hPa with 1 decimals of accuracy
[22:39:17][D][sensor:125]: 'DEV 1 BME280 Humidity': Sending state 30.15820 % with 1 decimals of accuracy
[22:39:17][D][sht3xd:065]: Got temperature=24.91°C humidity=29.57%
[22:39:17][D][sensor:125]: 'DEV 1 SHT30 Temperature': Sending state 24.90654 °C with 1 decimals of accuracy
[22:39:17][D][sensor:125]: 'DEV 1 SHT30 Humidity': Sending state 29.57199 % with 1 decimals of accuracy
```

lhoracek, 2022-01-14:
I'm getting similar issues with their RFID2 module, that shoudl be compatible with `rc522_i2c`. Starting to suspect my unit. Will get another one and retest. Thank @randybb