esphome/issues #2532: HA forgets color temperature of RGBWW light after power cycle

bdr99, 2021-10-09:
### The problem

I have an RGBWW bulb with ESPHome connected to HA. When I power cycle the bulb, after it reconnects to HA, HA has forgotten the color temperature and defaults to the coldest color temperature.

Steps to reproduce:
1. Change color temperature to warm in HA.
2. Turn off power to the bulb.
3. Wait for bulb to show as "Unavailable" in HA.
4. Restore power to the bulb. Bulb turns on with the same color temperature setting as before.
5. Wait for bulb to reconnect to HA.

Expected result: Color temperature shown in HA should match the actual color temperature of the bulb.
Actual result: HA shows the coldest color temperature even though the bulb is still a warm color temperature.

[image]

### Which version of ESPHome has the issue?

2021.9.3

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.10.1

### What platform are you using?

ESP8266

### Board

N/A

### Component causing the issue

light

### Additional information

<details>
  <summary>YAML config (click to expand)</summary>

```yaml
esphome:
  name: bulb3
  platform: ESP8266
  board: esp8285
  esp8266_restore_from_flash: true

wifi:
  networks:
  - ssid: !secret wifi_ssid
    password: !secret wifi_password
  manual_ip:
    static_ip: 192.168.1.202
    gateway: 192.168.1.1
    subnet: 255.255.255.0
  ap:
    ssid: Brandon's Room Lamp
    password: !secret ap_wifi_password

captive_portal:

api:

ota:

logger:
  baud_rate: 0
  level: WARN

light:
  - platform: rgbww
    name: Brandon's Room Lamp
    id: bulb3_light
    red: output_red
    green: output_green
    blue: output_blue
    cold_white: output_cold
    warm_white: output_warm
    cold_white_color_temperature: 6536 K
    warm_white_color_temperature: 2000 K
    color_interlock: true

output:
  - platform: esp8266_pwm
    id: output_red
    pin: GPIO4
  - platform: esp8266_pwm
    id: output_green
    pin: GPIO12
  - platform: esp8266_pwm
    id: output_blue
    pin: GPIO14
  - platform: esp8266_pwm
    id: output_cold
    pin: GPIO5
  - platform: esp8266_pwm
    id: output_warm
    pin: GPIO13
```
</details>

bkaufx, 2021-10-09:
This is an old problem that was supposed to have been fixed.  I guess maybe it wasn't.  https://github.com/esphome/esphome/pull/2150   https://github.com/esphome/issues/issues/2304 https://github.com/esphome/issues/issues/2314

closed 2021-10-11 by oxan

sairon, 2021-10-12:
@bdr99 The scenario you describe in the ticket should be fixed, as it fixed the bug that bothered me in my setup as well :) But it could get out of sync though if you change individual CW/WW channels by other means than HA color temperature slider, just like Oxan mentioned in the PR, so that deserves a more sophisticated fix.