esphome/issues #2536: esphome/esphome docker image only runs in privileged mode

chof747, 2021-10-10:
### The problem

I have a docker-compose configuration running esphome dashboard, but since the latest update to the image of the following commit: 42739f0 it can only run in privileged mode or I get the following error:
```
esphome          | Fatal Python error: init_interp_main: can't initialize time
esphome          | Python runtime state: core initialized
esphome          | PermissionError: [Errno 1] Operation not permitted
esphome          | 
esphome          | Current thread 0xb6f2b520 (most recent call first):
esphome          | <no Python frame>
```

I also tried to remove all volume mounts and it does not change anything. Adding `privileged : true` solves the problem but this is not a secure option. Please advise how I can fix this. Thx Christian

### Which version of ESPHome has the issue?

commit 42739f0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.10.2

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
docker-compose.yml:

  esphome:
    container_name: esphome
    image: esphome/esphome
    privileged: true
    network_mode: host
    environment:
      - TZ=${TZ}
      - PUID=${PUID}
      - PGID=${PGID}
    restart: unless-stopped
    volumes:
      - ${LOCALTIME}:/etc/localtime:ro
      - ${USERDIR}/esphome:/config${VOLUME_OPTIONS}
```

### Anything in the logs that might be useful for us?

```txt
esphome          | Fatal Python error: init_interp_main: can't initialize time
esphome          | Python runtime state: core initialized
esphome          | PermissionError: [Errno 1] Operation not permitted
esphome          | 
esphome          | Current thread 0xb6f2b520 (most recent call first):
esphome          | <no Python frame>
```

OttoWinter, 2021-10-11:
See also here https://community.home-assistant.io/t/migration-to-2021-7-fails-fatal-python-error-init-interp-main-cant-initialize-time/320648/9

It appears to be a bug in the host's libseccomp2 library

chof747, 2021-10-13:
Thanks installing the library fixed the issue.

closed 2021-10-13 by chof747