esphome/issues #2537: Use of delayed_on_off filter in Binary Sensor Component breaks HA sensor updates

DDomnick, 2021-10-10:
### The problem

Given code example, on program upload / reset, sensor in Home Assistant does not follow pin state nor change value at all.
Automations in ESPHome continue to function however.  I.E. "switch.turn_off: dev_genstartstopssr_plb150" would fire after 10 second delay.

### Which version of ESPHome has the issue?

2021.9.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.10.1

### What platform are you using?

ESP32

### Board

Adafruit Feather ESP32

### Component causing the issue

Binary Sensor Filters

### Example YAML snippet

```yaml
- platform: gpio
    pin: GPIO36
    name: "Dev-(+3v3_U)"
    id: util_power
    filters:
      - delayed_on_off: 10s
    on_release:
      then:
        - switch.turn_off: dev_genstartstopssr_plb150
```

Drealine, 2022-06-05:
Same issue here.

jazzmonger, 2022-12-22:
has this issue been fixed?  Im seeing the same thing...