esphome/issues #2544: single core esp32 bsec

tarbax, 2021-10-13:
### The problem

bsec not working on the single core esp32 version
there should be fixable from what i found:

https://community.bosch-sensortec.com/t5/MEMS-sensors-forum/Linking-BSEC-library-to-ESP-IDF-for-BME680/td-p/9741/page/2

### Which version of ESPHome has the issue?

2021.9 dev

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32-IDF

### Component causing the issue

bme680_bsec

OttoWinter, 2021-10-17:
Can you please include more information as to what specifically is not working?

tarbax, 2021-10-17:
> Can you please include more information as to what specifically is not working?

when compiling i get this error:
`2021-10-17 13:28:33,515 INFO Reading configuration /config/esphome/snos-single-core-bsec-test.yaml...
2021-10-17 13:28:33,582 INFO Generating C++ source...
2021-10-17 13:28:33,629 INFO Core config or version changed, cleaning build files...
2021-10-17 13:28:33,688 INFO Compiling app...
2021-10-17 13:28:33,834 INFO Running:  platformio run -d /config/esphome/snos-single-core-bsec-test
Processing snos-single-core-bsec-test (board: esp32doit-devkit-v1; framework: espidf; platform: espressif32 @ 3.3.2)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES: 
 - framework-espidf 3.40300.0 (4.3.0) 
 - tool-cmake 3.16.4 
 - tool-esptoolpy 1.30100.210531 (3.1.0) 
 - tool-ninja 1.7.1 
 - toolchain-esp32ulp 1.22851.191205 (2.28.51) 
 - toolchain-xtensa32 2.80400.210211 (8.4.0)
Reading CMake configuration...
Library Manager: Installing BSEC Software Library @ 1.6.1480

Unpacking  [------------------------------------]    0%
Unpacking  [------------------------------------]    2%
Unpacking  [#-----------------------------------]    5%
Unpacking  [##----------------------------------]    8%
Unpacking  [###---------------------------------]   10%
Unpacking  [####--------------------------------]   13%
Unpacking  [#####-------------------------------]   16%
Unpacking  [######------------------------------]   18%
Unpacking  [#######-----------------------------]   21%
Unpacking  [########----------------------------]   24%
Unpacking  [#########---------------------------]   27%
Unpacking  [##########--------------------------]   29%
Unpacking  [###########-------------------------]   32%
Unpacking  [############------------------------]   35%
Unpacking  [#############-----------------------]   37%
Unpacking  [##############----------------------]   40%
Unpacking  [###############---------------------]   43%
Unpacking  [################--------------------]   45%
Unpacking  [#################-------------------]   48%
Unpacking  [##################------------------]   51%
Unpacking  [###################-----------------]   54%
Unpacking  [####################----------------]   56%
Unpacking  [#####################---------------]   59%
Unpacking  [######################--------------]   62%
Unpacking  [#######################-------------]   64%
Unpacking  [########################------------]   67%
Unpacking  [#########################-----------]   70%
Unpacking  [##########################----------]   72%
Unpacking  [###########################---------]   75%
Unpacking  [############################--------]   78%
Unpacking  [#############################-------]   81%
Unpacking  [##############################------]   83%
Unpacking  [###############################-----]   86%
Unpacking  [################################----]   89%
Unpacking  [#################################---]   91%
Unpacking  [##################################--]   94%
Unpacking  [###################################-]   97%
Unpacking  [####################################]  100%
Library Manager: BSEC Software Library @ 1.6.1480 has been installed!
Generating assembly for certificate bundle...
LDF Modes: Finder ~ off, Compatibility ~ soft
No dependencies
Compiling /data/snos-single-core-bsec-test/.pioenvs/snos-single-core-bsec-test/src/esphome/components/api/api_connection.o
Compiling /data/snos-single-core-bsec-test/.pioenvs/snos-single-core-bsec-test/src/esphome/components/api/api_frame_helper.o
Compiling /data/snos-single-core-bsec-test/.pioenvs/snos-single-core-bsec-test/src/esphome/components/api/api_pb2.o
Compiling /data/snos-single-core-bsec-test/.pioenvs/snos-single-core-bsec-test/src/esphome/components/api/api_pb2_service.o
Compiling /data/snos-single-core-bsec-test/.pioenvs/snos-single-core-bsec-test/src/esphome/components/api/api_server.o
Compiling /data/snos-single-core-bsec-test/.pioenvs/snos-single-core-bsec-test/src/esphome/components/api/list_entities.o
Compiling /data/snos-single-core-bsec-test/.pioenvs/snos-single-core-bsec-test/src/esphome/components/api/proto.o
Compiling /data/snos-single-core-bsec-test/.pioenvs/snos-single-core-bsec-test/src/esphome/components/api/subscribe_state.o
Compiling /data/snos-single-core-bsec-test/.pioenvs/snos-single-core-bsec-test/src/esphome/components/api/user_services.o
Compiling /data/snos-single-core-bsec-test/.pioenvs/snos-single-core-bsec-test/src/esphome/components/api/util.o
Compiling /data/snos-single-core-bsec-test/.pioenvs/snos-single-core-bsec-test/src/esphome/components/bh1750/bh1750.o
Compiling /data/snos-single-core-bsec-test/.pioenvs/snos-single-core-bsec-test/src/esphome/components/binary_sensor/automation.o
Compiling /data/snos-single-core-bsec-test/.pioenvs/snos-single-core-bsec-test/src/esphome/components/binary_sensor/binary_sensor.o
Compiling /data/snos-single-core-bsec-test/.pioenvs/snos-single-core-bsec-test/src/esphome/components/binary_sensor/filter.o
Compiling /data/snos-single-core-bsec-test/.pioenvs/snos-single-core-bsec-test/src/esphome/components/bme680_bsec/bme680_bsec.o
Compiling /data/snos-single-core-bsec-test/.pioenvs/snos-single-core-bsec-test/src/esphome/components/deep_sleep/deep_sleep_component.o
In file included from src/esphome/components/bme680_bsec/bme680_bsec.cpp:1:
[... cut ...]
* Looking for bsec.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:bsec.h"
* Web  > https://platformio.org/lib/search?query=header:bsec.h
*
**************************************************************

 #include <bsec.h>
          ^~~~~~~~
compilation terminated.
*** [/data/snos-single-core-bsec-test/.pioenvs/snos-single-core-bsec-test/src/esphome/components/bme680_bsec/bme680_bsec.o] Error 1
========================= [FAILED] Took 14.44 seconds =========================`

tarbax, 2021-10-17:
my config:

```
esphome:
  name: snos-single-core-bsec-test

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y

# Enable logging
logger:

deep_sleep:
  run_duration: 3min
  sleep_duration: 20min
  wakeup_pin_mode: KEEP_AWAKE
  esp32_ext1_wakeup:
    pins: GPIO25
    mode: ANY_HIGH
ota:
  port: 3232

i2c:
  sda: 21
  scl: 22
  scan: true
  id: bus_a

output:
  - platform: ledc
    pin: GPIO27
    id: rtttl_out
  - platform: ledc
    pin: GPIO5
    id: redgpio
  - platform: ledc
    pin: GPIO19
    id: greengpio
  - platform: ledc
    pin: GPIO2
    id: bluegpio

rtttl:
  output: rtttl_out

bme680_bsec:
    # i2c address
    # -----------
    # Common values are:
    # - 0x76
    # - 0x77
    # Default: 0x76
    address: 0x76

    # Temperature offset
    # ------------------
    # Useful if device is in enclosure and reads too high
    # Default: 0
    temperature_offset: 3.9

    # IAQ calculation mode
    # --------------------
    # Available options:
    # - static (for fixed position devices)
    # - mobile (for on person or other moveable devices)
    # Default: static
    iaq_mode: static

    # Sample rate
    # -----------
    # Available options:
    # - lp (low power - samples every 3 seconds)
    # - ulp (ultra low power - samples every 5 minutes)
    # Default: lp
    sample_rate: ulp

    # Interval at which to save BSEC state
    # ------------------------------------
    # Default: 6h
    state_save_interval: 6h

sensor:
  - platform: bme680_bsec
    temperature:
      name: "BME680 Temperature"
    pressure:
      name: "BME680 Pressure"
    humidity:
      name: "BME680 Humidity"
    iaq:
      name: "BME680 IAQ"
      id: iaq_index
    co2_equivalent:
      name: "BME680 CO2 Equivalent"
    breath_voc_equivalent:
      name: "BME680 Breath VOC Equivalent"
      
  - platform: bh1750
    name: "BH1750 Illuminance"
    address: 0x5C
    measurement_duration: 69
    update_interval: 500ms
```