esphome/issues #2547: DSMR: sensor for actual tariff is missing

dhover, 2021-10-14:
### The problem

The DSMR telegram contains a line with the actual tariff.
But the sensor for the actual tariff is missing.

### Which version of ESPHome has the issue?

2021.9.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.10.4

### What platform are you using?

ESP8266

### Board

slimmelezer

### Component causing the issue

dsmr

### Additional information

In the DSMR 2 telegram the line with the actual tariff is:
0-0:96.14.0(0002)
0001 = off-peak
0002 = peak

StijnVdd, 2021-11-17:
Taking a look a the library that is used, the electricity_tariff is already exposed (https://platformio.org/lib/show/12591/Dsmr)
So I believe this "electricity_tariff" just needs to be added to the config schema in https://github.com/esphome/esphome/blob/dev/esphome/components/dsmr/sensor.py

mmakaay, 2021-11-18:
That makes sense @StijnVdd 
Are you able to provide a PR for that (+ documentaiton)?
If not, let me know, and I'll add it to my dsmr TODO list.

closed 2021-11-18 by oxan

dhover, 2021-11-26:
Today I've installed the new esphome 2021.11.2.
The Tariff sensor showed up!. Very nice.
Also the debugging is a very nice feature.

This is the result:

`[17:29:21][D][text_sensor:067]: 'Slimmelezer DSMR Identification': Sending state 'KMP5 ZABF001770315713'
[17:29:21][D][text_sensor:067]: 'Slimmelezer Gas Consumed Text': Sending state '(211126170000)(00)(60)(1)(0-1:24.2.1)(m3)(18852.601)'
[17:29:31][D][uart_debug:152]: <<< "/KMP5 ZABF001770315713\r\n"
[17:29:31][D][uart_debug:152]: <<< "\r\n"
[17:29:31][D][uart_debug:152]: <<< "0-0:96.1.1(205A414246303031373730333135373133)\r\n"
[17:29:31][D][uart_debug:152]: <<< "1-0:1.8.1(24551.800*kWh)\r\n"
[17:29:31][D][uart_debug:152]: <<< "1-0:1.8.2(28169.620*kWh)\r\n"
[17:29:31][D][uart_debug:152]: <<< "1-0:2.8.1(01361.491*kWh)\r\n"
[17:29:31][D][uart_debug:152]: <<< "1-0:2.8.2(03412.064*kWh)\r\n"
[17:29:31][D][uart_debug:152]: <<< "0-0:96.14.0(0002)\r\n"
[17:29:31][D][uart_debug:152]: <<< "1-0:1.7.0(0002.26*kW)\r\n"
[17:29:31][D][uart_debug:152]: <<< "1-0:2.7.0(0000.00*kW)\r\n"
[17:29:31][D][uart_debug:152]: <<< "0-0:96.13.1()\r\n"
[17:29:31][D][uart_debug:152]: <<< "0-0:96.13.0()\r\n"
[17:29:31][D][uart_debug:152]: <<< "0-1:24.1.0(3)\r\n"
[17:29:31][D][uart_debug:152]: <<< "0-1:96.1.0(4730303135353631303133373938323133)\r\n"
[17:29:31][D][uart_debug:152]: <<< "0-1:24.3.0(211126170000)(00)(60)(1)(0-1:24.2.1)(m3)\r\n"
[17:29:31][D][uart_debug:152]: <<< "(18852.601)\r\n"
[17:29:31][D][uart_debug:152]: <<< "!\r\n"
[17:29:31][D][sensor:113]: 'Slimmelezer Energy Consumed Tariff 1': Sending state 24551.80078 kWh with 3 decimals of accuracy
[17:29:31][D][sensor:113]: 'Slimmelezer Energy Consumed Tariff 2': Sending state 28169.61914 kWh with 3 decimals of accuracy
[17:29:31][D][sensor:113]: 'Slimmelezer Energy Produced Tariff 1': Sending state 1361.49097 kWh with 3 decimals of accuracy
[17:29:31][D][sensor:113]: 'Slimmelezer Energy Produced Tariff 2': Sending state 3412.06396 kWh with 3 decimals of accuracy
[17:29:31][D][sensor:113]: 'Slimmelezer Power Consumed': Sending state 2260.00000 W with 0 decimals of accuracy
[17:29:31][D][sensor:113]: 'Slimmelezer Power Produced': Sending state 0.00000 W with 0 decimals of accuracy
[17:29:31][D][text_sensor:067]: 'Slimmelezer DSMR Identification': Sending state 'KMP5 ZABF001770315713'
[17:29:31][D][text_sensor:067]: 'Slimmelezer Gas Consumed Text': Sending state '(211126170000)(00)(60)(1)(0-1:24.2.1)(m3)(18852.601)'`

One sensor which is still missing is the Gas Consumed.

mmakaay, 2021-11-26:
For Gas Consumed, integration of the DSMR parser into ESPHome is a requirement I think, but that has proven a bit hard to do on short term. It's on the radar though!
Thanks for reporting back.