esphome/issues #2553: NodeMCUv2 + sensor dht (SI7021) problem since last uodate

jcg747, 2021-10-16:
### The problem

Hello,
A while ago, I have built four temp/hum sensor based on a nodeMCU v2 and a Sonoff SI7021 ( with a compatible display) 

Everything was working nicely both on the local display and under HomeAssistant. until the latest ESPhome 2021.9.3 update. Since then, 3 on the 4 have stopped working properly ( no more temp/hum) with the following partial log :

[17:03:12][C][captive_portal:148]: Captive Portal:
[17:03:12][C][ota:029]: Over-The-Air Updates:
[17:03:12][C][ota:030]:   Address: 192.168.10.64:8266
[17:03:12][C][ota:032]:   Using Password.
[17:03:12][C][api:135]: API Server:
[17:03:12][C][api:136]:   Address: 192.168.10.64:6053
[17:03:12][C][api:140]:   Using noise encryption: NO

**[17:03:22][W][dht:167]: Falling edge for bit 40 failed!
[17:03:22][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number).**

[17:03:22][D][sensor:121]: 'SenseNode1 Temp': Sending state nan °C with 1 decimals of accuracy
[17:03:22][D][sensor:121]: 'SenseNode1 Humidity': Sending state nan % with 0 decimals of accuracy

Curiously all 4 are absolutely identical (I tripple checked)

I'm a newcomer on Domotic so I apologize in advance if it look's obvious for more experienced user.
I'm also new to Github, I would appreciate any help to get it sorted
Best regards 
Jean Claude

### Which version of ESPHome has the issue?

20219.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Core 2021.10.5 / Supervisor 2021.10.0

### What platform are you using?

ESP32

### Board

NodeMCU v2

### Component causing the issue

Sonof SI7021

### Example YAML snippet

```yaml
name: sensor_node_1
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: "JCG_2G"
  password: "Boeing747-400a"
  # Optional manual IP
  manual_ip:
    static_ip: 192.168.10.64
    gateway: 192.168.10.1
    subnet: 255.255.255.0
    dns1: 192.168.10.1
    dns2: 192.168.10.1

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Sensor Node 1 Fallback Hotspot"
    password: "xalXOJV4SB2T"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "Boeing747-400"

ota:
  password: "Boeing747-400"

time:
  - platform: homeassistant
    id: esptime
    
sensor:
  - platform: dht
    pin: D4
    model: SI7021
    temperature:
      name: "SenseNode1 Temp"
      id: SenseNode1_Temp 
    humidity:
      name: "SenseNode1 Humidity"
      id: SenseNode1_Hum
    update_interval: 30s

font:
  - file: 'OpenSans-Regular.ttf'
    id: font1
    size: 12
  - file: 'arial.ttf'
    id: font2
    size: 13
    
i2c:
  sda: D2
  scl: D1
  scan: False
```

srnoth, 2021-10-21:
Same. Goes away went reverting to previous version.

srnoth, 2021-10-31:
@ottuzzi good catch. I was able too resolve this on an ESP8266 board by specifying the latest framework (instead of recommended):

`esp8266:
  board: d1_mini
  restore_from_flash: true
  framework:
    version: latest`

gregoinc, 2021-10-31:
Hi @ottuzzi Piero, 

Thanks for the tip... That sounds interesting, and might be the way forward. I tried what you had above but it doesn't work for me? Here's what I have now that is working...

`esphome:
  name: esp1-studio
  platform: ESP8266
  board: nodemcuv2`

If I add...

`framework:
    version: latest`

It complains with an error 'framework is not a valid option for esphome' so not sure why that is happening?

ssieb, 2021-10-31:
You need to look closer at what he wrote.  It's not `esphome:`, it's `esp8266:`.

gregoinc, 2021-10-31:
> You need to look closer at what he wrote. It's not `esphome:`, it's `esp8266:`.

Hi @ssieb I did look closely... and when I changed 'esphome' to 'esp8266' I received a different error. Yes, I am not an expert on this, and it is likely I may not be across all the nuances of what is required. Perhaps you could clarify?

ssieb, 2021-10-31:
Device configuration has changed significantly with the last release.  See the docs at https://esphome.io/components/esphome.html and https://esphome.io/components/esp8266.html for where each parameter goes now.

gregoinc, 2021-10-31:
> Device configuration has changed significantly with the last release. See the docs at https://esphome.io/components/esphome.html and https://esphome.io/components/esp8266.html for where each parameter goes now.

Hi again... so from reading the documentation... I understand correctly, it sounds like there are now two areas within the esphome configuration file? One like below...

esp8266:
  board: nodemcuv2
  framework:
    version: recommended

And then this...

esphome:
  name: esp1-studio
  platform: ESP8266
  board: nodemcuv2

Or am I missing something?

ssieb, 2021-10-31:
Close, but you are missing something.  See the section about "Platform options that have been moved".  `platform:` has to go in the `esp8266:` section.

gregoinc, 2021-10-31:
> Close, but you are missing something. See the section about "Platform options that have been moved". `platform:` has to go in the `esp8266:` section.

After some more reading and changes it appears to be happy with...

esp8266:
 board: nodemcuv2
 framework:
  version: recommended

esphome:
  name: esp1-studio

gregoinc, 2021-10-31:
An update... changing from **recommended** to **latest** didn't resolve the issue I was trying to solve... still seeing this in the logs:-

[10:43:42][W][dht:167]: Requesting data from DHT failed!
[10:43:42][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number).

So I guess I wait to see what happens with future versions.

samuelmcconnell, 2021-11-01:
I noticed that I am now facing this issue on a Sonoff TH16 with the AM2302 one-wire sensor (configured as SI7021 on GPIO14). Was working previously, but after updating to 2021.10.3, it fails with the following error:

```
[08:48:28][W][dht:167]: Requesting data from DHT failed!
[08:48:28][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
[08:48:28][D][sensor:112]: 'Gecko Tank Temperature': Sending state nan °C with 1 decimals of accuracy
[08:48:28][D][sensor:112]: 'Gecko Tank Humidity': Sending state nan % with 0 decimals of accuracy

```
I updated the Arduino platform declaration as srnoth did, but it did not seem to help. Device config available [here](https://gist.github.com/samuelmcconnell/38c6bd26cd1116abebe8127581b227dd).

gregoinc, 2021-11-02:
> I noticed that I am now facing this issue on a Sonoff TH16 with the AM2302 one-wire sensor (configured as SI7021 on GPIO14). Was working previously, but after updating to 2021.10.3, it fails with [...]

I have attempted a number of fixes for the DHT issue... but none have resolved it. I have also not taken the step of rolling HA back to an earlier version. I am keen to move forward, however there are a few challenges with my setup (DHT, ZHA, etc), so may have no choice but to roll back and see what happens.

srnoth, 2021-11-02:
For the folks with TH16s... might be worth checking if the plug for the sensor is in all the way. I also had a TH16 throw errors, and went down this rabbit hole thinking it was the update... but it was actually coincidence that I only noticed the errors when I did the update.

The plugs seem to pop out very easily and it only takes being out by a mm for the sensor to lose connection.

ChrisPotDev, 2021-12-01:
I have the same issue with a Sonoff SI7021 and a d1_min, sadly no backup to restore.

ESPHome version: 2021.11.4

`esphome:
  name: ${devicename}_${device_type}

esp8266:
  board: d1_mini
  restore_from_flash: true
  framework:
    version: latest`
[22:20:34][D][sensor:112]: 'Main Bedroom Bathroom Temperature': Sending state 25.50000 °C with 2 decimals of accuracy
[22:20:34][D][sensor:112]: 'Main Bedroom Bathroom Humidity': Sending state 57.70000 % with 2 decimals of accuracy
[22:20:36][D][dht:048]: Got Temperature=25.5°C Humidity=57.7%
[22:20:36][D][sensor:112]: 'Main Bedroom Bathroom Temperature': Sending state 25.50000 °C with 2 decimals of accuracy
[22:20:36][D][sensor:112]: 'Main Bedroom Bathroom Humidity': Sending state 57.70000 % with 2 decimals of accuracy
[22:20:38][W][dht:170]: Falling edge for bit 39 failed!
[22:20:38][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number).
[22:20:38][D][sensor:112]: 'Main Bedroom Bathroom Temperature': Sending state nan °C with 2 decimals of accuracy
[22:20:38][D][sensor:112]: 'Main Bedroom Bathroom Humidity': Sending state nan % with 2 decimals of accuracy
[22:20:40][D][dht:048]: Got Temperature=25.5°C Humidity=57.8%
[22:20:40][D][sensor:112]: 'Main Bedroom Bathroom Temperature': Sending state 25.50000 °C with 2 decimals of accuracy
[22:20:40][D][sensor:112]: 'Main Bedroom Bathroom Humidity': Sending state 57.80000 % with 2 decimals of accuracy
[22:20:42][D][dht:048]: Got Temperature=25.5°C Humidity=57.8%
[22:20:42][D][sensor:112]: 'Main Bedroom Bathroom Temperature': Sending state 25.50000 °C with 2 decimals of accuracy
[22:20:42][D][sensor:112]: 'Main Bedroom Bathroom Humidity': Sending state 57.80000 % with 2 decimals of accuracy
[22:20:44][W][dht:170]: Falling edge for bit 39 failed!
[22:20:44][W][dht:060]: Invalid readings! Please check your wiring (pull-up resistor, pin number).

srnoth, 2021-12-30:
Confirming this is still an issue on the latest ESP Home (2021.12.1). All "solved" by using the latest framework but of course that is just a work around.

srnoth, 2022-02-21:
Anyone still experiencing this issue after the latest update bumps the default framework automatically? My configs are now working without specifying 'version: latest' for the framework.