esphome/issues #2555: BME680 BSEC - Wemos D1 - ESPHome

SHHA24, 2021-10-16:
### The problem

Hi running the ESPHome config for the BME680 with the BSec as per instructions.
All seems to be working and levels are the same as previous configuration. 
BUT with the new BSec and ESPHome the IAQ reading is very high 1651 IAQ ? 
it is in calibrated state, but this sensor reading doesn’t change a lot, only 1625 to 1660 ….. any ideas? 

### Which version of ESPHome has the issue?

2021.9.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.10.5

### What platform are you using?

ESP8266

### Board

Wemos D1

### Component causing the issue

BME680

### Example YAML snippet

```yaml
binary_sensor:
  - platform: status
    name: "US Temp Status"

bme680_bsec:
    address: 0x76
    sample_rate: lp   
    iaq_mode: static
    state_save_interval: 6h
logger:
  logs:
    custom_components.iaquk: debug
    
    
    
sensor:
  - platform: bme680_bsec
    temperature:
      name: "US Temperature"
      sample_rate: lp
      filters:
        - median
    pressure:
      name: "US Pressure"
      sample_rate: lp
      filters:
        - median
    humidity:
      name: "US Humidity"
      sample_rate: lp
      filters:
        - median
    gas_resistance:
      # Gas resistance in Ω
      name: "Gas Resistance"
      filters:
        - median
    iaq:
      name: "Home IAQ"
      filters:
        - median
    iaq_accuracy:
      # IAQ accuracy as a numeric value of 0, 1, 2, 3
      name: "Home IAQ Accuracy"    
        
        
    co2_equivalent:
      name: "Home CO2 Equivalent"
    breath_voc_equivalent:
      name: "Home Breath VOC Equivalent"
 

  - platform: sds011
    pm_2_5:
      name: "PM <2.5µm Concentration"
    pm_10_0:
      name: "PM <10.0µm Concentration"
    update_interval: 15min
 
text_sensor:
  - platform: bme680_bsec
    iaq_accuracy:
      # IAQ accuracy as a text value of Stabilizing, Uncertain, Calibrating, Calibrated
      name: "Accuracy"
```

prashnts, 2021-10-31:
I can confirm that the Gas Resistance reading does not seem to recover. The "test case" in question is based on cigarette smoke in proximity of the sensor. Earlier without using the BSEC algorithm, the smoke in proximity would cause the resistance to go down sharply and then recover slowly but steadily. Check the graph below for difference between the two algorithms:

[image]

On the right after the flat line (sensor was offline), the value is only decreasing, while on the left you can see that it changed in response to cigarette smoke, every time I smoked.

This makes the sensor very unusable.

mithrandir42, 2023-05-16:
I also got strange eCO2 / IAQ readings after time. Even when stabilizing after power on (usually the IAQ value is 25 then, if I remember correctly)  I got an IAQ reading of 229. Strange enough, only one setup was affected, others worked like usual. For my setup I was able to solve this by erasing the flash and afterwards flashing esphome again.

```
../.local/bin/esptool.py --chip esp32 erase_flash    (press boot button)
../.local/bin/esphome upload bme680.yaml
```
Perhaps this also works in your case.