esphome/issues #2574: GPIO Pin 0 does not support Pullup mode

jrstough, 2021-10-21:
### The problem

After updating ESPHome to version 2021.10.0 I get "GPIO Pin 0 does not support pullup pin mode. Please choose another pin."  I have 3 boards that are doing the same thing.  These worked without any errors before the update.  See code attached.

### Which version of ESPHome has the issue?

2021.10.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.10.6

### What platform are you using?

ESP8266

### Board

D1 Mini and Node MCU

### Component causing the issue

GPIO

### Example YAML snippet

```yaml
esphome:
  name: basement_utilities
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: !secret iot_wifi_ssid
  password: !secret iot_wifi_pw
  manual_ip:
    static_ip: 192.168.10.106
    gateway: 192.168.10.1
    subnet: 255.255.255.0
# Enable logging
logger:
  level: debug

# Enable Home Assistant API
api:
   password: !secret esp_node_pw

ota:

# Blue Light 1  
output:
  - platform: gpio
    inverted: true
    pin: D0
    id: frezzer_ok
  - platform: gpio
    pin: D1
    id: frezzer_alarm
  - platform: gpio
    pin: D2
    id: frezzer_not_ok    
    
  - platform: esp8266_pwm
    pin: D8
    id: alarm_test
    
light:
  - platform: monochromatic
    name: "pwm alarm"
    output: alarm_test
    
binary_sensor:
  - platform: gpio
    pin:
      number: D2
    name: "Deep Frezer Temp"
    device_class: heat
    
  - platform: gpio
    pin:
      number: D3
      mode: INPUT_PULLUP
      inverted: True
    name: "Frezzer Alarm Silenced"
    
#Blue Light 2 on Board
status_led:
  pin:
    number: D4
    inverted: True

#Read Dallas Comms
dallas:
  - pin: D6
  
#Read Deep Freezer Temp  
sensor:
  - platform: dallas
    address: 0xCC01192E0E1C1828
    name: "Deep Freezer Temperature"
    #Setpoints are in Degrees C
    on_value_range:
        #Turn on Below 17.8 F or -17.8 C
        - below: -17.8
          then:
            - output.turn_on: frezzer_ok
            - output.turn_off: frezzer_not_ok
        #Turn Off Above 5 F or -15 C
        - above: -15
          then:
            - output.turn_off: frezzer_ok
            - output.turn_on: frezzer_not_ok
            
#Basement Node MCU Wifi Strength
  - platform: wifi_signal
    name: "Basement Node MCU WiFi"
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/basement_utilities.yaml...
WARNING 'basement_utilities': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
Failed config

binary_sensor.gpio: [source /config/esphome/basement_utilities.yaml:52]
  platform: gpio
  pin:  [source /config/esphome/basement_utilities.yaml:54]
    number: D3
    
    GPIO Pin 0 does not support pullup pin mode. Please choose another pin.
    mode: INPUT_PULLUP [source /config/esphome/basement_utilities.yaml:55]
    inverted: True
  name: Frezzer Alarm Silenced
```

randonkade, 2021-10-21:
Same issue here

santa81, 2021-10-21:
gpio 0 is always pulled up, so   you don't need to pull up them.

mvoort1337, 2021-10-21:
I have the same issue with the Sonoff Basic devices.
Sample code on the Sonoff basic on the ESPHome site even gives the GPIO0 as a reference to use the with code for the Sonoff basic devices:

[(https://esphome.io/devices/sonoff_basic.html)]

`binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: true
    name: "Sonoff Basic Button"
    on_press:
      - switch.toggle: relay`

Previous versions up to 2021.9.3 worked fine with this configuration.

On the generic esp8266 page the GPIO0 is listed as a boot mode controller:
[(https://esphome.io/devices/esp8266.html)]

santa81, 2021-10-21:
mode = INPUT

then is everything fine again

mvoort1337, 2021-10-21:
that does seem to work for me.
Could recompile without an error message.
Flased the sonoff basic with the mode: INPUT (instead of mode: INPUT_PULLUP)

Tested and works fine, buttons also still work.

0n3-70uch, 2021-10-21:
> no, only gpio 0, because that INPUT is always pulled up. so you don't can configure that.

Yeah I know but the docs are wrong for GPIO0. Look here (one example):
https://esphome.io/devices/sonoff_basic.html