esphome/issues #2576: Warning! Could not find the package with 'Update' requirements for your system 'linux_x86_64'

5A52, 2021-10-21:
### The problem

Can not compile after upgrading to  2021.10.0
[image]

### Which version of ESPHome has the issue?

2021.10.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

core-2021.10.6

### What platform are you using?

ESP8266

### Component causing the issue

undefined reference to `esphome::light::ESPColorView:

### Example YAML snippet

```yaml
substitutions:
  devicename: espblb1_rroom
  

esphome:
  name: $devicename
  platform: ESP8266
  board: esp_wroom_02

packages:
  common: !include common/common.yaml

output:
  - platform: esp8266_pwm
    id: output_red
    pin: GPIO4
  - platform: esp8266_pwm
    id: output_green
    pin: GPIO12
  - platform: esp8266_pwm
    id: output_blue
    pin: GPIO14
  - platform: esp8266_pwm
    id: output_white
    pin: GPIO5

e131:
  method: unicast

light:
  - platform: rgbw
    internal: true
    id: l1
    red: output_red
    green: output_green
    blue: output_blue
    white: output_white

  - platform: partition
    name: $devicename
    id: $devicename
    segments:
      - id: l1
        from: 0
        to: 0      
    effects:
      - e131:
          universe: 1
          channels: RGBW    
      - random:
      - flicker:
```

Marlor, 2021-10-21:
I can confirm the missing package on Home Assistant OS generic-x86-64 with minimal yaml:

```
esphome:
  name: mini
  platform: ESP8266
  board: esp01_1m

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "7a41383e14ef4acd88afe1f4b98bea24"

wifi:
  ssid: ""
  password: ""

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Mini Fallback Hotspot"
    password: "RkMXPqFp0E6T"

captive_portal:
```

```
Library Manager: Installing Update
Warning! Could not find the package with 'Update' requirements for your system 'linux_x86_64'
```

there seems to be a lot of ambiguity from the packages on an cleen install:

```
Tool Manager: Warning! More than one package has been found by toolchain-xtensa @ ~2.40802.191122 requirements:
 - platformio/toolchain-xtensa @ 2.100300.210717
 - mcspr/toolchain-xtensa @ 5.100300.210830
 - tasmota/toolchain-xtensa @ 5.100200.210303
Tool Manager: Please specify detailed REQUIREMENTS using package owner and version (showed above) to avoid name conflicts
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-xtensa @ 2.40802.200502 has been installed!
```

```
Tool Manager: Installing tool-esptool @ <2
Tool Manager: Warning! More than one package has been found by tool-esptool @ <2 requirements:
 - platformio/tool-esptool @ 1.413.0
 - volcas/tool-esptool @ 1.413.1
Tool Manager: Please specify detailed REQUIREMENTS using package owner and version (showed above) to avoid name conflicts
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-esptool @ 1.413.0 has been installed!
```

```
Library Manager: Warning! More than one package has been found by AsyncTCP-esphome requirements:
 - ottowinter/AsyncTCP-esphome @ 1.2.1
 - esphome/AsyncTCP-esphome @ 1.2.2
Library Manager: Please specify detailed REQUIREMENTS using package owner and version (showed above) to avoid name conflicts
```

Full log:
[log.txt](https://github.com/esphome/issues/files/7388586/log.txt)

OttoWinter, 2021-10-21:
@Marlor I meant I can't reproduce the compile error "undefined reference to ..." with the config. The title of this issue is a bit misleading, the Update warning is just that, a warning. iirc the warning is printed because we need to add the Update library from arduino core, _but_ PIO somehow can't find it (but we still need to define it so that the includes work) - in short: it's complicated

Re the more than one found: please try with 2021.10, most of those should be gone now

5A52, 2021-10-21:
> Can't reproduce with the config you provided. Compile works fine here.
> 
> Please try pressing the clean button in the overflow menu on the right, then compile again.
No, it is not compiling even  clean, sorry. Here is a full log if it helps some kind.
[esp.log](https://github.com/esphome/issues/files/7390853/esp.log)

oxan, 2021-10-21:
I can trigger it by removing this line from the generated main.cpp:
```cpp
light_addressablelightwrapper = new light::AddressableLightWrapper(l1);
```

Smells like a compiler bug to me, I really don't see how that line has any relation to the missing symbols it complains about.

oxan, 2021-10-21:
Can't reproduce it with Arduino 3 which uses GCC 10, so yeah, definitely a compiler bug.