esphome/issues #2577: Checksum mismatch

bkbartk, 2021-10-21:
### The problem

After installing ESPHome and update the device via home assistant I only get checksum mismatches

`[09:04:00][C][wifi_signal.sensor:009]: WiFi Signal 'Wi-Fi Signal'
[09:04:00][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[09:04:00][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[09:04:00][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[09:04:00][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[09:04:00][C][wifi_info:009]: WifiInfo IPAddress 'IP Address'
[09:04:00][C][wifi_info:011]: WifiInfo SSID 'Wi-Fi SSID'
[09:04:00][C][wifi_info:012]: WifiInfo BSSID 'Wi-Fi BSSID'
[09:04:00][E][dsmr:154]: !4119

 ^

Checksum mismatch
[09:04:01][E][dsmr:154]: !4023

 ^

Checksum mismatch
[09:04:02][E][dsmr:154]: !5906

 ^

Checksum mismatch`

### Which version of ESPHome has the issue?

2021.10.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.9.7

### What platform are you using?

ESP8266

### Board

slimmelezer

### Example YAML snippet

```yaml
https://raw.githubusercontent.com/zuidwijk/dsmr/main/slimmelezer.yaml
```

### Anything in the logs that might be useful for us?

```txt
[09:04:00][C][wifi_signal.sensor:009]: WiFi Signal 'Wi-Fi Signal'
[09:04:00][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[09:04:00][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[09:04:00][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[09:04:00][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[09:04:00][C][wifi_info:009]: WifiInfo IPAddress 'IP Address'
[09:04:00][C][wifi_info:011]: WifiInfo SSID 'Wi-Fi SSID'
[09:04:00][C][wifi_info:012]: WifiInfo BSSID 'Wi-Fi BSSID'
[09:04:00][E][dsmr:154]: !4119

 ^

Checksum mismatch
[09:04:01][E][dsmr:154]: !4023

 ^

Checksum mismatch
[09:04:02][E][dsmr:154]: !5906

 ^

Checksum mismatch
```

### Additional information

there were no isseus with v 2021.9

michaelarnauts, 2021-10-21:
I see the same issue, I was using the following options on 2021.9, and since 2021.10 uses `glmnet/Dsmr@0.5` now by default, the change should be somewhere in the changes made in ESPHome itself.

```
  platformio_options:
    lib_deps: glmnet/Dsmr@0.5
    lib_ignore: glmnet/Dsmr@0.3
``` 

I'm using a DSMRv5 meter, so I should not have to enable the disable_crc option.

michaelarnauts, 2021-10-21:
I think the issue is due to this PR: https://github.com/esphome/esphome/pull/2382

zuidwijk, 2021-10-21:
@bkbartk the Wemos is consuming about 180~200mA in "normal" use. When configuring wifi, that can easily got to the max 250mA or a little above and than you get brownouts. I always advice to use usb power during wifi setup.

Maybe the wifi usage is too high during transfer of OTA files.

bkbartk, 2021-10-21:
good to know,
still the fix doesn't work for me, not sure why

defl, 2021-10-30:
I can confirm above issues as well. Increasing the UART buffer to 1024 worked as a fix for me using dsmr 0.5.0

denevecon, 2021-10-31:
Thanks to your investigations and solution I got my update to 2020.10.3 running again. Still some Checksum mismatch errors between sensor updates. See below: 

[17:14:53][C][api:135]:   Address: slimmelezer.local:6053
[17:14:53][C][api:139]:   Using noise encryption: NO
[17:14:53][C][wifi_signal.sensor:009]: WiFi Signal 'Wi-Fi Signal'
[17:14:53][C][wifi_signal.sensor:009]:   Device Class: 'signal_strength'
[17:14:53][C][wifi_signal.sensor:009]:   State Class: 'measurement'
[17:14:53][C][wifi_signal.sensor:009]:   Unit of Measurement: 'dBm'
[17:14:53][C][wifi_signal.sensor:009]:   Accuracy Decimals: 0
[17:14:53][C][wifi_info:009]: WifiInfo IPAddress 'IP Address'
[17:14:53][C][wifi_info:011]: WifiInfo SSID 'Wi-Fi SSID'
[17:14:53][C][wifi_info:012]: WifiInfo BSSID 'Wi-Fi BSSID'
[17:14:54][E][dsmr:168]: !68E0
 ^
Checksum mismatch
[17:14:55][E][dsmr:168]: !6FD5
 ^
Checksum mismatch
[17:14:56][E][dsmr:168]: !D225
 ^
Checksum mismatch
[17:14:57][E][dsmr:168]: !A0D5
 ^
Checksum mismatch
[17:14:59][E][dsmr:168]: !8CDF
 ^
Checksum mismatch
[17:15:00][E][dsmr:168]: !F33F
 ^
Checksum mismatch
[17:15:01][E][dsmr:168]: !9F96
 ^
Checksum mismatch
[17:15:02][E][dsmr:168]: !B6DA
 ^
Checksum mismatch
[17:15:03][E][dsmr:168]: !0206
 ^
Checksum mismatch
[17:15:04][D][sensor:113]: 'Energy Consumed Tariff 1': Sending state 9534.44434 kWh with 3 decimals of accuracy
[17:15:04][D][sensor:113]: 'Energy Consumed Tariff 2': Sending state 10324.56836 kWh with 3 decimals of accuracy
[17:15:04][D][sensor:113]: 'Energy Produced Tariff 1': Sending state 0.00000 kWh with 3 decimals of accuracy
[17:15:04][D][sensor:113]: 'Energy Produced Tariff 2': Sending state 0.00000 kWh with 3 decimals of accuracy
[17:15:04][D][sensor:113]: 'Power Consumed': Sending state 898.00000 kW with 0 decimals of accuracy
[17:15:04][D][sensor:113]: 'Power Produced': Sending state 0.00000 kW with 0 decimals of accuracy
[17:15:04][D][sensor:113]: 'Electricity Failures': Sending state 8.00000  with 0 decimals of accuracy
[17:15:04][D][sensor:113]: 'Long Electricity Failures': Sending state 2.00000  with 0 decimals of accuracy
[17:15:04][D][sensor:113]: 'Voltage Phase 1': Sending state 233.10001 V with 1 decimals of accuracy
[17:15:04][D][sensor:113]: 'Current Phase 1': Sending state 4.00000 A with 1 decimals of accuracy
[17:15:04][D][sensor:113]: 'Power Consumed Phase 1': Sending state 898.00000 kW with 0 decimals of accuracy
[17:15:04][D][sensor:113]: 'Power Produced Phase 1': Sending state 0.00000 kW with 0 decimals of accuracy
[17:15:04][D][sensor:113]: 'Gas Consumed': Sending state 5743.83301 m³ with 3 decimals of accuracy
[17:15:04][D][text_sensor:067]: 'DSMR Identification': Sending state 'ISK5\2M550E-1012'
[17:15:04][D][text_sensor:067]: 'DSMR Version': Sending state '50'
[17:15:06][E][dsmr:168]: !4C21
 ^
Checksum mismatch
[17:15:07][E][dsmr:168]: !78E3
 ^
Checksum mismatch
[17:15:08][E][dsmr:168]: !1589
 ^
Checksum mismatch
[17:15:09][D][sensor:113]: 'Energy Consumed Tariff 1': Sending state 9534.44531 kWh with 3 decimals of accuracy
[17:15:09][D][sensor:113]: 'Energy Consumed Tariff 2': Sending state 10324.56836 kWh with 3 decimals of accuracy
[17:15:09][D][sensor:113]: 'Energy Produced Tariff 1': Sending state 0.00000 kWh with 3 decimals of accuracy
[17:15:09][D][sensor:113]: 'Energy Produced Tariff 2': Sending state 0.00000 kWh with 3 decimals of accuracy
[17:15:09][D][sensor:113]: 'Power Consumed': Sending state 887.00000 kW with 0 decimals of accuracy
[17:15:09][D][sensor:113]: 'Power Produced': Sending state 0.00000 kW with 0 decimals of accuracy
[17:15:09][D][sensor:113]: 'Electricity Failures': Sending state 8.00000  with 0 decimals of accuracy
[... 219 lines cut ...]
Checksum mismatch
[17:15:51][E][dsmr:168]: !8082
 ^
Checksum mismatch
[17:15:52][E][dsmr:168]: !BA22
 ^
Checksum mismatch
[17:15:53][E][dsmr:168]: !E8D9
 ^
Checksum mismatch
[17:15:54][D][sensor:113]: 'Energy Consumed Tariff 1': Sending state 9534.45703 kWh with 3 decimals of accuracy
[17:15:54][D][sensor:113]: 'Energy Consumed Tariff 2': Sending state 10324.56836 kWh with 3 decimals of accuracy
[17:15:54][D][sensor:113]: 'Energy Produced Tariff 1': Sending state 0.00000 kWh with 3 decimals of accuracy
[17:15:54][D][sensor:113]: 'Energy Produced Tariff 2': Sending state 0.00000 kWh with 3 decimals of accuracy
[17:15:54][D][sensor:113]: 'Power Consumed': Sending state 885.00000 kW with 0 decimals of accuracy
[17:15:54][D][sensor:113]: 'Power Produced': Sending state 0.00000 kW with 0 decimals of accuracy
[17:15:54][D][sensor:113]: 'Electricity Failures': Sending state 8.00000  with 0 decimals of accuracy
[17:15:54][D][sensor:113]: 'Long Electricity Failures': Sending state 2.00000  with 0 decimals of accuracy
[17:15:54][D][sensor:113]: 'Voltage Phase 1': Sending state 232.50000 V with 1 decimals of accuracy
[17:15:54][D][sensor:113]: 'Current Phase 1': Sending state 4.00000 A with 1 decimals of accuracy
[17:15:54][D][sensor:113]: 'Power Consumed Phase 1': Sending state 879.00000 kW with 0 decimals of accuracy
[17:15:54][D][sensor:113]: 'Power Produced Phase 1': Sending state 0.00000 kW with 0 decimals of accuracy
[17:15:54][D][sensor:113]: 'Gas Consumed': Sending state 5743.83301 m³ with 3 decimals of accuracy
[17:15:54][D][text_sensor:067]: 'DSMR Identification': Sending state 'ISK5\2M550E-1012'
[17:15:54][D][text_sensor:067]: 'DSMR Version': Sending state '50'

mmakaay, 2021-10-31:
The main reason that keeps popping up for errors like this, after incorporating the fixes (rx_buffer_size increase + my patched code) is that the UART is not hardware-based but software-based.
What does your YAML configuration look like?

Two things that might disturb the use of hardware UART: 
* Not having `baud_rate: 0` for the `logger` component
* Using an RX pin that is not hardware-UART capable. E.g. on ESP8266 the pin ought to be GPIO13 (i.e. D7 on a D1 mini)

A quick way to check is connecting to the logger, and inspecting the setup information for the UART component. The information will show hardware or software UART.

denevecon, 2021-10-31:
I misinterpreted the instructions and commented the rx_buffer_size increase. Changing this and the baud_rate for the logger resulted in an error-less log output. Btw the UART is hardware based. Thanks again.

patvdleer, 2021-10-31:
I'm running `ESPHOME 2021.10.3` with the zuidwijk 1.1.1v on a Kaifa E0025 with DSMI 4.2, sadly setting the `rx_buffer_size` did not fix it. Even with the `crc_check` set to false I still get the `Checksum mismatch` errors.

```yaml
uart:
  baud_rate: 115200
  rx_pin: D7
  rx_buffer_size: 1024

dsmr:
  id: dsmr_instance
  crc_check: false
```

mmakaay, 2021-12-10:
Various fixes have been done to the DSMR component and these likely have fixed the reported issues.
All fixes are in the upcoming version 2021.12.0 of ESPHome. If anybody still has problems with that version, then please create a new issue for it, with information about the log output, uart debugger and meter type.

closed 2021-12-10 by mmakaay

zuidwijk, 2021-12-23:
Hey @mmakaay ... I've compiled a firmware on 2021.12.0 and 2021.12.1 and notice some Checksum mismatch errors. Do you still need to add the `rx_buffer_size` to the uart? And do you still need to add your `fix-dsmr-read-chunk-size`?

mmakaay, 2021-12-23:
No, remove any external component that you have in your config. All fixes have been incorporated.
With my changes, the rx_buffer_size >= max telegram size is not required, though it is recommended. When using tn rx_buffer_size that matches the telegram size, the code won't have to do any forced filling of the telegram buffer anymore.

zuidwijk, 2021-12-24:
With and without your external code, and with or without rx_buffer_size, I keep getting those error messages. Opening new case? or continu an existing case?

mmakaay, 2021-12-24:
Definitely a new one, the issues from this were fixed and it is unwieldy as is.

zuidwijk, 2021-12-24:
Found the issue, and I don't like it...

I'm using `improv_serial:` and that requires the logger baud_rate to be not 0.
Disable improv_serial and set baud_rate to 0 works.

zuidwijk, 2021-12-24:
https://github.com/esphome/issues/issues/2873