esphome/issues #2595: Compile error with latest update

truglodite, 2021-10-22:
### The problem

I just updated a bunch of my devices from esphome 2021.9.x to 10.1, and one of them is giving me compile errors. The device with the same yaml worked fine before the update. I tried removing some features I didn't really need on it, like the hotspot and encryption, but the same error keeps recurring no matter what I try.

### Which version of ESPHome has the issue?

2021.10.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.10.6

### What platform are you using?

ESP8266

### Board

Wemos D1

### Component causing the issue

Not apparent

### Example YAML snippet

```yaml
substitutions:
  name: "master"
  friendly_name: "Master"

esphome:
  name: ${name}
  platform: ESP8266
  board: nodemcuv2

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

logger:

api:
  password: !secret master_password
  encryption:
    key: !secret encryption_key
# Enable ota
ota:
  password: !secret ota_password

# Device Specific Configs ###########################################
# Wemos D1 Mini
#
# dallas -> GPIO16 (D0)
# sda -> GPIO5 (D1)
# scl -> GPIO4 (D2)
# soft rx -> GPIO0 (D3) [keep high on boot!!!]
# built in LED -> GPIO2 (D4) [keep high on boot!!!]
# DHT -> GPIO14(D5)
# motion -> GPIO12(D6)
# soft tx -> GPIO13(D7)
# button (to Vcc) -> GPIO15(D8) [keep low on boot!!!]

uart:
  tx_pin: D7
  rx_pin: D3
  baud_rate: 9600

i2c:
  sda: D1
  scl: D2

dfplayer:

time:
  - platform: homeassistant
    id: ha_time
    
text_sensor:
  - platform: homeassistant
    name: "ESPhome Voice"
    internal: true
    entity_id: input_text.esphome_voice
    on_value:
      then:
        - dfplayer.set_volume: 25
        - delay: 150ms
        - dfplayer.play_folder:
            folder: 1
            file: !lambda return atoi(x.c_str());
            
font:
  - file: "verdana.ttf"
    id: my_fontlg
    size: 20
  - file: "verdana.ttf"
    id: my_fontmd
    size: 18
  - file: "verdana.ttf"
    id: my_fontsm
    size: 10
    
display:
  - platform: ssd1306_i2c
    id: my_display
    pages:
      - id: page1
        lambda: |-
          auto nowTime = id(ha_time).now();
          int date = nowTime.day_of_month;
          
          it.strftime(120, 20, id(my_fontlg), TextAlign::BOTTOM_RIGHT, "%I:%M", nowTime);
          if (nowTime.hour > 0 && nowTime.hour < 10) {
            it.filled_rectangle(57, 0, 15, 21, COLOR_OFF);
          }
          else if (nowTime.hour >= 13 && nowTime.hour <= 21)  {
            it.filled_rectangle(57, 0, 15, 21, COLOR_OFF);
          }
          
          if (nowTime.hour < 12) {
            it.print(121, 10, id(my_fontsm), TextAlign::BOTTOM_LEFT, "A");
          }
          else {
            it.print(121, 18, id(my_fontsm), TextAlign::BOTTOM_LEFT, "P");
          }
          
          it.strftime(0, 10, id(my_fontsm), TextAlign::BOTTOM_LEFT, "%A", nowTime);
          
          it.strftime(0, 18, id(my_fontsm), TextAlign::BOTTOM_LEFT, "%b. ", nowTime);
          if (nowTime.day_of_month == 1 || nowTime.day_of_month == 21 || nowTime.day_of_month == 31)  {
            it.printf(49, 18, id(my_fontsm), TextAlign::BOTTOM_RIGHT, "%dst", date);
          }
          else if (nowTime.day_of_month == 2 || nowTime.day_of_month == 22) {
            it.printf(49, 18, id(my_fontsm), TextAlign::BOTTOM_RIGHT, "%dnd", date);
          }
          else if (nowTime.day_of_month == 3 || nowTime.day_of_month == 23) {
            it.printf(49, 18, id(my_fontsm), TextAlign::BOTTOM_RIGHT, "%drd", date);
          }
          else {
            it.printf(49, 18, id(my_fontsm), TextAlign::BOTTOM_RIGHT, "%dth", date);
          }
          
          it.print(it.get_width() / 4, 16, id(my_fontsm), TextAlign::TOP_CENTER, "Inside");
          it.printf(it.get_width() / 4, 24, id(my_fontmd), TextAlign::TOP_CENTER, "%.0f %%", id(thermostat_humid).state);
          it.printf(it.get_width() / 4, 42, id(my_fontlg), TextAlign::TOP_CENTER, "%.0f F", id(inside_temp).state);
          it.print(3*it.get_width() / 4, 16, id(my_fontsm), TextAlign::TOP_CENTER, "Outside");
          it.printf(3*it.get_width() / 4, 24, id(my_fontmd), TextAlign::TOP_CENTER, "%.0f %%", id(backyard_humid).state);
          it.printf(3*it.get_width() / 4, 42, id(my_fontlg), TextAlign::TOP_CENTER, "%.0f F", id(backyard_temp).state);
          
          it.line(0, 16, 127, 16);
          it.line(63, 16, 63, 63);
          
          if (id(button).state) {
            it.rectangle(0, 0, 128, 64);
          }
    model: "SSD1306 128x64"
    reset_pin: D0
    address: 0x3C
interval:
  - interval: 3s
    then:
      - display.page.show_next: my_display
      - component.update: my_display

# Binaries
binary_sensor:
  - platform: status
    name: "${friendly_name} Status"
    
  - platform: gpio
    pin:
      number: GPIO15
      mode: INPUT_PULLUP
    name: "${friendly_name} Button"
    id: button
    filters:
      - delayed_on: 10ms
      - delayed_off: 10ms
    
  - platform: gpio
    pin:
      number: D6
      mode: INPUT
    name: "${friendly_name} Motion"
    device_class: motion
    filters:
      - delayed_off: 5000ms
  
dallas:
  - pin: D0
  
status_led:
  pin: D4

sensor:
  - platform: dallas
    index: 0
    resolution: 12
    accuracy_decimals: 1
    name: "${friendly_name} Temperature"
    id: celsius
    unit_of_measurement: "°F"
    filters:
      - lambda: return x * (9.0 / 5.0) + 32.0;
      
  - platform: wifi_signal
    name: "${friendly_name} WiFi Signal"
    update_interval: 60s
    
  - platform: dht
    pin: D5
    model: DHT22
    humidity:
      accuracy_decimals: 0
      name: "${friendly_name} Humidity"
    update_interval: 60s
        
  - platform: bh1750
    name: "${friendly_name} Luminance"
    address: 0x23
    update_interval: 10s
        
  - platform: homeassistant
    name: "Backyard Temp"
    id: backyard_temp
    entity_id: sensor.backyard_temp
    internal: true
  - platform: homeassistant
    name: "Backyard Humid"
    id: backyard_humid
    entity_id: sensor.backyard_humid
    internal: true
  - platform: homeassistant
    name: "Inside Temp"
    id: inside_temp
    entity_id: sensor.inside_temp_avg
    internal: true
  - platform: homeassistant
    name: "Inside Humid"
    id: thermostat_humid
    entity_id: sensor.thermostat_humidity
    internal: true
```

### Anything in the logs that might be useful for us?

```txt
Linking /data/master/.pioenvs/master/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld: /data/master/.pioenvs/master/firmware.elf section `.text1' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
*** [/data/master/.pioenvs/master/firmware.elf] Error 1
========================= [FAILED] Took 88.89 seconds =========================
```

### Additional information

My complete yaml is pasted above. The device has like I said worked before the latest esphome update, for years in fact.

ssieb, 2022-02-26:
I can verify that it is a problem.  It's caused by this line in the linker script:
```
MEMORY
{
  iram1_0_seg : org = 0x40100000, len = 0x8000
}
```
I don't know why it's this size.  I made it bigger and it compiled.  The memory use line went up to:
```
RAM:   [=====     ]  49.3% (used 40416 bytes from 81920 bytes)
```
(It was around 39000 when I reduced the config enough to compile with the original setting.)
I haven't verified that it will actually run on the device yet.
I also don't know if that changed at some point or if the memory use of components has increased.

monsieurhannes, 2022-04-01:
I have the same issue flashing a Gosund SP01 (ESP8266) with ESPHome 2022.3.2:

```
INFO Reading configuration /config/esphome/esphome-web-2252df.yaml...
INFO Detected timezone 'Europe/Stockholm'
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Deleting /data/power-01/.pioenvs
INFO Compiling app...
Processing power-01 (board: esp8285; framework: arduino; platform: platformio/espressif8266 @ 3.2.0)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|-- <ArduinoJson> 6.18.5
Compiling /data/power-01/.pioenvs/power-01/src/esphome/components/api/api_connection.cpp.o
Compiling /data/power-01/.pioenvs/power-01/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/power-01/.pioenvs/power-01/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/power-01/.pioenvs/power-01/src/esphome/components/api/api_pb2_service.cpp.o
```

### My config
## System Health

version | core-2022.3.8
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.92-v8
arch | aarch64
timezone | UTC

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4961
Installed Version | 1.24.0
Stage | running
Available Repositories | 1080
Downloaded Repositories | 6

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.5
-- | --
update_channel | stable
supervisor_version | supervisor-2022.03.5
docker_version | 20.10.9
disk_total | 29.0 GB
disk_used | 8.6 GB
healthy | true
supported | true
board | rpi3-64
supervisor_api | ok
version_api | ok
installed_addons | ESPHome (2022.3.2), JupyterLab (0.7.0), Bookstack (1.0.2), Studio Code Server (4.2.0), TasmoAdmin (0.17.0), deCONZ (6.12.0), Terminal & SSH (9.3.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 0
views | 1
mode | storage

</details>

I use a Raspberry Pi 3. Is there any workaround?

Lafred, 2022-04-27:
Running into a similar problem today.  

`INFO Reading configuration /config/esphome/asdf-90.yaml...
WARNING GPIO0 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Generating C++ source...
INFO Compiling app...
Processing asdf-90 (board: esp32cam; framework: arduino; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <ESPmDNS> 1.0
Linking /data/asdf-90/.pioenvs/asdf-90/firmware.elf
collect2: fatal error: ld terminated with signal 11 [Segmentation fault], core dumped
compilation terminated.
*** [/data/asdf-90/.pioenvs/asdf-90/firmware.elf] Error 1
========================= [FAILED] Took 15.65 seconds =========================
`

ssieb, 2022-08-02:
For all those posting about compiling crashing or not completing, this issue is not about that.  That's a problem with using a too small device to compile with (usually a pi3).
This issue is about the `section `.text1' will not fit in region `iram1_0_seg'` error.