esphome/issues #2662: DSMR message parsing only partially

bennierex, 2021-10-30:
### The problem

I've been running my custom hardware and DSMR component now for about a year without any major issues (apart for some random WiFi and API disconnects). This was based on the [P1Parser created by Matthijs Kooijman](https://github.com/matthijskooijman/arduino-dsmr/blob/master/src/dsmr/parser.h).

Yesterday, I wanted to check out the new ESPHome native DSMR component, so I flashed my ESP32 with a new firmware. Unfortunately I haven't been able to get it to work properly. The parsing of the messages (my smart meter is using v5.0) seems very unreliable and only the first part can be parsed until it starts erring out (see the logs below); the pattern keeps repeating itself, only being able to parse the first part. Enabling the CRC just skips every message because it fails to correctly cheksum the whole message.

For now I've reverted back to my custom code using the P1Parser of Matthijs, which works without problems (ruling out hardware problems afaik).

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.10.6

### What platform are you using?

ESP32

### Board

esp32doit-devkit-v1

### Component causing the issue

DSMR

### Example YAML snippet

```yaml
uart:
  - rx_pin: GPIO16
    tx_pin: GPIO17
    baud_rate: 115200
    id: uart_bus

dsmr:
  uart_id: uart_bus
  crc_check: false

switch:
  - platform: gpio
    name: "DSMR enable"
    pin: GPIO32
    restore_mode: ALWAYS_ON
    internal: true
```

### Anything in the logs that might be useful for us?

```txt
[10:28:09][D][sensor:113]: 'Electricity Delivered Tariff 1': Sending state 8821.79688 kWh with 3 decimals of accuracy
[10:28:09][D][sensor:113]: 'Electricity Delivered Tariff 2': Sending state 9095.10254 kWh with 3 decimals of accuracy
[10:28:09][D][sensor:113]: 'Electricity Returned Tariff 1': Sending state 516.06201 kWh with 3 decimals of accuracy
[10:28:09][D][sensor:113]: 'Electricity Returned Tariff 2': Sending state 1223.71497 kWh with 3 decimals of accuracy
[10:28:09][D][sensor:113]: 'Actual Power Delivered': Sending state 0.41600 kW with 3 decimals of accuracy
[10:28:09][D][sensor:113]: 'Actual Power Returned': Sending state 0.00000 kW with 3 decimals of accuracy
[10:28:09][D][sensor:113]: 'Voltage L2': Sending state 228.70000 V with 1 decimals of accuracy
[10:28:09][D][sensor:113]: 'Power Delivered L2': Sending state 0.13600 kW with 3 decimals of accuracy
[10:28:09][D][text_sensor:067]: 'DSMR Identification': Sending state 'ISK5\2M550T-1012'
[10:28:09][D][text_sensor:067]: 'DSMR Version': Sending state '50'
[10:28:09][D][text_sensor:067]: 'Electricity Tariff': Sending state '0001'
[10:28:23][E][dsmr:154]: 1-0:!0718
    ^
Last dataline not CRLF terminated
[10:28:24][E][dsmr:154]: 1-0:21.7.0(01-0:42.7.0(00.000*kW)
             ^
Invalid number
[10:28:26][E][dsmr:154]: 1-0:32.7.0(229.1-0:72.7.0(228.6*V)
                ^
Invalid number
[10:28:36][E][dsmr:154]: 1-0:!FA66
    ^
Last dataline not CRLF terminated
[10:28:38][E][dsmr:154]: 1-0:!0026
    ^
Last dataline not CRLF terminated
[10:28:44][E][dsmr:154]: 1-0:!7C2E
    ^
Last dataline not CRLF terminated
```

### Additional information

The first part of the DSMR message it actually parses, seems to contain the expected values.

StijnVdd, 2021-10-31:
I have the same issue when flashing ESPHome to the SlimmeLezer by Marcel Zuidwijk

jsuanet, 2021-10-31:
This is a known issue and a PR (2622) for this problem is in review.  For now you can use the workaround described here: https://github.com/esphome/issues/issues/2621#issuecomment-950863235

bennierex, 2021-11-01:
@jsuanet Thanks! I had missed the issue has already been reported. I just tried the workaround and it looks like it's working again!

I'll try to have a look at optimizing this when I got some time to spare.

closed 2021-11-01 by bennierex