esphome/issues #2663: Modbus sensor variables value_type S_DWORD wrong

songchan322, 2021-10-30:
### The problem

Modbus sensor variables **value_type** _S_DWORD_ actual is _unsigned float from 2 registers_
same as _U_DWORD_ 

### Which version of ESPHome has the issue?

2021.9.0

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.6

### What platform are you using?

ESP8266

### Board

sonoff basic r3

### Example YAML snippet

```yaml
esphome:
  name: chint-meter
  platform: ESP8266
  board: esp01_1m

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:

ota:
  password: "*****"

wifi:
  ssid: "ssid"
  password: "password"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Chint-Meter Fallback Hotspot"
    password: "cH4Klu1Yw0op"

captive_portal:

external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: dev
    components: [ modbus, modbus_controller ]
  
uart:
  id: mod_bus
  tx_pin: 1
  rx_pin: 3
  baud_rate: 9600
  stop_bits: 2

modbus:
  id: modbus1
  send_wait_time: 2000ms

modbus_controller:
  - id: epever
    ## the Modbus device addr
    address: 0x24
    modbus_id: modbus1
    setup_priority: -10
    update_interval: 10s
    
sensor:
  - platform: modbus_controller
    modbus_controller_id: epever
    id: power_meter_voltage
    name: "power-meter-voltage"
    address: 0x2000
    unit_of_measurement: "V"
    register_type: holding
    value_type: S_DWORD
    accuracy_decimals: 1
  - platform: modbus_controller
    modbus_controller_id: epever
    id: power_meter_energy
    name: "power-meter-energy"
    address: 0x4000
    unit_of_measurement: "kWh"
    register_type: holding
    value_type: S_DWORD
    accuracy_decimals: 2
```

### Anything in the logs that might be useful for us?

```txt
when [value_type: S_DWORD]
**************
[19:25:13][D][modbus_controller.sensor:030]: Sensor new state: 1131262720.00
[19:25:13][D][sensor:118]: 'power-meter-voltage': Sending state 1131262720.00000 V with 1 decimals of accuracy
[19:25:13][D][modbus_controller.sensor:030]: Sensor new state: 1168062464.00
[19:25:13][D][sensor:118]: 'power-meter-energy': Sending state 1168062464.00000 kWh with 2 decimals of accuracy
*********************

when [value_type: U_DWORD]
**********
[19:27:35][D][modbus_controller.sensor:030]: Sensor new state: 1131269376.00
[19:27:35][D][sensor:118]: 'power-meter-voltage': Sending state 1131269376.00000 V with 1 decimals of accuracy
[19:27:36][D][modbus_controller.sensor:030]: Sensor new state: 1168062592.00
[19:27:36][D][sensor:118]: 'power-meter-energy': Sending state 1168062592.00000 kWh with 2 decimals of accuracy
***************
```

martgras, 2021-10-30:
Do you have a link to the register description for your device?
Can you try FP32 or
  FP32_R for value type (missing in the documentation)

oxan, 2021-11-03:
@martgras Is this esphome/esphome#2654?

martgras, 2021-11-03:
Probably not. Don't know what this issue is with more details.
Maybe `value_type: FP32` fixes the issue. 
Documentation is updated in https://github.com/esphome/esphome-docs/pull/1590

closed 2022-03-18 by github-actions[bot]