esphome/issues #2664: OTA Fails

Miro-Georgiev, 2021-10-31:
### The problem

Flashing a BIN file with esphome-flasher do work, but OTA fails. The problem started after updating ESPHome to v2021.10.0 and continues in v2021.10.3. 

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.10.6

### What platform are you using?

ESP32

### Board

ESP32 NodeMCU

### Component causing the issue

Not sure

### Example YAML snippet

```yaml
#GPIO01 DO NOT used for anything! Used for dummy control of Power Supply!

substitutions:
  devicename: living_room
  upper_devicename: Living Room
  
packages:
  device_base: !include common/ESP32_base.yaml

wifi:
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.########

globals:
  # RGB Light RF remote codes - 16 bit, 128 codes
  - id: rgb_map
    type: bool[128][16]
    restore_value: no
    initial_value: '{0,0,0,0,0,0,0,0,0,0,1,0,0,1,0,0,
                     0,0,0,0,0,0,0,1,0,0,1,0,0,1,0,1,
                     0,0,0,0,0,0,1,0,0,0,1,0,0,1,1,0,
                     0,0,0,0,0,0,1,1,0,0,1,0,0,1,1,1,
                     0,0,0,0,0,1,0,0,0,0,1,0,1,0,0,0,
                     0,0,0,0,0,1,0,1,0,0,1,0,1,0,0,1,
                     0,0,0,0,0,1,1,0,0,0,1,0,1,0,1,0,
                     0,0,0,0,0,1,1,1,0,0,1,0,1,0,1,1,
                     0,0,0,0,1,0,0,0,0,0,1,0,1,1,0,0,
                     0,0,0,0,1,0,0,1,0,0,1,0,1,1,0,1,
                     0,0,0,0,1,0,1,0,0,0,1,0,1,1,1,0,
                     0,0,0,0,1,0,1,1,0,0,1,0,1,1,1,1,
                     0,0,0,0,1,1,0,0,0,0,1,1,0,0,0,0,
                     0,0,0,0,1,1,0,1,0,0,1,1,0,0,0,1,
                     0,0,0,0,1,1,1,0,0,0,1,1,0,0,1,0,
                     0,0,0,0,1,1,1,1,0,0,1,1,0,0,1,1,
                     0,0,0,1,0,0,0,0,0,0,1,1,0,1,0,0,
                     0,0,0,1,0,0,0,1,0,0,1,1,0,1,0,1,
                     0,0,0,1,0,0,1,0,0,0,1,1,0,1,1,0,
                     0,0,0,1,0,0,1,1,0,0,1,1,0,1,1,1,
[... 785 lines cut ...]
Compiling /data/living_room/.pioenvs/living_room/src/main.cpp.o
Linking /data/living_room/.pioenvs/living_room/firmware.elf
RAM:   [=         ]  12.8% (used 41984 bytes from 327680 bytes)
Flash: [=====     ]  54.7% (used 1003362 bytes from 1835008 bytes)
Building /data/living_room/.pioenvs/living_room/firmware.bin
========================= [SUCCESS] Took 40.99 seconds =========================
INFO Successfully compiled program.
INFO Connecting to 192.#####
INFO Uploading /data/living_room/.pioenvs/living_room/firmware.bin (1003472 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.##### using esphome API
WARNING Can't connect to ESPHome API for 192.######: Timeout while connecting to ('192.#######', 6053)
INFO Trying to reconnect to 192.###### in the background
```

### Additional information

Only the above YAML config fails on OTA, all other devices I have are performing fine on a 100%.
I did replace the ESP board with new one, but experiencing the same issue.

I can only guess, but could it be the global variable being too large? I had an issue more than an year ago (cannot remember ESPHome version), where I was able to create a bin file, but after flashing it was never starting.

Miro-Georgiev, 2021-10-31:
Removing the initial_value: '{.......}' from the global variable resolved the OTA.

Also noticed that manual upload of the original YAML above is only working on first boot after bin upload with ESPHome-flasher. After reboot will not start.

OttoWinter, 2021-11-02:
Yes very likely due to the `initial_value`. GCC (the compiler) has this so-called "initializer_list" type expression implemented by the array being allocated on the stack. And with this huge object size that could explain why some methods work and others don't (some bootloader init path use more stack than others).

This is not a problem with esphome per se, you really should try to make the init expression smaller. For example encoding it in 16-bit integers (and bit operations) would make the expression shorter to represent on the stack.

Miro-Georgiev, 2021-11-03:
Thank you for looking into this, I will change my approach towards bit operations instead of this huge object.

closed 2021-11-03 by Miro-Georgiev