esphome/issues #2667: cant compile device with HLW8012 sensor as missing pulse_counter_sensor.h

rob8811, 2021-11-02:
### The problem

After updating to latest core and esphome 2021.10.3 all my esphome devices are marked for update.
I tried updating one with a HLW8012 sensor but it failed during compilation as the compiler could not find pulse_counter_sensor.h.  I looked at the devices src/esphome/components/pulse_counter and the directory is empty.
Before this I was using a custom version of the Hlw8012 tailored to the bl0937 made a long time before the hlw8012 could handle it. I removed references to the custom component in the yaml file and setup with hlw8012 as per documentation.

I also tried removing the esphome integration and readdingbit without success. I do not know how to look at the source location for the pulse counter to see if files are there and just not being transferred to the device directories, or if there is a dependancy issue.

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.10.7

### What platform are you using?

ESP8266

### Board

esp01_1m

### Component causing the issue

HLW8012

### Example YAML snippet

```yaml
esphome:
  name: telkin_1
  platform: ESP8266
  board: esp01_1m 

# I have not included other lines from file such as wifi etc 

sensor: 
    
  - platform: hlw8012
    sel_pin: 12
    cf_pin: 5
    cf1_pin: 14
    current_resistor: 0.001f
    voltage_divider: 2401
    model: "BL0937"
    voltage:
      name: "T_1 Voltage"
      unit_of_measurement: V
    current:
      name: "T_1 Current"
      unit_of_measurement: A
    power:
      name:  "T_1 Power"
      unit_of_measurement: Watt
      accuracy_decimals: 1
    energy:
      name: "T_1 Energy"
```

### Anything in the logs that might be useful for us?

```txt
Processing telkin_1 (board: esp01_1m; framework: arduino; platform: platformio/espressif8266 @ 2.6.3) -------------------------------------------------------------------------------- HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash Dependency Graph |-- <ESPAsyncTCP-esphome> 1.2.3 | |-- <ESP8266WiFi> 1.0 |-- <ESPAsyncWebServer-esphome> 2.0.0 | |-- <ESPAsyncTCP-esphome> 1.2.3 | | |-- <ESP8266WiFi> 1.0 | |-- <Hash> 1.0 | |-- <ESP8266WiFi> 1.0 |-- <ESP8266WiFi> 1.0 |-- <Wire> 1.0 |-- <ESP8266mDNS> 1.2 | |-- <ESP8266WiFi> 1.0 |-- <noise-c> 0.1.4 | |-- <libsodium> 1.10018.1 |-- <DNSServer> 1.1.1 | |-- <ESP8266WiFi> 1.0 Compiling /data/telkin_1/.pioenvs/telkin_1/src/esphome/components/hlw8012/hlw8012.cpp.o Compiling /data/telkin_1/.pioenvs/telkin_1/src/esphome/components/light/light_state.cpp.o Compiling /data/telkin_1/.pioenvs/telkin_1/src/esphome/components/logger/logger.cpp.o Compiling /data/telkin_1/.pioenvs/telkin_1/src/esphome/components/md5/md5.cpp.o
In file included from src/esphome/components/hlw8012/hlw8012.cpp:1:0: src/esphome/components/hlw8012/hlw8012.h:6:67: fatal error: esphome/components/pulse_counter/pulse_counter_sensor.h: No such file or directory #include "esphome/components/pulse_counter/pulse_counter_sensor.h" ^ compilation terminated.
```

### Additional information

running on raspberry pi

OttoWinter, 2021-11-02:
Can't reproduce. It's probably an issue with how you set up the custom component.

Please post the directory structure of your config folder. If there's even just a directory called `hlw8012` or `pulse_counter` in `custom_components` (or files named that way with `.py` suffix) that would explain it. Or best just delete the folder if you don't use any custom components.

rob8811, 2021-11-02:
Thanks. There was some custom files left in the custom folder, and also in the device folder. Deleting these allowed the compulation to complete sucessfully. No issues now, thanks again for speedy response.

closed 2021-11-02 by rob8811