esphome/issues #2668: Away warning for ESPHome Climate component

GrKoR, 2021-11-02:
### The problem

I created custom component based on Climate. During the compilation I receive this warning:
```
Compiling .pioenvs\test_aux_ac_local\src\esphome\components\climate\climate_traits.cpp.o
src\esphome\components\climate\climate.cpp: In constructor 'esphome::climate::Climate::Climate(const string&)':
src\esphome\components\climate\climate.cpp:443:60: warning: 'esphome::climate::Climate::away' is deprecated (declared at src\esphome\components\climate\climate.h:198): away is deprecated, use 
preset instead [-Wdeprecated-declarations]
 Climate::Climate(const std::string &name) : EntityBase(name) {}
                                                            ^
```

My code doesn't use away. In source code of Climate class I can't find what generates this warning too.

Here is [the source code of my custom component](https://github.com/GrKoR/esphome_aux_ac_component/tree/dev/components/aux_ac) if it is needed for tests. Yaml can be found [in tests folder](https://github.com/GrKoR/esphome_aux_ac_component/tree/dev/tests).
Warning is generated for DEV branch. The master branch has old version of component.

### Which version of ESPHome has the issue?

2021.10.1

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

esp12e

### Component causing the issue

Climate

### Anything in the logs that might be useful for us?

```txt
Compiling .pioenvs\test_aux_ac_local\src\esphome\components\climate\climate_traits.cpp.o
src\esphome\components\climate\climate.cpp: In constructor 'esphome::climate::Climate::Climate(const string&)':
src\esphome\components\climate\climate.cpp:443:60: warning: 'esphome::climate::Climate::away' is deprecated (declared at src\esphome\components\climate\climate.h:198): away is deprecated, use 
preset instead [-Wdeprecated-declarations]
 Climate::Climate(const std::string &name) : EntityBase(name) {}
                                                            ^
```

OttoWinter, 2021-11-02:
The warning is caused by the constructor here:

https://github.com/GrKoR/esphome_aux_ac_component/blob/505af50dce833701b74c074da17d7a53671c0be2/components/aux_ac/aux_ac.h#L1560-L1562

A solution would be to remove the constructor and use setters - that is also good because `initAC` should really be called from `setup`, not during the construction process.

I already looked into this a couple of weeks ago and there's not really any way for us to get rid of those warnings until we remove the deprecated parameter completely.

GrKoR, 2021-11-02:
Thnx!

closed 2021-11-03 by OttoWinter