esphome/issues #2669: No buffer space available

roberttco, 2021-11-02:
### The problem

After a new install of the ESPHome add-in in Home Assistant, the logs show a no buffer space available error. 

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

Core 2021.10.8, Supervisor 2021.10.8

### What platform are you using?

ESP8266

### Board

d1-mini

### Component causing the issue

n/a

### Example YAML snippet

```yaml
This issue is independent of the board code.  In fact there are no devices configured.
```

### Anything in the logs that might be useful for us?

```txt
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 00-banner.sh: executing... 
-----------------------------------------------------------
 Add-on: ESPHome
 ESPHome add-on for intelligently managing all your ESP8266/ESP32 devices
-----------------------------------------------------------
 Add-on version: 2021.10.3
 You are running the latest version of this add-on.
 System: Debian GNU/Linux 11 (bullseye)  (amd64 / qemux86-64)
 Home Assistant Core: 2021.10.7
 Home Assistant Supervisor: 2021.10.8
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
[cont-init.d] 00-banner.sh: exited 0.
[cont-init.d] 01-log-level.sh: executing... 
[cont-init.d] 01-log-level.sh: exited 0.
[cont-init.d] 10-requirements.sh: executing... 
[cont-init.d] 10-requirements.sh: exited 0.
[cont-init.d] 20-nginx.sh: executing... 
[cont-init.d] 20-nginx.sh: exited 0.
[cont-init.d] 30-dirs.sh: executing... 
[cont-init.d] 30-dirs.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[10:43:20] INFO: Waiting for dashboard to come up...
[services.d] done.
[10:43:20] INFO: Starting ESPHome dashboard...
2021-11-02 10:43:20,724 INFO Starting dashboard web server on unix socket /var/run/esphome.sock and configuration dir /config/esphome...
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/esphome/esphome/dashboard/dashboard.py", line 617, in run
    zc = EsphomeZeroconf()
  File "/usr/local/lib/python3.9/dist-packages/zeroconf/_core.py", line 447, in __init__
    listen_socket, respond_sockets = create_sockets(interfaces, unicast, ip_version, apple_p2p=apple_p2p)
  File "/usr/local/lib/python3.9/dist-packages/zeroconf/_utils/net.py", line 359, in create_sockets
    if add_multicast_member(cast(socket.socket, listen_socket), i):
  File "/usr/local/lib/python3.9/dist-packages/zeroconf/_utils/net.py", line 276, in add_multicast_member
    listen_socket.setsockopt(socket.IPPROTO_IP, socket.IP_ADD_MEMBERSHIP, _value)
OSError: [Errno 105] No buffer space available
[10:43:21] INFO: Starting NGINX...
```

### Additional information

I'm not sure when this started happening and I started looking into it when I saw devices were offline in HA.  The server is running Debian Bullseye with the latest updates (as of 11/1/2021) - 
5.10.0-9-amd64 kernel
8GB RAM (2GB free)

hakha4, 2021-11-11:
Found this issue solved by  following commands:

sudo sysctl -w net.ipv4.igmp_max_memberships=50
sudo sysctl -w net.ipv4.igmp_max_msf=30

javilopezalarcon, 2021-11-27:
> Found this issue solved by following commands:
> 
> sudo sysctl -w net.ipv4.igmp_max_memberships=50 sudo sysctl -w net.ipv4.igmp_max_msf=30

You just solved the problem that had been behind him for a long time and I did not know how to solve it, which also had it in raspbian busted 32, raspian bullseye 32 and in the beta raspberry pi os 64 ...

Thank you very much, really 👍🏻👍🏻

mramirezd, 2021-11-28:
> Found this issue solved by following commands:
> 
> sudo sysctl -w net.ipv4.igmp_max_memberships=50 sudo sysctl -w net.ipv4.igmp_max_msf=30

Thanks a lot for this! In my case I just had to change the `net.ipv4.igmp_max_memberships` to fix the issue. In order to keep it fixed after rebooting the raspberry, I had to edit the `/etc/sysctl.conf` and add a line with `net.ipv4.igmp_max_memberships=50`

roberttco, 2021-12-31:
Thank you @mramirezd and @hakha4 !  I just updated by host OS (running supervised) and this happened again.  Alas I forgot I changed this setting before.  Now it will remain fixed.

rdehuyss, 2022-08-05:
Quick question: I'm running esphome in a docker container. Should I run these fixes in the docker container or on my host system?

roberttco, 2022-08-05:
This happens in a supervised install of Home assistant - thus a container.

jpaulickcz, 2023-02-19:
FWIW I had the same issue (ESPHome in a Docker Stack with `network_mode = host`) and the `OSError: [Errno 105] No buffer space available` was solved by running the
```
sysctl -w net.ipv4.igmp_max_memberships=50
```
on the host system (not in eshphome the container)

closed 2024-09-07 by ssieb