esphome/issues #2671: ERROR Error binary size: Error: ESP has been flashed with wrong flash size. Please choose the correct 'board' option (esp01_1m always works) and then flash over USB.

curt7000, 2021-11-02:
### The problem

I have a Sonoff TH16 that has been running well for 3yrs.  Whenever I update ESPHome, I've never encountered a problem OTA.

Just updated to 2021.10.3 and I am unable to update OTA.  I receive the error;

ERROR Error binary size: Error: ESP has been flashed with wrong flash size. Please choose the correct 'board' option (esp01_1m always works) and then flash over USB.

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

Not Related

### What platform are you using?

ESP8266

### Board

sonoff basic

### Component causing the issue

OTA

### Example YAML snippet

```yaml
esphome:
  name: sv_controller
  platform: ESP8266
  board: esp8285
  on_boot:
    priority: -10
    then:
      - output.turn_on: gpio_powerled

wifi:
  ssid: "Wireless"
  password: "pwd"

time:
  - platform: homeassistant
    id: homeassistant_time

# Power Status
output:
  - platform: gpio
    pin: GPIO13
    inverted: true
    id: gpio_powerled

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: 'pwd'

ota:
  password: 'pwd'

binary_sensor:
  - platform: status
    name: "Shed Ventilation Controller Status"
  - platform: gpio
    pin:
      number: GPIO0
      inverted: True
    id: manualshedfan
    on_press:
      then:
        - switch.toggle: shedfan

text_sensor:
  - platform: version
    name: "Shed Ventilation Controller Version"
  - platform: wifi_info
    ip_address:
      name: "Shed Ventilation Controller IP Address"
    bssid:
      name: "Shed Ventilation Controller BSSID"

sensor:
  - platform: uptime
    name: "Shed Ventilation Controller Uptime"
  - platform: wifi_signal
    name: "Shed Ventilation Controller RSSI"
    update_interval: 60s
  - platform: dht
    pin: GPIO14
    temperature:
      name: "Shed Temperature"
#      filters:
#       - offset: 0.8
      id: shedtemp
      on_value_range:
        - above: 28.0
          then:
            - switch.turn_on: shedfan
        - below: 27.0
          then:
            - switch.turn_off: shedfan
    humidity:
      name: "Shed Humidity"
#      filters:
#        - offset: 0.0
      id: shedhum
    model: SI7021
    update_interval: 60s

switch:
  - platform: restart
    name: "Shed Ventilation Controller Restart"
  - platform: gpio
    pin: GPIO12
    name: "Shed Ventilation Fan"
    id: shedfan
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /home/homeassistant/.homeassistant/esphome/config/sv_controller.yaml...
WARNING 'sv_controller': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
INFO Detected timezone 'Etc/UTC'
INFO Generating C++ source...
INFO Compiling app...
Processing sv_controller (board: esp8285; framework: arduino; platform: platformio/espressif8266 @ 2.6.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 1MB Flash
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/sv_controller/src/main.cpp.o
Linking .pioenvs/sv_controller/firmware.elf
Building .pioenvs/sv_controller/firmware.bin
RAM:   [====      ]  37.3% (used 30568 bytes from 81920 bytes)
Flash: [====      ]  37.6% (used 385096 bytes from 1023984 bytes)
========================= [SUCCESS] Took 4.90 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of sv_controller.local
INFO  -> 10.0.1.62
INFO Uploading /home/homeassistant/.homeassistant/esphome/config/sv_controller/.pioenvs/sv_controller/firmware.bin (389248 bytes)
ERROR Error binary size: Error: ESP has been flashed with wrong flash size. Please choose the correct 'board' option (esp01_1m always works) and then flash over USB.
```

smoke007, 2022-01-22:
Try changing the following in your config:

From:
esphome:
  name: sv_controller
  platform: ESP8266
  board: esp8285

To:
esphome:
  name: sv_controller
  platform: ESP8266
  board: esp01_1m

curt7000, 2022-01-23:
Unfortunately, it didn’t work.  I’ve tried compiling and updating on ESPHOME 2022.01.0 as well.

baerrs, 2022-03-14:
I'm having the same issue... have you found the fix for this ?

curt7000, 2022-03-14:
> I'm having the same issue... have you found the fix for this ?

Nothing has worked OTA.  Each time I update ESPHome to a new release I hope it will resolve, but no luck.  I need to take it down and manually update it.  But, I t’s in a shed, up in the rafters, as a ventilation fan controller, so I’ve been too busy/lazy to do it.