esphome/issues #2672: Alerting the st7789v component - copying to custom_components

scargill, 2021-11-02:
### The problem

I've a problem (I'm not alone) with the ST7789v component - NOT using a TTGO display but a full 320*240 version. The CPP and H files sit in esphome/myproject/ src/esphome/components/st7789v folder but of course you can't alter that.  SO I tried copying that folder into esphome/custom_components/ folder and without ANY mods the compilation fails. This is in Windows not Linux so I'm not expecting permissions issues. If I BIN the new folder, compilation works again.

I THINK the answer for the ST7789v is to change width and height variables starting at line 222, but I simply cannot find a way to alter the source file.

### Which version of ESPHome has the issue?

2021.9.1

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.9.1

### What platform are you using?

ESP32

### Board

nodemcu-32s

### Component causing the issue

st7789v

### Example YAML snippet

```yaml
The issue is making a change as above - I'm sure the change will fix the issue (or one of them - the colours are also reversed on a standard 320*240 st7789v display compared to TTGO) but I'm not able to change the source. HELP.
```

### Anything in the logs that might be useful for us?

```txt
I simply get a compile failure -   Platform not found: 'display.st7789v'.
```

OttoWinter, 2021-11-03:
You need to copy the folder from https://github.com/esphome/esphome/tree/dev/esphome/components/st7789v. The code in the generated project only contains the generated h/cpp source files, not the python files that validate/generate the code.

scargill, 2021-11-03:
I did and thank ytou very much - that compiled - sadly it did not run. I changed offsets and instead of a centred TTGO display I move it to the top right (no idea why not top left) - BUT when I changed the width and height - all I got out was a white flashing screen instead of my original display. See image - the code was intended for the TTGO display which is 240*135 instad of the more normal 320*240. 
[image]

scargill, 2021-11-03:
So, with the full folder as advised in custom_components - I can move that display up and right.... by altering st7789v.cpp and re-compiling and running - no problem..

fraxinas, 2022-02-24:
so i've worked on this in my personal branch https://github.com/fraxinas/esphome/tree/fix_st7789v because i have a 240x240 display. however i found out that this only works when using the `esp-idf` platform instead of arduino.
with this a lot of other things that i need don't work unfortunately.
when i originally worked on this in April 2021, this didn't need esp-idf.