esphome/issues #2673: ESP32-C3 only GPIO1 can receive ir or rf code

liwei19920307, 2021-11-03:
### The problem

I want to design an all-in-one module, based on eps-c3-13u (ESP32-C3), which supports both infrared and radio frequency transmission and reception. I tested GPIO0~5, 9~10, 18~19 after several tests, only GPIO1 can receive the message

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.10.7

### What platform are you using?

ESP32

### Board

esp_c3_13u

### Component causing the issue

remote_receiver

### Example YAML snippet

```yaml
esphome:
  name: esp32_c3_test
  platform: ESP32
  board: esp32-c3-devkitm-1
  platformio_options:
    platform: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
    platform_packages:
      - framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.0
    board_build.variant: esp32c3
    board_build.f_cpu: 160000000L
    board_build.f_flash: 40000000L
    upload_protocol: esptool
    board_build.flash_mode: dio

remote_receiver:
  - id: esp32_c3_test_rr_0
    pin:
      number: 0
    dump:
      - rc_switch
    tolerance: 50%
    filter: 250us
    idle: 4ms
    buffer_size: 2kb
  - id: esp32_c3_test_rr_1
    pin:
      number: 1
    dump:
      - raw
  - id: esp32_c3_test_rr_2
    pin:
      number: 2
    dump:
      - rc_switch
    tolerance: 50%
    filter: 250us
    idle: 4ms
    buffer_size: 2kb
  - id: esp32_c3_test_rr_3
    pin:
      number: 3
    dump:
      - rc_switch
    tolerance: 50%
    filter: 250us
    idle: 4ms
    buffer_size: 2kb
  - id: esp32_c3_test_rr_4
    pin:
      number: 4
    dump:
      - rc_switch
    tolerance: 50%
    filter: 250us
    idle: 4ms
    buffer_size: 2kb
  - id: esp32_c3_test_rr_5
    pin:
      number: 5
    dump:
      - rc_switch
    tolerance: 50%
    filter: 250us
    idle: 4ms
    buffer_size: 2kb
  - id: esp32_c3_test_rr_9
    pin:
      number: 9
    dump:
      - rc_switch
    tolerance: 50%
    filter: 250us
    idle: 4ms
    buffer_size: 2kb
  - id: esp32_c3_test_rr_10
    pin:
      number: 10
    dump:
      - rc_switch
    tolerance: 50%
    filter: 250us
    idle: 4ms
    buffer_size: 2kb
  - id: esp32_c3_test_rr_18
    pin:
      number: 18
    dump:
      - rc_switch
    tolerance: 50%
    filter: 250us
    idle: 4ms
    buffer_size: 2kb
  - id: esp32_c3_test_rr_19
    pin:
      number: 19
    dump:
      - rc_switch
    tolerance: 50%
    filter: 250us
    idle: 4ms
    buffer_size: 2kb
```

### Additional information

only GPIO1 can receive the message

glmnet, 2021-11-06:
Try defining only two receivers as you only need two (rf and ir). Esp32 cannot have a remote receiver on each pin all at the same time. 

Also there is a bug in esp32 receiver I'm fixing it soon

liwei19920307, 2021-11-07:
> Try defining only two receivers as you only need two (rf and ir). Esp32 cannot have a remote receiver on each pin all at the same time.
> 
> Also there is a bug in esp32 receiver I'm fixing it soon

Thanks Reply.  I tried to set it separately, but only GPIO1 is still valid. Continue to test after you fix it.

liwei19920307, 2021-11-10:
1. After several tests, I found a strange phenomenon. Only GPIO1 can not receive anything, but if you add any receiver in front of GPIO1, GPIO1 can work and receive messages, but any other GPIO cannot receive anything.
2. Receiving and sending cannot work at the same time, the order of configuration will have different results
3. The following are the test results of 7 different configurations

````
esphome:
  name: s6in1_demo
  platform: ESP32
  board: esp32-c3-devkitm-1
  platformio_options:
    platform: https://github.com/platformio/platform-espressif32.git#feature/arduino-upstream
    platform_packages:
      - framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.0
    board_build.variant: esp32c3
    board_build.f_cpu: 160000000L
    board_build.f_flash: 40000000L
    upload_protocol: esptool
    board_build.flash_mode: dio

wifi:
  ssid: "XXX"
  password: "XXX"

captive_portal:

logger:

api:
  password: 'XXX'

ota:
  password: 'XXX'

web_server:
  port: 80

````

#1 can receive

````
remote_receiver:
  - id: esp32_c3_test_rr_0
    pin:
      number: 0
      inverted: yes
    dump: 
      raw
  - id: esp32_c3_test_rr_1
    pin:
      number: 1
      inverted: yes
````

#2 can receive

````
remote_receiver:
  - id: esp32_c3_test_rr_12
    pin:
      number: 12
      inverted: yes
    dump: 
      raw
  - id: esp32_c3_test_rr_1
    pin:
      number: 1
      inverted: yes
````

#3 cannot receive anything

````
remote_receiver:
  - id: esp32_c3_test_rr_1
    pin:
      number: 1
      inverted: yes
    dump: 
      raw
  - id: esp32_c3_test_rr_12
    pin:
      number: 12
      inverted: yes
    dump: 
      raw
````

#4 cannot receive anything

````
remote_receiver:
  - id: esp32_c3_test_rr_1
    pin:
      number: 1
      inverted: yes
    dump: 
      raw
````

#5 cannot receive anything

````
remote_receiver:
  pin:
    number: 1
    inverted: yes
  dump: 
    raw
````

#6 rt work rr cannot receive anything

````
remote_transmitter:
  pin: 2
  carrier_duty_percent: 50%

remote_receiver:
  - id: esp32_c3_test_rr_12
    pin:
      number: 12
      inverted: yes
    dump: 
      raw
  - id: esp32_c3_test_rr_1
    pin:
      number: 1
      inverted: yes
    dump: 
      raw

switch:
  - platform: template
    name: s6in1_demo_ir
    turn_on_action:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38kHz
          code: [9054, -4491, 582, ...]
````

#7 rt not work rr can receive

````
remote_receiver:
  - id: esp32_c3_test_rr_12
    pin:
      number: 12
      inverted: yes
    dump: 
      raw
  - id: esp32_c3_test_rr_1
    pin:
      number: 1
      inverted: yes
    dump: 
      raw
      
remote_transmitter:
  pin: 2
  carrier_duty_percent: 50%
  
switch:
  - platform: template
    name: s6in1_demo_ir
    turn_on_action:
      - remote_transmitter.transmit_raw:
          carrier_frequency: 38kHz
          code: [9054, -4491, 582, ...]
````

liwei19920307, 2021-11-18:
> 1. After several tests, I found a strange phenomenon. Only GPIO1 can not receive anything, but if you add any receiver in front of GPIO1, GPIO1 can work and receive messages, but any other GPIO [...]

ESPHome 2021.11.1 is not resolved

rpatel3001, 2022-01-14:
if anyone comes across this, this may be related to my issue #2934. TLDR, the ESP32C3 RMT peripheral channels 0 and 1 can only transmit and channels 2 and 3 can only receive. If you need a receiver, you need two dummy transmitters created first to block the TX only channels (or maybe only one dummy transmitter, based on the above tests).

ssieb, 2022-09-16:
I expect this is a duplicate, but I'll leave it open until a fix for the other issue can be tested.

rpatel3001, 2024-04-23:
likely closed by https://github.com/esphome/esphome/pull/6497

ssieb, 2024-04-23:
Yes.  No logs have been included, so I assume that's it.

closed 2024-04-23 by ssieb