esphome/issues #2674: ESPHome Time Component - .timestamp not  updating

behnam85, 2021-11-03:
### The problem

Time component .timestamp not  updating.
The .second is working fine.
The .timestamp is working fine on ESP_LOGI and .timestamp updated.
 The .timestamp only updates every 128 seconds

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.10.7

### What platform are you using?

ESP32

### Component causing the issue

time

### Example YAML snippet

```yaml
substitutions:
  
  
  textname: templatexx
#globals:
#   - id: template_sens
#     type: float  
#    restore_value: no
#    initial_value: '000000000000000' 
#   - id: my_vmin_int
#     type: float
#     restore_value: no
#     initial_value: '250'   

esphome:
  name: esp321
  platform: ESP32
  board: lolin32

wifi:
  ssid: "Mi wifi"
  password: "behnam123"
  
  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esp321 Fallback Hotspot"
    password: "coCKxLsID1nI"

captive_portal:

# Enable logging
logger: 
  level: debug

# 6Enable Home Assistant API
mqtt:

  broker: 192.168.2.31
  username: homeassistant
  password: pho7weeN3Yie9ohNgoh8yeom5ou5iNgeichiNgaeheeneR1auhae8Vief6oophut
[... 221 lines cut ...]
[11:31:20][D][gps:040]: Location:
[11:31:20][D][gps:041]:   Lat: ****
[11:31:20][D][gps:042]:   Lon: ****
[11:31:20][D][gps:047]: Speed:
[11:31:20][D][gps:048]:   0.203720 km/h
[11:31:20][D][gps:052]: Course:
[11:31:20][D][gps:053]:   0.000000 °
[11:31:20][I][gps:074]: 11:31:20:1635926480
[11:31:20][D][sensor:113]: 'Template Sensor': Sending state 1635926528.00000  with 0 decimals of accuracy
[11:31:20][D][text_sensor:067]: 'Template Text Sensor': Sending state '2021-11-03 11:31:20'
[11:31:20][D][gps:040]: Location:
[11:31:20][D][gps:041]:   Lat: ****
[11:31:20][D][gps:042]:   Lon: ****
[11:31:20][D][gps:057]: Altitude:
[11:31:20][D][gps:058]:   1330.800049 m
[11:31:20][D][gps:062]: Satellites:
[11:31:20][D][gps:063]:   7
[11:31:20][D][sensor:113]: 'Current time stamp': Sending state 1635926528.00000  with 1 decimals of accuracy
[11:31:20][D][sensor:113]: 'Current time': Sending state 20.00000  with 1 decimals of accuracy
[11:31:20][D][sensor:113]: 'Latitude': Sending state 38.43681 ° with 6 decimals of accuracy
```

### Additional information

[image]

OttoWinter, 2021-11-03:
Try sending the value to a text sensor instead.

Sensor encodes the value as a float, likely the timestamp is large enough that floats can't represent the number accurately anymore.

Jpsy, 2021-11-03:
I just worked on that same problem. 
It is definitely the result of the fact that publish_state uses a float input parameter and the 4 byte floats have not enough resolution to store all digits of a Unix epoche value measured in milliseconds. The quantization error is about +/- 64 seconds on the values that appear in HA.

A text_sensor is no replacement as it accepts no device_class.
In HA this will only be a string, never a timestamp.

I see two options, both being a feature request:
1. Add device_class properties to text_sensors. From all possilble device classes I currently see "date" and "timestamp" as possible targets for this feature.
2. Add a seconds publish_state method to sensors that accepts doubles.

If there is any chance that one of these options could be implemented I would be happy to issue a feature request. Could you give me some feedback.

Jpsy, 2021-11-03:
Related to #1099

oxan, 2021-11-03:
> Sensor encodes the value as a float, likely the timestamp is large enough that floats can't represent the number accurately anymore.

Yes, this is the problem, we should probably add a note about it to the documentation though, not the first report of this (#1099).

OttoWinter, 2021-11-03:
> > Sensor encodes the value as a float, likely the timestamp is large enough that floats can't represent the number accurately anymore.
> 
> Yes, this is the problem, we should probably add a note about it to the documentation though, not the first report of this (#1099).

Yeah probably we should even remove `timestamp` support from sensor completely (it's the only instance i know of where the values get that high) when we have device class support for text sensors (and perhaps integration in the uptime integration as well)

Jpsy, 2021-11-03:
> Yeah probably we should even remove `timestamp` support from sensor completely...

That would be a breaking change but much cleaner. 
The sensor for timestamps will probably continue to confuse users  because of the quantization effects.

tungmeister, 2022-01-10:
Is there a solution to this? I'm currently rendering a progress bar on a display, I'm calculating the position of the progress bar using the media_duration, media_position and media_position_updated_at (cast to a float) attributes from a media_player. because the timestamp is only updating every couple of minutes the position of the progress bar only updates comparably and I'd rather it be much more granular.

tungmeister, 2022-01-12:
For anyone else having this issue, if you cast the timestamp to an int or double it will update correctly e.g:
````
static_cast<double>(id(esptime).timestamp_now())
````

akomelj, 2022-01-25:
> Yeah probably we should even remove `timestamp` support from sensor completely (it's the only instance i know of where the values get that high) when we have device class support for text sensors (and perhaps integration in the uptime integration as well)

Is anyone already working on the device class implementation for text sensors? I can implement underlying functionality (adding configuration key to schema & validations & publishing the setting to discovery topics) and submit a PR, however - before jumping in - I would just like to check if devs have bigger plans for this or someone is already working on it? Someone would probably still need to go through existing text sensor integrations and classify them accordingly afterwards...

mbuzina, 2022-03-25:
The HASS API changed to allow python object (native objects). Maybe the best solution would be a dedicated sensor_timestamp which returns a python dateteime object to home assistant, taking in an esphome::time::ESPTime?

Snip from https://developers.home-assistant.io/docs/core/entity/sensor/#available-device-classes
<html>
<body>

timestamp |   | Timestamp. Requires native_value to return a Python datetime.datetime object, with time zone information, or None.
-- | -- | --

</body>
</html>

fakuivan, 2023-10-28:
Any updates on this? It feels weird for the solution to be switching to text sensors. Are floats hardcoded into the filter code? Could templates be used for this?

ssieb, 2023-10-28:
Nothing to do with filters.  An esphome `sensor:` of whatever type holds a float value and the API to HA transfers a float value.

fakuivan, 2023-10-28:
Thanks for clarifying, mind pointing to the code that defines how the sensor values are sent? How hard would it be to make it accept doubles or integral types too?

ssieb, 2023-10-29:
It's in the protobuf I guess.  But the state value that the sensor holds is a float, so you can't get past that.  If you wanted to change it to other types, then you would have to modify everything about how sensors work and change the api protocol all the way to the HA integration.

dougiteixeira, 2024-02-10:
I created a PR to add support for the TIMESTAMP and DATE device classes in the text sensor, reflecting in the API and MQTT: https://github.com/esphome/esphome/pull/6202

After starting the PR review, I will create the PR for the changes to the `aioesphomeapi` package and the ESPHome integration in Home Assistant, the changes are ready and tested locally.