esphome/issues #2675: GPIO16 error for ESP8266

sthorley, 2021-11-03:
### The problem

Hi,

It seems that the validation introduced in PR [2572](https://github.com/esphome/esphome/pull/2572) broke my deployed config for [Deta Grid Connect Switches](https://blog.mikejmcguire.com/2020/05/22/deta-grid-connect-3-and-4-gang-light-switches-and-home-assistant/). I was using the [template here: ](https://www.esphome-devices.com/devices/DETA-Grid-Connect-Smart-Switch) which was certainly working. I have had it deployed for a year with all switches working so I am certain this switch is supported and works using GPIO16 in the way it's coded. See the excerpt below:

`# Buttons
binary_sensor:
  # 1st button
  - platform: gpio
    pin:
      number: GPIO16
      mode: INPUT_PULLUP
      inverted: True
    name: "${friendly_name} 1st Button"
    #toggle relay on push
    on_press:
      - light.toggle: light1`

I assume this might be caused by an issue with the template rather than ESPhome but am unsure how to fix it and I note the change was marked "bugfix/non breaking" when it has in fact broken this template.

The error given is "    GPIO Pin 16 does not support pullup pin mode. Please choose another pin.
    mode: INPUT_PULLUP "

Thanks for any assistance

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.10.

### What platform are you using?

ESP8266

### Board

esp01_1m

### Component causing the issue

gpio

### Example YAML snippet

```yaml
`# Buttons
binary_sensor:
  # 1st button
  - platform: gpio
    pin:
      number: GPIO16
      mode: INPUT_PULLUP
      inverted: True
    name: "${friendly_name} 1st Button"
    #toggle relay on push
    on_press:
      - light.toggle: light1`
```

### Anything in the logs that might be useful for us?

```txt
GPIO Pin 16 does not support pullup pin mode. Please choose another pin.
    mode: INPUT_PULLUP
```

OttoWinter, 2021-11-03:
It's not a breaking change if validation just gets stricter for something that's not working correctly.

ESP8266s have no internal pullup resistors on pin GPIO16. I think in your case the INPUT_PULLUP just acted _like_ INPUT (but that would have been undefined behavior and the arduino framework could have changed those semantics at any time)

-> change the mode to INPUT

sthorley, 2021-11-04:
Thanks Otto, Replacing the "INPUT_PULLUP" with simply "INPUT" worked!

I really appreciate the work you've done making ESPhome, it's an amazing framework that I'm learning more and more about every month. Thank you

closed 2022-04-11 by github-actions[bot]