esphome/issues #2676: rgbww light integration mqtt communication broken with HA

dylan09, 2021-11-03:
### The problem

When using MQTT to communicate between my ESP RGBWW light with Home Assistant I get errors and warnings in HA log file. Errors and warnings saying `state topic` and `color_mode` are not properly configured or contain wrong values.

And, maybe related to this,  the icon color in HA isn't correctly changed when switching from color to white light (color temperature). When light is in RGB mode the icon shows correct color. After changing to color temperature and setting light to white the icon stays on old rgb color.

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.10.6

### What platform are you using?

ESP8266

### Board

esp_wroom_02

### Component causing the issue

rgbww

### Example YAML snippet

```yaml
substitutions:
   node_name: xx-li-wiz2
   id_prefix: xx_li_wiz2
   name_prefix: xx_li_wiz2
 
packages:
  wifi: !include wifi.yaml
  web_server: !include web_server.yaml
  mqtt:  !include mqtt.yaml
  time:  !include time.yaml
  sensor: !include def_sensor.yaml
  text_sensor: !include def_text_sensor.yaml

esphome:
  name: ${node_name}
  platform: ESP8266
  board: esp_wroom_02
  on_boot:
    priority: -10
    # ...
    then:
      - light.turn_off: ${id_prefix}

captive_portal:

logger:
  level: DEBUG

ota:
  password: !secret passwd

output:
  - platform: esp8266_pwm
    id: out_red
    pin: GPIO15
    frequency: 200Hz
  - platform: esp8266_pwm
    id: out_green
    pin: GPIO12
    frequency: 200Hz
  - platform: esp8266_pwm
    id: out_blue
    pin: GPIO14
    frequency: 200Hz
  - platform: esp8266_pwm
    id: out_warm_white
    pin: GPIO4
    frequency: 1000Hz
  - platform: esp8266_pwm
    id: out_cold_white
    pin: GPIO13
    frequency: 1000Hz

light:
  - platform: rgbww
    name: ${name_prefix}
    id: ${id_prefix}
    red: out_red
    green: out_green
    blue: out_blue
    warm_white: out_warm_white
    cold_white: out_cold_white
    cold_white_color_temperature: 6500 K
    warm_white_color_temperature: 2200 K
    color_interlock: true # avoids simultaneous RGB and W/W
    default_transition_length: 500ms
    restore_mode: ALWAYS_OFF
    effects:
      - lambda:
          name: Random
          update_interval: 4.0s
          lambda: |-
            auto call = id(${id_prefix}).turn_on();
            call.set_transition_length(2000);
            Color c = Color::random_color();
            call.set_rgb((c.red)/255.0, (c.green)/255.0, (c.blue)/255.0);
            call.perform();
      - lambda:
          name: "Christmas"
          update_interval: 5.0s
          lambda: |-
            static int state = 0;
            auto call = id(${id_prefix}).turn_on();
            call.set_transition_length(3000);
            if (state == 0) {
              call.set_rgb(1.0, 0.0, 0.0);
            } else if (state == 1) {
              call.set_rgb(0.0, 1.0, 0.0);
            } else if (state == 2) {
              call.set_rgb(1.0, 0.0, 0.0);
            } else {
              call.set_rgb(0.0, 1.0, 0.0);
            }
            call.perform();
            state += 1;
            if (state == 4)
              state = 0;
```

### Anything in the logs that might be useful for us?

```txt
Home Assistant Log after initial boot of light.

2021-11-03 16:16:58 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.mqtt entity: sensor.xx_li_wiz2_ip
2021-11-03 16:16:58 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.mqtt entity: sensor.xx_li_wiz2_version
2021-11-03 16:16:58 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.mqtt entity: sensor.xx_li_wiz2_uptime
2021-11-03 16:16:58 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.mqtt entity: sensor.xx_li_wiz2_vcc
2021-11-03 16:16:58 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new sensor.mqtt entity: sensor.xx_li_wiz2_rssi
2021-11-03 16:16:58 INFO (MainThread) [homeassistant.helpers.entity_registry] Registered new light.mqtt entity: light.xx_li_wiz2
2021-11-03 16:16:59 ERROR (MainThread) [homeassistant.util.logging] Exception in state_received when handling msg on 'esphome/xx_li_wiz2/light/xx_li_wiz2/state': '{"effect":"None","color":{}}'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/mqtt/debug_info.py", line 37, in wrapper
    msg_callback(msg)
  File "/usr/src/homeassistant/homeassistant/components/mqtt/light/schema_json.py", line 299, in state_received
    if values["state"] == "ON":
KeyError: 'state'

And after changing light from RGB to color temperature I see this warning in log file:

2021-11-03 16:18:18 WARNING (MainThread) [homeassistant.components.mqtt.light.schema_json] Invalid color mode received
```

### Additional information

Here the MQTT announcement for the above configuration:
```
{
   "schema": "json",
   "color_mode": true,
   "supported_color_modes": ["color_temp", "rgb"],
   "brightness": true,
   "effect": true,
   "effect_list": ["Random", "Christmas", "None"],
   "name": "xx_li_wiz2",
   "state_topic": "esphome/xx_li_wiz2/light/xx_li_wiz2/state",
   "command_topic": "esphome/xx_li_wiz2/light/xx_li_wiz2/command",
   "availability_topic": "esphome/xx_li_wiz2/status",
   "unique_id": "ESPlightxx_li_wiz2",
   "device": {
      "identifiers": "XXXXXXXXXXXX",
      "name": "xx-li-wiz2",
      "sw_version": "esphome v2021.10.3 Nov  3 2021, 16:16:16",
      "model": "esp_wroom_02",
      "manufacturer": "espressif"
   }
}
```

MQTT state topic after booting light. This MQTT message/state is responsible for the above error message.
```
{
   "effect": "None",
   "color": {}
}
``` 

MQTT state topic after switching from RGB to color temperature. This topic produces the warning in HA log.
``` 
{
   "effect": "None",
   "color_mode": "cwww",
   "state": "ON",
   "brightness": 255,
   "color": {
      "c": 250,
      "w": 255
   }
}
```

In the announce topic `supported_color_modes":["color_temp","rgb"]`  is sent. But the second state topic contains `"color_mode": "cwww"`.

oxan, 2021-11-03:
This is probably #2388.

dylan09, 2021-11-03:
Yes. I think its related. But for me at first start I get an error in HA because of incomplete status message.
And the icon color in HA GUI didn't show the correct color, when light goes from RGB to color temperature (white).

And currently I have to stay with MQTT.

dylan09, 2021-11-06:
I have tried to get some information out of the source code for rgbww light and light call.
For me with `color_interlock` set, the implementation of rgbww light is broken and didn't work without errors in HA.
If I set `color_interlock` to false, the light is working with HA and I get  no more errors and warnings in the logs.
But now I have to set white values using 2 sliders for warm white and cold white. Not very intuitive.

The pre color_mode model implemented until ESPHome 1.20.4 works as expected. I could set my white light using one slider for color temperature and one for brightness. For me its more intuitiv to set a color temperature in kelvin (or mireds) than to have to slider with wich I have to change color temperature step by step tuning cold white and warm white LEDs.

Maybe there is a way to change behavior, so that rgbww with `color_interlock` works either as a RGB light or as a light with color temperature.

dylan09, 2021-11-06:
As a test I have tried to change the definition in `rgbww_light_output.h` line 24 from `light::ColorMode::COLD_WARM_WHITE` to `light::ColorMode::COLOR_TEMPERATURE`.
With that I get the color temperature control and white brightness slider. But the color temperature setting didn't get useful white values with my lights.

dylan09, 2022-03-11:
There is also one small glitch with the discovery message I found a few days ago. Maybe is it possible to add the min and max Mireds value to the discovery telegram?

Or I will try to create a PR later. Unfortunately I am not so experienced with git.