esphome/issues #2677: Logs not displaying in ESPHome UI when using SSL and nginx_proxy on latest HA 2021.11

craigueh, 2021-11-04:
### The problem

Logs are not displaying in the ESPHome UI when using SSL and nginx_proxy on latest HA 2021.11
They worked fine up until I installed the most recent update. if I access my HA over SSL there is no log output visible but if I go back to local IP I can see the logs as norma
[image]
l.

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.0

### What platform are you using?

ESP8266

leonardpitzu, 2022-03-31:
same for me on 2022.03.* and 2202.04 beta

MACscr, 2022-06-27:
Issue still exists with ` Home Assistant Core: 2022.6.7
 Home Assistant Supervisor: 2022.05.3` and last dev version of the addon. I find it odd that there arent more clear instructions for using the official HA ssl proxy with the esphome addon.

mmakaay, 2022-06-30:
The problem lies in the HA ssl proxy, which makes the ESPHome project the wrong target for the issue.
I am running ESPHome behind an nginx reverse proxy without issues. My home lab setup has quite a few services running behind this proxy, and my generic proxy config rules (which work for all my services) contain more than needed for ESPHome, but this is what I use:

**File: `nginx_reverse_proxy.conf` (included from main config)**
```
proxy_http_version 1.1;
proxy_pass_request_headers on;
proxy_cache_bypass $http_upgrade;
proxy_set_header Upgrade $http_upgrade;
proxy_set_header Connection "upgrade";
proxy_set_header Host $http_host;
proxy_set_header Origin "${scheme}://${http_host}";
proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
proxy_set_header X-Real-IP $remote_addr;
proxy_set_header X-Forwarded-Proto $scheme;
proxy_set_header X-Scheme $scheme;
proxy_set_header X-Forwarded-Host $host;
proxy_set_header X-Forwarded-Port $request_port;

# Some apps may require large uploads.
client_max_body_size 0;
```

**Reverse proxy section from main config**
```
upstream upstream-esphome {
        server 192.168.100.173:6052;
    }
    
    server {
        listen 443 ssl http2;
        server_name my.esphome.hostname;
        include /local/nginx_ssl.conf;

        location / {
            proxy_pass http://upstream-esphome;
            include /local/nginx_reverse_proxy.conf;
            allow 192.168.100.0/24; # LAN users
            allow 192.168.102.0/24; # VPN users
            deny all;
        }
    }
```

oddlama, 2023-03-17:
FYI: The issue occurs when using `proxy_set_header Host $host;` instead of `proxy_set_header Host $http_host;`.
I didn't yet investigate why esphome fails when using `$host`, but it should be noted that using `$host` would be preferable since it prevents host spoofing ([more information here](https://github.com/yandex/gixy/blob/master/docs/en/plugins/hostspoofing.md)).

speed47, 2023-04-21:
Damn, thank you @oddlama! This was driving me crazy and your comment nailed it! Replacing `$host` by `$http_host` does fix the issue. This should be documented.

thetxpopulist, 2023-07-23:
can confirm as well, I lost access to view logs when dialing in my nginx config; reverting from `$host` back to `$http_host` was the fix as pointed out by @oddlama , and in the directive:

> This variable may have a different value from $http_host in such cases: 1) when the Host input header is absent or has an empty value, $host equals to the value of server_name directive; 2)when the value of Host contains port number, $host doesn't include that port number. $host's value is always lowercase since 0.8.17.

renatoccosta, 2023-11-22:
For me it worked only when I added these lines:

```
proxy_set_header    Upgrade             $http_upgrade;
proxy_set_header    Connection          "upgrade";
proxy_http_version 1.1;
``` 

Full location block:

```
  location / {
    proxy_pass http://esphome;

        proxy_set_header    X-Real-IP           $remote_addr;
        proxy_set_header    X-Forwarded-For     $proxy_add_x_forwarded_for;
        proxy_set_header    X-Forwarded-Proto   $scheme;
        proxy_set_header    Host                $host;
        proxy_set_header    X-Forwarded-Host    $host;
        proxy_set_header    X-Forwarded-Port    $server_port;
        proxy_set_header    Upgrade             $http_upgrade;
        proxy_set_header    Connection          "upgrade";

        proxy_http_version 1.1;

  }
```

aghosh0605, 2024-05-08:
> For me it worked only when I added these lines:
>
> ```
> proxy_set_header    Upgrade             $http_upgrade;
> proxy_set_header    Connection          "upgrade";
> proxy_http_version 1.1;
> ```
>
> Full [...]

Thanks for the help! Because of this small issue, my update was crashed as I had to stop  without knowing what is actually happening in logs!