esphome/issues #2678: Where is default OTA password stored and how do I change it's value?

timdaman, 2021-11-04:
### The problem

First, thank you for making such a wonderful tool. 

Now to my issue. When I create a new "node" in ESPHome (using the wizard) I am asked the following questions

- What to name the node
- What are my WiFi credentials
- What board "type"

When I look at the generated yaml file for my new configurations I see the OTA password is already set with a block like this
```
ota:
  password: "REDACTED"
```

My question is where is the value stored? I see all new node seem to get the same password but I don't see from where or how to update that password.

I have a OTA password I was using before, it seems like this "default" was added in the last few months. The result is devices that used my older templates have one OTA password and my newer "hand configured" devices have another. 

I am trying to move some of these hand configured devices to templated configuration but the OTA password mismatch is preventing this with out some manual work. 

I spent a hour digging through the code but I couldn't figure it out.

P.S. I found the note on how to reset the OTA password, that is what I am doing to "repair" the mismatch but I would rather ensure it never happens again.

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.10.6

### What platform are you using?

ESP8266

### Board

Sonoff S31, A.K.A. esp01_1m

### Component causing the issue

OTA

oxan, 2021-11-04:
It uses a random generated secret: https://github.com/esphome/esphome/blob/dev/esphome/dashboard/dashboard.py#L368

I don't think all nodes are supposed to get the same password...

timdaman, 2021-11-05:
Well that is embarrassing, indeed. I just tested and it behaves just as you described. 

I must of copied the configurations I was looking at and missed the hard-coded password. Thanks for the help.

closed 2021-11-05 by timdaman