esphome/issues #2680: thermostat component state inverted after reboot

nliaudat, 2021-11-05:
### The problem

The thermostat component state (on/off) is inverted in hassio after esphome reboot.

In fact a thermostat in "auto" mode become "off" and a component in "off" mode become "idle"

### Which version of ESPHome has the issue?

2021.10.0

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.10.6

### What platform are you using?

ESP32

### Board

nodemcu-32s

### Component causing the issue

thermostat

### Example YAML snippet

```yaml
https://github.com/nliaudat/esp32_8ch_motor_shield/blob/main/firmware/esphome_firmware_sample.yaml
```

### Anything in the logs that might be useful for us?

```txt
[I][restart:016]: Restarting device...
[I][app:130]: Rebooting safely...
[D][esp32.preferences:113]: Saving preferences to flash...
...
[I][ota:102]: Boot seems successful, resetting boot loop counter.
[D][esp32.preferences:113]: Saving preferences to flash...
```

### Additional information

Before esp reboot : 

[image]

After esp reboot : 
[image]

nliaudat, 2021-11-05:
If I do an automation in hassio
```
  action:
  - service: climate.set_temperature
    target:
      entity_id: climate.bedroom
    data:
      target_temp_high: 18
      target_temp_low: 16
      hvac_mode: auto
```
esphome report : 
`[W][climate:053]:   Mode AUTO is not supported by this device!`

kbx81, 2021-11-11:
Yeah, we need to disambiguate `auto` and `heat_cool` in this component -- its been on my mind a bit for a while. I'll take a look.

I'm not sure about the state being inverted -- I've not observed this behavior myself. 🤔

kbx81, 2021-11-11:
@nliaudat what happens if you change [this line](https://github.com/nliaudat/esp32_8ch_motor_shield/blob/b295ba7fa26a65aae856cbe2f545bf77897c0d83/firmware/esphome_firmware_sample.yaml#L287) to `default_mode: heat_cool`? I think that's what you want, anyway.

nliaudat, 2021-11-11:
`default_mode: heat_cool` is not "inverted" after reboot.
But : 
1) The state is not saved in memory as expected. 
2) Nor the target temp low/high set trough hassio.

```
[08:02:55][I][restart:016]: Restarting device...
[08:02:55][I][app:130]: Rebooting safely...
[08:02:55][D][esp32.preferences:113]: Saving preferences to flash...
```

kbx81, 2021-11-11:
> The state is not saved in memory as expected.

Can you elaborate? The climate mode _should_ be restored after the reboot, but the climate action will be recalculated. The climate action itself is not restored; this is by design and is not a bug.

> Nor the target temp low/high set through hassio.

Are you requesting a climate preset via hassio? If a preset is activated, the target temperatures are reset as assigned by the preset. Presets store fixed set points to be recalled later; they are not modified as the temperature is changed.

nliaudat, 2021-11-11:
The climate mode (heat_cool or off) is restored to "default_mode" after esp32 reboots.
The documentation says it should restore from its previous value. 
> default_mode (Optional, climate mode): The default climate mode the controller should use if it is unable to restore it from memory

For the temperatures, the preset in hassio is the default one : here _(not away)_

I'll try to remove the "away_config" from climate configuration to see if's better.

nliaudat, 2021-11-12:
I tested without the "away_config" and you are right ! The temperature set survives reboot. One point fixed. Many Thanks.

closed 2022-04-07 by github-actions[bot]