esphome/issues #2681: Compiling error with bme680_bsec sensor platform

lehaleha21, 2021-11-06:
### The problem

I'm using ESP32 with BME680 sensor wit BSEC library.
ESPHome version 2021.9.3 works well. But i can't update my ESP's.
I get compiling error with newer (2021.10.X) versions of ESPHome.
Compiling without   "libraries:     - SPI" causes the error:
```
Compiling /data/bme01/.pioenvs/bme01/src/esphome/components/bme680_bsec/bme680_bsec.cpp.o
In file included from src/esphome/components/bme680_bsec/bme680_bsec.h:10:0,
                 from src/esphome/components/bme680_bsec/bme680_bsec.cpp:3:
/data/bme01/.piolibdeps/bme01/BSEC Software Library/src/bsec.h:44:17: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://platformio.org/lib/search?query=header:SPI.h
*
*************************************************************

 #include "SPI.h"
                 ^
compilation terminated.
*** [/data/bme01/.pioenvs/bme01/src/esphome/components/bme680_bsec/bme680_bsec.cpp.o] Error 1
```

When I added SPI library I got another error:
```
Compiling /data/bme01/.pioenvs/bme01/src/esphome/components/bme680_bsec/bme680_bsec.cpp.o
Compiling /data/bme01/.pioenvs/bme01/src/esphome/components/esp32/core.cpp.o
src/esphome/components/bme680_bsec/bme680_bsec.cpp: In member function 'void esphome::bme680_bsec::BME680BSECComponent::load_state_()':
src/esphome/components/bme680_bsec/bme680_bsec.cpp:349:42: error: request for member 'make_preference' in 'esphome::global_preferences', which is of pointer type 'esphome::ESPPreferences*' (maybe you meant to use '->' ?)
   this->bsec_state_ = global_preferences.make_preference<uint8_t[BSEC_MAX_STATE_BLOB_SIZE]>(hash, true);
                                          ^
src/esphome/components/bme680_bsec/bme680_bsec.cpp:349:65: error: expected primary-expression before '[' token
   this->bsec_state_ = global_preferences.make_preference<uint8_t[BSEC_MAX_STATE_BLOB_SIZE]>(hash, true);
                                                                 ^
src/esphome/components/bme680_bsec/bme680_bsec.cpp:349:99: warning: left operand of comma operator has no effect [-Wunused-value]
   this->bsec_state_ = global_preferences.make_preference<uint8_t[BSEC_MAX_STATE_BLOB_SIZE]>(hash, true);
                                                                                                   ^
Compiling /data/bme01/.pioenvs/bme01/src/esphome/components/esp32/gpio_arduino.cpp.o
*** [/data/bme01/.pioenvs/bme01/src/esphome/components/bme680_bsec/bme680_bsec.cpp.o] Error 1
========================= [FAILED] Took 25.13 seconds =========================

```

### Which version of ESPHome has the issue?

2021.10.X

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.11.1

### What platform are you using?

ESP32

### Board

ESP32-WROOM-32

### Component causing the issue

bme680_bsec

[... 138 lines cut ...]
Compiling /data/bme01/.pioenvs/bme01/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/bme01/.pioenvs/bme01/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/bme01/.pioenvs/bme01/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/bme01/.pioenvs/bme01/src/esphome/components/api/api_server.cpp.o
Compiling /data/bme01/.pioenvs/bme01/src/esphome/components/api/list_entities.cpp.o
Compiling /data/bme01/.pioenvs/bme01/src/esphome/components/api/proto.cpp.o
Compiling /data/bme01/.pioenvs/bme01/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/bme01/.pioenvs/bme01/src/esphome/components/api/user_services.cpp.o
Compiling /data/bme01/.pioenvs/bme01/src/esphome/components/api/util.cpp.o
Compiling /data/bme01/.pioenvs/bme01/src/esphome/components/bme680_bsec/bme680_bsec.cpp.o
Compiling /data/bme01/.pioenvs/bme01/src/esphome/components/esp32/core.cpp.o
src/esphome/components/bme680_bsec/bme680_bsec.cpp: In member function 'void esphome::bme680_bsec::BME680BSECComponent::load_state_()':
src/esphome/components/bme680_bsec/bme680_bsec.cpp:349:42: error: request for member 'make_preference' in 'esphome::global_preferences', which is of pointer type 'esphome::ESPPreferences*' (maybe you meant to use '->' ?)
   this->bsec_state_ = global_preferences.make_preference<uint8_t[BSEC_MAX_STATE_BLOB_SIZE]>(hash, true);
                                          ^
src/esphome/components/bme680_bsec/bme680_bsec.cpp:349:65: error: expected primary-expression before '[' token
   this->bsec_state_ = global_preferences.make_preference<uint8_t[BSEC_MAX_STATE_BLOB_SIZE]>(hash, true);
                                                                 ^
src/esphome/components/bme680_bsec/bme680_bsec.cpp:349:99: warning: left operand of comma operator has no effect [-Wunused-value]
   this->bsec_state_ = global_preferences.make_preference<uint8_t[BSEC_MAX_STATE_BLOB_SIZE]>(hash, true);
                                                                                                   ^
Compiling /data/bme01/.pioenvs/bme01/src/esphome/components/esp32/gpio_arduino.cpp.o
*** [/data/bme01/.pioenvs/bme01/src/esphome/components/bme680_bsec/bme680_bsec.cpp.o] Error 1
========================= [FAILED] Took 25.13 seconds =========================
```

mmakaay, 2021-11-06:
With `xiaomi_lywsd03mmc` commented out and a `wifi` entry added, I can compile the code just fine.
Can you try doing a clean build? Possibly there are stale source files in your build directory, causing the issue.

lehaleha21, 2021-11-06:
> With `xiaomi_lywsd03mmc` commented out and a `wifi` entry added, I can compile the code just fine. Can you try doing a clean build? Possibly there are stale source files in your build directory, causing the issue.

LYWSD03MMC disabled. Build files cleaned.
No success ((((

mmakaay, 2021-11-06:
Then what *exact* version of ESPHome are you using currently?
There was an issue with SPI not being included for bme680_bsec, but that was fixed in: https://github.com/esphome/esphome/pull/2516
Possibly you are using a version that was not patched for this yet.

When we know the exact version, instead of 2021.10.X, then we can try an exactly identical build.

I've seen cases where a clean build wasn't clean after all. To test this, can you copy the contents of your YAML config to a new file, and change the esphome name in the new YAML file. That will force a guaranteed fresh build (which on my system works with ESPHome dev).

lehaleha21, 2021-11-06:
Current version in the ESP32-WROOM-32 module: ESPHome version 2021.9.3 compiled on Oct 13 2021 (this is latest version wich compiled successfuly).
After updating ESPHome Home Assistant Add-On to 2021.10.1 I got the error "fatal error: SPI.h: No such file or directory"
I added "libraries:    - SPI" in my yaml file and got compilation  error.
Updating ESPHome Home Assistant Add-On to 2021.10.2 - the same result
Updating ESPHome Home Assistant Add-On to 2021.10.3 (current) - the same result
Thats why i wrote 2021.10.X

Right now i connect ESP32 module to the host with Home Assistant via USB.
Create yaml with new name and try to install using USB connection.
With no result (((
```
INFO Reading configuration /config/esphome/bme-01.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing bme-01 (board: esp32dev; framework: arduino; platform: espressif32 @ 3.3.2)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Library Manager: Installing BSEC Software Library @ 1.6.1480

Unpacking  [------------------------------------]    0%
Unpacking  [------------------------------------]    2%
Unpacking  [#-----------------------------------]    5%
Unpacking  [##----------------------------------]    8%
Unpacking  [###---------------------------------]   10%
Unpacking  [####--------------------------------]   13%
Unpacking  [#####-------------------------------]   16%
Unpacking  [######------------------------------]   18%
Unpacking  [#######-----------------------------]   21%
Unpacking  [########----------------------------]   24%
Unpacking  [#########---------------------------]   27%
Unpacking  [##########--------------------------]   29%
Unpacking  [###########-------------------------]   32%
Unpacking  [############------------------------]   35%
Unpacking  [#############-----------------------]   37%
Unpacking  [##############----------------------]   40%
Unpacking  [###############---------------------]   43%
Unpacking  [################--------------------]   45%
Unpacking  [#################-------------------]   48%
Unpacking  [##################------------------]   51%
Unpacking  [###################-----------------]   54%
Unpacking  [####################----------------]   56%
Unpacking  [#####################---------------]   59%
Unpacking  [######################--------------]   62%
Unpacking  [#######################-------------]   64%
Unpacking  [########################------------]   67%
Unpacking  [#########################-----------]   70%
Unpacking  [##########################----------]   72%
Unpacking  [###########################---------]   75%
Unpacking  [############################--------]   78%
Unpacking  [#############################-------]   81%
Unpacking  [##############################------]   83%
Unpacking  [###############################-----]   86%
Unpacking  [################################----]   89%
Unpacking  [#################################---]   91%
Unpacking  [##################################--]   94%
Unpacking  [###################################-]   97%
Unpacking  [####################################]  100%
Library Manager: BSEC Software Library @ 1.6.1480 has been installed!
Dependency Graph
|-- <SPI> 1.0
|-- <WiFi> 1.0
|-- <Update> 1.0
|-- <Wire> 1.0.1
|-- <BSEC Software Library> 1.6.1480
|-- <ESPmDNS> 1.0
Compiling /data/bme-01/.pioenvs/bme-01/src/esphome/components/api/api_connection.cpp.o
Compiling /data/bme-01/.pioenvs/bme-01/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/bme-01/.pioenvs/bme-01/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/bme-01/.pioenvs/bme-01/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/bme-01/.pioenvs/bme-01/src/esphome/components/api/api_server.cpp.o
Compiling /data/bme-01/.pioenvs/bme-01/src/esphome/components/api/list_entities.cpp.o
Compiling /data/bme-01/.pioenvs/bme-01/src/esphome/components/api/proto.cpp.o
Compiling /data/bme-01/.pioenvs/bme-01/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/bme-01/.pioenvs/bme-01/src/esphome/components/api/user_services.cpp.o
Compiling /data/bme-01/.pioenvs/bme-01/src/esphome/components/api/util.cpp.o
Compiling /data/bme-01/.pioenvs/bme-01/src/esphome/components/bme680_bsec/bme680_bsec.cpp.o
Compiling /data/bme-01/.pioenvs/bme-01/src/esphome/components/esp32/core.cpp.o
Compiling /data/bme-01/.pioenvs/bme-01/src/esphome/components/esp32/gpio_arduino.cpp.o
src/esphome/components/bme680_bsec/bme680_bsec.cpp: In member function 'void esphome::bme680_bsec::BME680BSECComponent::load_state_()':
src/esphome/components/bme680_bsec/bme680_bsec.cpp:349:42: error: request for member 'make_preference' in 'esphome::global_preferences', which is of pointer type 'esphome::ESPPreferences*' (maybe you meant to use '->' ?)
   this->bsec_state_ = global_preferences.make_preference<uint8_t[BSEC_MAX_STATE_BLOB_SIZE]>(hash, true);
                                          ^
src/esphome/components/bme680_bsec/bme680_bsec.cpp:349:65: error: expected primary-expression before '[' token
   this->bsec_state_ = global_preferences.make_preference<uint8_t[BSEC_MAX_STATE_BLOB_SIZE]>(hash, true);
                                                                 ^
src/esphome/components/bme680_bsec/bme680_bsec.cpp:349:99: warning: left operand of comma operator has no effect [-Wunused-value]
   this->bsec_state_ = global_preferences.make_preference<uint8_t[BSEC_MAX_STATE_BLOB_SIZE]>(hash, true);
                                                                                                   ^
*** [/data/bme-01/.pioenvs/bme-01/src/esphome/components/bme680_bsec/bme680_bsec.cpp.o] Error 1
========================= [FAILED] Took 24.87 seconds =========================
```

mmakaay, 2021-11-06:
In your error messages, there is this:
```
this->bsec_state_ = global_preferences.make_preference<uint8_t[BSEC_MAX_STATE_BLOB_SIZE]>(hash, true);
```

In the source code for 2021.10.3, there is only this (on a very different line number):
```
this->bsec_state_ = global_preferences->make_preference<uint8_t[BSEC_MAX_STATE_BLOB_SIZE]>(hash, true);
```

Your version has an object (and there's a dot before the `make_preference(...)` method, while the current version of ESPHome uses an object reference (and there's an `->` arrow before that method). This has been the case for all the versions of 2021.10.X, including the first 2021.10.0b1 beta release.
The code with a dot can be found in 2021.9.X.

So I don't know what's going on exactly, but I am pretty sure that your code is not being compiled now using 2021.10.3, since the failing source code simply is not in the code base.

azrael783, 2021-12-06:
I do have the same problem. I'm using a D1 mini and esphome 2021.11.4 as a standalone docker container and I get this error message:

```
INFO Reading configuration /config/wemos_bme680.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing wemos-bme680-lux (board: d1_mini; framework: arduino; platform: platformio/espressif8266 @ 2.6.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|-- <ESP8266WiFi> 1.0
|-- <Wire> 1.0
|-- <BSEC Software Library> 1.5.1474+sha.179de71
|-- <ESP8266mDNS> 1.2
Compiling .pioenvs/wemos-bme680-lux/src/esphome/components/bme680_bsec/bme680_bsec.cpp.o
Compiling .pioenvs/wemos-bme680-lux/src/esphome/components/esp8266/core.cpp.o
In file included from src/esphome/components/bme680_bsec/bme680_bsec.h:10:0,
                 from src/esphome/components/bme680_bsec/bme680_bsec.cpp:3:
.piolibdeps/wemos-bme680-lux/bsec/src/bsec.h:45:17: fatal error: SPI.h: No such file or directory

*************************************************************
* Looking for SPI.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SPI.h"
* Web  > https://platformio.org/lib/search?query=header:SPI.h
*
*************************************************************

 #include "SPI.h"
                 ^
compilation terminated.
Compiling .pioenvs/wemos-bme680-lux/src/esphome/components/esp8266/gpio.cpp.o
Compiling .pioenvs/wemos-bme680-lux/src/esphome/components/esp8266/preferences.cpp.o
*** [.pioenvs/wemos-bme680-lux/src/esphome/components/bme680_bsec/bme680_bsec.cpp.o] Error 1
========================== [FAILED] Took 4.27 seconds ==========================
```

And I am getting the error `[sample rate] is an invalid option for [0]. Please check the indentation` when I uncomment the sample rate.

```yml
bme680_bsec:
    address: 0x77
    temperature_offset: 0
    iaq_mode: static
    sample_rate: lp
    state_save_interval: 6h
```
The Pi4 where the container is running on was completely set up new with RaspiOS Bullseye arm64. But I had both problems also on RaspiOS Buster 32bit (see  [#2073](https://github.com/esphome/issues/issues/2073)).

mmakaay, 2021-12-06:
Did you try a clean build? The ultimate clean build being making a copy of the yaml and changing the esphome/name option in the new config. SPI not being included ought to be fixed in 2021.11.4, so maybe something's going wrong with the build output directory right now.

trvrnrth, 2021-12-06:
@azrael783 Looks like you're pulling in a very old version of the BSEC component (the giveaway is the version of the BSEC Library being pulled in is a temporary fork I had before the necessary fixes were merged upstream in 1.6.1480). You'll need to find where it's coming from and remove it as you want to be using the component directly from core now.

mmakaay, 2021-12-07:
Inlcuded it as an external component maybe? Then simply remove that one to get it fixed.

trvrnrth, 2021-12-07:
The version in use pre-dates external components so it'll be hanging out in `custom_components` and will need removing from there.

azrael783, 2021-12-07:
> Inlcuded it as an external component maybe? Then simply remove that one to get it fixed.

yep, that solved the problem for me. Thank you.

mmakaay, 2021-12-07:
Okay great @azrael783 , thanks for confirming the fix.

Closing this issue now, since the unlying issue was solved a while ago.

closed 2021-12-07 by mmakaay