esphome/issues #2682: Preferences restore doesn't work with ESP-IDF

vooon, 2021-11-06:
### The problem

Since i switched my devices to ESP-IDF i notice trouble with state restore options.

For example, here all globals always have initial_value after reboot:
https://github.com/vooon/smart-door-lock/blob/c5343cf334c4ce40818b402702fb3b2d4b19fcf2/entrance-lock.yaml#L113-L137

Same applies to light brightness and switch states, so i think trouble somewhere in preferences module.
It doesn't looks like there are any troublesome leftovers from arduino, as i cleared flash entirely on part of devices.

### Which version of ESPHome has the issue?

2021.11.0-dev, 2021.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.10.7

### What platform are you using?

ESP32-IDF

### Board

doesn't really matter, all modules has same issue

### Component causing the issue

preferences

### Anything in the logs that might be useful for us?

```txt
Didn't noticed anything useful. Maybe i need to pass some flag to enable debugging for prefs module?
```

mmakaay, 2021-11-06:
I think I see the same thing with my Xiaomi lamp project.
After flashing, the light always turns on in the standard white light mode.
Haven't looked into that further, but the behavior sounds familiar.

vooon, 2021-11-06:
@mmakaay yes, it's very likely. Recently i had to modify MJXDD01YL (after preferences restore I'm unable to use it in HA).
And having restore trouble too.

https://gist.github.com/vooon/47469aad914449f9b24f02b5a586f735
[image]

jakosek, 2021-11-11:
on Arduino builds I have same problems, globals, light state - nothing restore on ESP32.

mmakaay, 2021-11-26:
I have added some extra logging to explicitly log when saving works. First I change the lamp color, and I log part of the data that are saved:
```
[16:04:46][V][light:290]: Stored RGB = 0.000000 1.000000 0.000000
```

There are many other parameters in the stored data, but this is enough to see if I'm storing correct data. In this case, I can confirm that the RGB green value matches the color that I switched the lamp to.
After changing the color and waiting for a little while, I see that the settings flush operation takes place:
```
[16:05:06][D][esp32.preferences:119]: Saving preferences to flash @ 1315220032, size=44 ...
[16:05:06][V][esp32.preferences:138]: Saving preferences successful

```
So I don't run into a not enough space error here. 

**Note: Waiting for this "Saving preferences" log message is important.**
One thing that has changed fairly recently, is that the settings are not stored to flash on every single change. Instead, periodically a check is done to see if the data to store are "dirty". When they, are, the data are stored. This prevents the system from writing too often to the flash, possibly causing more wear than required.
Because of this, when unplugging the lamp right after changing the color, the settings are expected to not be saved.

When I unplug my lamp after the save and plug it in again, I get the following when the settings are read:
```
[16:04:46][V][esp32.preferences:057]: nvs_get_blob('1315220032'): ESP_ERR_NVS_NOT_FOUND - the key might not be set yet
[16:04:46][V][light:273]: Restore from saved failed
```

The key "1315220032" matches in the save and restore log messages, so the problem is not in using the wrong key.

mmakaay, 2021-11-26:
Tracking further what happens, I found that the `nvs_open(...)` call failed. And when that happens, the `ESP32Preferences::open(...)` code erases NVS. The error code that I get, is `ESP_ERR_NVS_NOT_INITIALIZED`.

mmakaay, 2021-11-26:
I think I fixed it. The `nvs_flash_init()` function was not called before doing `nvs_open(...)`. That causes that error that I saw.
I created a PR for the fix (single line change): https://github.com/esphome/esphome/pull/2805

If others can test this change, that would be nice.

closed 2021-11-26 by oxan