esphome/issues #2683: Status LED stopping RGB Light from working on connected pin

BertrumUK, 2021-11-06:
### The problem

Since updating to ESPHome 2021.10.3 last night I noticed that whatever pin the status_led is connected to does not work as intended. The LED would light up for a second then go off. If I remove status_led from the configuration then the LED colour stays lit.

I have replicated this issue on a breadboard using the code below and a 4 pin LED using another Wemos D1 Mini. I tried rolling back to ESPHome 2021.10.2 but the issue persists. An older version of ESPHome 2021.8.2 would not start on my Pi4 so could not be tested..

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.1

### What platform are you using?

ESP8266

### Board

Wemos D1 Mini

### Component causing the issue

status_led

### Example YAML snippet

```yaml
esphome:
  name: test-led
  platform: ESP8266
  board: d1_mini

wifi:
  ssid: 'RogueOne'
  password: !secret wifipw
  manual_ip:
    static_ip: 192.168.0.100
    gateway: 192.168.0.1
    subnet: 255.255.255.0

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test LED"
    password: !secret wifipw

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:    

      
output:
  - platform: esp8266_pwm
    pin: D6
    id: red
  - platform: esp8266_pwm
    pin: D7
    id: green
  - platform: esp8266_pwm
    pin: D8
    id: blue
    
light:
  - platform: rgb
    name: "Test LED"
    red: red
    green: green
    blue: blue
    
status_led:
  pin: D6
```

oxan, 2021-11-10:
What exactly are you seeing? The status_led on D6 is supposed to remain off as long as there are no errors.

BertrumUK, 2021-11-11:
As the wemos D1 mini boots the status led flashes as expected and turns off once it is up and connected. The led has 4 pins, a Gnd and a red,green&blue pin. Whatever colour pin that is sharing the same led status pin does not light up and stay lit if that colour is selected. For example if the red led shares the same gpio as the status led it will light up red for about a second then go off. It did not do this before I updated to the latest versions. 

My code above is a basic set up for any esp8266 and should exhibit the same behaviour as I have tried it on a breadboard with a spare wemos and led to prove to myself that it was an issue.

BertrumUK, 2021-11-11:
Here is a short video showing the red led status light flashing on boot but if I try and select red it lights then goes off. Blue and green work as expected. If I move the status led confit to blue or green then they have this issue. 

https://share.icloud.com/photos/04Y8TviC6KHaEzUOyn3bdHCgw

oxan, 2021-11-14:
Ah, you're sharing a pin between components, that's not supported. I don't think it's likely for that to change anytime soon.

There's the [status_led](https://esphome.io/components/light/status_led.html) light component that allows sharing a light with a status LED, but it only works for monochromatic lights.

BertrumUK, 2021-11-14:
Is this a new thing? It used to work just fine for me until the recent updates.

oxan, 2021-11-14:
It's not a new policy, probably worked purely by accident before. I don't immediately see which change would've caused this, though.

BertrumUK, 2021-11-14:
It would be a shame to lose this functionality as you can use 1 LED and 1 LED to provide at least 2 functions. As I said it worked perfectly before so I use it on all my LED devices where a status led is used.

BertrumUK, 2021-11-14:
I tried monochromatic as well and that has the same issue 

`output:
  - platform: esp8266_pwm
    pin: D6
    id: red
  - platform: esp8266_pwm
    pin: D7
    id: green
  - platform: esp8266_pwm
    pin: D8
    id: blue
    
light:
  - platform: monochromatic
    name: "Red"
    output: red
  - platform: monochromatic
    name: "Green"
    output: green
  - platform: monochromatic
    name: "Blue"
    output: blue`

oxan, 2021-11-15:
I honestly don't see how this would ever have worked, the `status_led` component has been unconditionally turning off the pin it's connected to when there's no error ever since v1.13.

BertrumUK, 2021-11-15:
It have some sonoff basics that are still running 1.20.3 so that is probably the version where it was still working. Can I force Esphome to rollback and install that version on my test led setup as a proof of concept?

BertrumUK, 2021-11-16:
I went back through previous versions and found [this](https://github.com/esphome/esphome/pull/1974)

By changing the code use the [LED Status Light](https://esphome.io/components/light/status_led.html?highlight=status) component it now works 

`light:
  - platform: rgb
    name: "Test LED"
    red: red
    green: green
    blue: blue
  - platform: status_led
    name: "Switch state"
    pin: D6`

BertrumUK, 2021-11-16:
@oxan You mentioned the status_led light component above - sorry I missed that it was a different component from the normal status_led component.

I'm all good now thanks.

closed 2021-11-16 by BertrumUK