esphome/issues #2685: ESP8266 - Restoring global variable of type std::vector<std::string> from flash causing bootloop

pkejval, 2021-11-06:
### The problem

Hello, I just found bug when restoring saved variable of type std::vector<std::string>

When variable is set to restore: false no bootloop is present.
Program is crashing at last line in lambda:
`id(tag_list) = result;`

I tried clearing existing variable and assigning new values into:
```
            std::stringstream data(x);
            std::string line;
            
            id(tag_list).clear();
            while(std::getline(data,line,',')) {
              ESP_LOGD("main", "Inserting card %s", line.c_str());
              id(tag_list).push_back(line);
            }
```

Attached minimal YAML for easy reproduction.

Test data for sensor:
`97-C3-C9-1C,97-C3-C9-1A,97-C3-C9-1B,97-C3-C9-1Y,97-C3-C9-1X,97-C3-C9-1E`

### Which version of ESPHome has the issue?

v2021.10.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP8266

### Board

nodemcu

### Example YAML snippet

```yaml
esphome:
  platform: ESP8266
  board: nodemcuv2
  esp8266_restore_from_flash: true
  includes:
    - sstream.h

globals:  
  - id: tag_list
  type: std::vector<std::string>
  restore_value: true

text_sensor:
  - platform: homeassistant
    id: allowed_tags
    entity_id: sensor.allowed_tags
    on_value:
      then:
        - lambda: |-
            std::vector<std::string> result;
            std::stringstream data(x);
            std::string line;
            
            while(std::getline(data,line,',')) {
              ESP_LOGD("main", "Inserting card %s", line.c_str());
              result.push_back(line);
            }

            id(tag_list) = result;
```

oxan, 2021-11-10:
Yes, this ain't gonna work, because `std::string` holds a pointer to heap memory where the actual string is, and that doesn't get restored. I don't think that's something that will be changed anytime soon...

pkejval, 2021-11-10:
@oxan Thanks for looking at it.
My no-so-good workaround is hardcoding initial values into YAML like so:

```
substitutions:
  allowed_tags: '{"ABC", "CBA"}'

globals:  
  - id: tag_list
  type: std::vector<std::string>
  restore_value: false
  initial_value: $allowed_tags
```
This way my code works as intended but initial values are not updated from HA... Should not be a big deal because HA is almost always available to update tags from HA (DB).

paul-at-pebbletech, 2022-02-25:
@oxan Is this the same for a single string eg `std::string` as well as `std::vector<std::string>`. If so is there a way to reserve say 400 bytes of flash that can be preserved during power interruptions, by using the `EEPROM.h` Arduino library?

pkejval, 2022-02-25:
My propose for this is to create compile-time check for these unsupported types and disallow to use `restore_value: true` for them. Current behaviour is just crashing ESP at runtime which is very bad.My propose for this is to create compile-time check for these unsupported types and disallow to use `restore_value: true` for them. Current behaviour is just crashing ESP at runtime which is very bad.

paul-at-pebbletech, 2022-02-25:
Yes I spent a couple days thinking the problem was the HA python service call. As when I assign the string from HA to the global string the ESP device reboots. Oddly if I assign the service data to template sensor and then use button press automation I can assign the template sensor to the global. But the global is preserved on reboot

OttoWinter, 2022-02-25:
> My propose for this is to create compile-time check for these unsupported types and disallow to use `restore_value: true` for them. Current behaviour is just crashing ESP at runtime which is very bad.

Exactly that already exists. Just the esp8266 compiler is so old that the check is not supported by the compiler (https://github.com/esphome/esphome/blob/bf60e40d0bda314bccf1dd7c870ef9af6af87e56/esphome/core/helpers.h#L61)

paul-at-pebbletech, 2022-02-25:
@OttoWinter could I seek your opinion if I may, on how to store large compex data as per my first post on this issue eg 200 plus bytes. As i am so close to having a scheduler component that gets it schedule from HA but maintains it over a power fail / HA not available which should be the holy grail of automation eg intelligence at the edge closest to the sensors/ actuators 
Thank you paul

Gohper, 2024-05-05:
After two weeks of scratching searching I found this post.
It seems that it still does not work, or?

pkejval, 2024-05-05:
> [Sledovací obrázek][https://github.com/notifications/beacon/ALOKP43ZQQTZI4IX7PH4GFTZAX7SJA5CNFSM5HP5HP5KYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOPTNPJMA.gif]

I didn't tried it since I created this issue but I'm assuming it won't ever work.

Maybe YAML parser should stop user doing this because it hardlocks ESP in bootloop.