esphome/issues #2687: Compile error after update to 2021.10.3

theflorianmaas, 2021-11-07:
### The problem

I get this errors when I try to update existing nodes, This issue did not occour with the previous version. Any idea where and what to check?
Thank you

> INFO Reading configuration /config/esphome/Cam_terrazza.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing camext (board: esp32dev; framework: arduino; platform: espressif32 @ 3.3.2)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <WiFi> 1.0
|-- <Update> 1.0
|-- <ESPmDNS> 1.0
Compiling /data/camext/.pioenvs/camext/src/esphome/components/api/api_connection.cpp.o
sh: 1: xtensa-esp32-elf-g++: Permission denied
Compiling /data/camext/.pioenvs/camext/src/esphome/components/api/api_frame_helper.cpp.o
sh: 1: xtensa-esp32-elf-g++: Permission denied
Compiling /data/camext/.pioenvs/camext/src/esphome/components/api/api_pb2.cpp.o
sh: 1: xtensa-esp32-elf-g++: Permission denied
*** [/data/camext/.pioenvs/camext/src/esphome/components/api/api_connection.cpp.o] Error 126
*** [/data/camext/.pioenvs/camext/src/esphome/components/api/api_frame_helper.cpp.o] Error 126
Compiling /data/camext/.pioenvs/camext/src/esphome/components/api/api_pb2_service.cpp.o
*** [/data/camext/.pioenvs/camext/src/esphome/components/api/api_pb2.cpp.o] Error 126
sh: 1: xtensa-esp32-elf-g++: Permission denied
*** [/data/camext/.pioenvs/camext/src/esphome/components/api/api_pb2_service.cpp.o] Error 126
========================== [FAILED] Took 5.23 seconds ==========================

[image]

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.1

### What platform are you using?

ESP32

### Board

ESP32-cam, ESP32 mini D1,  TTGO-CAM

### Component causing the issue

api_connection.cpp.o, api_pb2.cpp.o, api_pb2_service.cpp.o

### Example YAML snippet

```yaml
# ESP32-CAM

substitutions:
  node_name: "camext"
  device_verbose_name: "cam terrazza"
  static_ip: 192.168.178.105
  resolution: "1600x1200"
  #resolution: "800x600"
  jpeg_quality: "10"
  vertical_flip: "False"
  horizontal_mirror: "False"
  contrast: "0" 
  brightness: "2"
  saturation: "-1"
  password: "1234"  
  ota_password: "1234"

packages: 
  wifi: !include common/wifi.yaml
  device_base: !include common/device_esp32cam.yaml

sensor:
  - platform: wifi_signal
    name: "WiFi Signal Sensor Cam Terrazza"
    update_interval: 60s
```

mmakaay, 2021-11-07:
It looks like your toolchain isn't installed correctly. The sh shell is complaining that the compiler program is not readable or executable, permission-wise.

How are you running ESPHome?

theflorianmaas, 2021-11-07:
ESPHome is running as Home Assistant addon.
I have removed and installed the addon.
The toolchain is installed again with no issues

[image]

but then I get the same errors

[image]

I wonder if after the addon is removed then there is something to delete manually before installing the addon again.

mmakaay, 2021-11-07:
I don't have ESPHome running as a Home Assistant addon. Using the standard dashboard container, I can't reproduce the issue. 
Can you please share the device YAML configuration that is failing for you, so I can be sure that I'm testing the same thing here?

mmakaay, 2021-11-07:
Another issue that might be related is https://github.com/esphome/issues/issues/2607
In that issue, the system reported that toolchain-xtense 2.40802 was not available for the Raspberry Pi architecture.
Are you running on Rasperry Pi? If yes, then possibly a package was created by now, but in a broken state.

theflorianmaas, 2021-11-07:
No, it is running on x64 desktop.
Thanks

mmakaay, 2021-11-07:
The other question then: "Can you please share the device YAML configuration that is failing for you?"

theflorianmaas, 2021-11-07:
All YAML configurations fail. 
See the attached file, containing a YAML returning the error

Thank you

theflorianmaas, 2021-11-07:
> All YAML configurations fail. See the attached file, containing a YAML returning the error
> 
> Thank you

Here is the file....

[ESPHome_example.txt](https://github.com/esphome/issues/files/7493020/ESPHome_example.txt)

mmakaay, 2021-11-07:
I tried compiling the exact example, and for me compilation worked.
But my environment is not the same as yours. Let's hope somebody else can replicate the issue that you're seeing here.

jsuanet, 2021-11-08:
I did a test with the HA add on version ESPhome 2021.10.3 and the yaml in your ESPHome_example.txt compiled without errors (had to extend the wifi password to  8 chars first), see attachted file. You could try to clean your build directories first. If that is not solves the problem remove and reinstall the ESPHome add on again. 

[corrente compile results.txt](https://github.com/esphome/issues/files/7495803/corrente.compile.results.txt)

theflorianmaas, 2021-11-08:
I tried with a fresh intallation of HA and ESPHome and all YAML have been compiled correctly.
Thank you all for your support!

mmakaay, 2021-11-09:
Ok, curiouis why and how this happen, but good to know that it's not a general issue and that you got it to work 👍

closed 2021-11-09 by mmakaay