esphome/issues #2689: FastLED compilation error

bulburDE, 2021-11-08:
### The problem

After upgrading esphome I get an error when compiling the code. I am implementing an [Custom Light Output](https://esphome.io/components/light/custom.html) where I import `FastLED.h` in my custom code.
```
src/main.cpp:26:34: error: invalid use of '::'
 #define delay(x) esphome::delay(x)
                                  ^
.piolibdeps/wordclock/FastLED/src/FastLED.h:522:7: note: in expansion of macro 'delay'
  void delay(unsigned long ms);
       ^
```

This may be related to this pull request: https://github.com/esphome/esphome/pull/2575.

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

Wemos D1 mini

### Anything in the logs that might be useful for us?

```txt
WARNING Calling ESPHome with the configuration before the command is deprecated and will be removed in the future. 
WARNING Please instead use:
WARNING    esphome run wordclock.yaml
INFO Reading configuration wordclock.yaml...
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
INFO Compiling app...
Processing wordclock (board: d1_mini; framework: arduino; platform: platformio/espressif8266 @ 2.6.3)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- <FastLED> 3.4.0
|   |-- <SPI> 1.0
|   |-- <EspSoftwareSerial> 6.8.5
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
|   |-- <ESP8266WiFi> 1.0
Compiling .pioenvs/wordclock/src/main.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/IPAddress.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/MD5Builder.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/Print.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/Schedule.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/StackThunk.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/Stream.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/StreamString.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/Tone.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/TypeConversion.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/Updater.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/WMath.cpp.o
In file included from src/wordclock_light.h:2:0,
                 from src/main.cpp:28:
.piolibdeps/wordclock/FastLED/src/FastLED.h:14:21: note: #pragma message: FastLED version 3.004.000
 #    pragma message "FastLED version 3.004.000"
                     ^
Compiling .pioenvs/wordclock/FrameworkArduino/WString.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/abi.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/base64.cpp.o
In file included from .piolibdeps/wordclock/FastLED/src/FastLED.h:65:0,
                 from src/wordclock_light.h:2,
                 from src/main.cpp:28:
.piolibdeps/wordclock/FastLED/src/fastspi.h:135:23: note: #pragma message: No hardware SPI pins defined.  All SPI access will default to bitbanged output
 #      pragma message "No hardware SPI pins defined.  All SPI access will default to bitbanged output"
                       ^
Compiling .pioenvs/wordclock/FrameworkArduino/cbuf.cpp.o
src/main.cpp:26:34: error: invalid use of '::'
 #define delay(x) esphome::delay(x)
                                  ^
.piolibdeps/wordclock/FastLED/src/FastLED.h:522:7: note: in expansion of macro 'delay'
  void delay(unsigned long ms);
       ^
Compiling .pioenvs/wordclock/FrameworkArduino/cont.S.o
Compiling .pioenvs/wordclock/FrameworkArduino/cont_util.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/core_esp8266_app_entry_noextra4k.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/core_esp8266_eboot_command.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/core_esp8266_features.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/core_esp8266_flash_quirks.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/core_esp8266_flash_utils.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/core_esp8266_i2s.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/core_esp8266_main.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/core_esp8266_noniso.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/core_esp8266_phy.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/core_esp8266_postmortem.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/core_esp8266_si2c.cpp.o
Compiling .pioenvs/wordclock/FrameworkArduino/core_esp8266_sigma_delta.cpp.o
*** [.pioenvs/wordclock/src/main.cpp.o] Error 1
```

bulburDE, 2021-11-08:
In previous versions (here 2021.10.0) there were even more errors regarding the functions `delayMicroseconds(int)`, `millis()`, `micros()`
```
In file included from .piolibdeps/wordclock/FastLED/src/FastLED.h:44:0,
                 from src/wordclock_light.h:2,
                 from src/main.cpp:23:
.piolibdeps/wordclock/FastLED/src/fastled_delay.h: In member function 'void CMinWait<WAIT>::wait()':
.piolibdeps/wordclock/FastLED/src/fastled_delay.h:22:19: error: call of overloaded 'micros()' is ambiguous
    diff = (micros() & 0xFFFF) - mLastMicros;
                   ^
```

After downgrading to 2021.9.3 it is working again.

mmakaay, 2021-11-08:
Not trying to solve the issue, but doing a quick jot down of what I think that's going on here:

The FastLED library contains a declaration
```
void delay(unsigned long ms);
```
but `esphome/home/core.py` adds something called `ARDUINO_GLUE_CODE`, which amongs other things defines `delay(x)` as `esphome::delay(x)`. That basically changes the declaration from above to:
```
void esphome::delay(unsigned long ms);
```
and this is invalid.

mmakaay, 2021-11-08:
Long shot: maybe using an `#undef delay(x)` before doing the FastLED.h include helps as a work-around?

bulburDE, 2021-11-08:
Thanks for the quick response and the working work-around :)
Using the following code in my custom code is working:
```
#include "esphome.h"
#undef delay
#include "FastLED.h"
```

mmakaay, 2021-11-08:
Ah, great! Of course, you can't use delay() after this, but esphome::delay() would work as intended.

oxan, 2021-11-10:
Including third-party libraries in custom components is usually a bad idea due to how we pollute the global namespace... I'd like to find a solution for that someday, but easier said than done.

Syralist, 2022-01-26:
Since a few versions the above mentioned workaround no longer works. Now the errormessage says:
`In file included from /data/lixieclock/.piolibdeps/lixieclock/Lixie/src/Lixie.cpp:7:0:
/data/lixieclock/.piolibdeps/lixieclock/Lixie/src/Lixie.h:18:21: fatal error: FastLED.h: No such file or directory`

ssieb, 2022-12-19:
This is not an esphome issue.

closed 2022-12-19 by ssieb

nielsnl68, 2022-12-19:
duplicate of #3926, #3906