esphome/issues #2690: esp32_ble_tracker breaks mDNS and wifi stability

Goz3rr, 2021-11-08:
### The problem

After enabling `esp32_ble_tracker` in the config, the device will no longer be reachable via the mDNS hostname (breaking the log and OTA). If I manually set a DNS entry for the correct hostname and IP, both will start working again, however the log will lose connection exactly every 3 minutes and then reconnect after a few seconds. In addition to this, the device itself logs losing connection to Home Assistant at seemingly random times:
`[W][api.connection:080]: Home Assistant 2021.11.1 (192.168.0.10): Connection reset`
Rarely I also see:
`[W][wifi_esp32:386]: Event: Disconnected ssid='xxx' bssid=xxx reason='Not Authenticated'`
This is quite annoying because after it reconnects, Home Assistant will record a new value even the actual value hasn't changed for hours/days.

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.11.1

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

esp32_ble_tracker

### Example YAML snippet

```yaml
esphome:
  name: esp32-test
  platform: ESP32
  board: esp32dev
  
wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: !secret ota_password

# Comment this out and the issue disappears
esp32_ble_tracker:
```

### Anything in the logs that might be useful for us?

```txt
Log from ESP32 when not working, but I don't see anything weird:

ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
[I][logger:214]: Log initialized
[C][ota:452]: There have been 0 suspected unsuccessful boot attempts.
[D][esp32.preferences:113]: Saving preferences to flash...
[I][app:029]: Running through setup()...
[D][esp32_ble_tracker:214]: Starting scan...
[C][wifi:037]: Setting up WiFi...
[D][wifi:372]: Starting scan...
[D][esp32_ble_tracker:719]: Found device xx:xx:xx:1E:A3:C2 RSSI=-92
[D][esp32_ble_tracker:740]:   Address Type: RANDOM
[D][esp32_ble_tracker:719]: Found device xx:xx:xx:7B:81:D5 RSSI=-70
[D][esp32_ble_tracker:740]:   Address Type: RANDOM
[D][esp32_ble_tracker:719]: Found device xx:xx:xx:0E:8A:0C RSSI=-65
[D][esp32_ble_tracker:740]:   Address Type: RANDOM
[D][wifi:387]: Found networks:
[I][wifi:431]: - '[my ssid]' (xx:xx:xx:8A:6B:7A) ▂▄▆█
[D][wifi:432]:     Channel: 6
[D][wifi:433]:     RSSI: -62 dB
[D][wifi:436]: - '[my ssid]-iot' (xx:xx:xx:8A:6B:7A) ▂▄▆█
[D][wifi:436]: - '' (xx:xx:xx:8A:6B:7A) ▂▄▆█
[D][wifi:436]: - '[my ssid]-iot' (xx:xx:xx:B7:C1:12) ▂▄▆█
[I][wifi:247]: WiFi Connecting to '[my ssid]'...
[D][esp32_ble_tracker:719]: Found device xx:xx:xx:0E:5E:F9 RSSI=-78
[D][esp32_ble_tracker:740]:   Address Type: RANDOM
[I][wifi:504]: WiFi Connected!
[C][wifi:352]:   Local MAC: 24:0A:C4:81:E5:08
[C][wifi:353]:   SSID: '[my ssid]'
[C][wifi:354]:   IP Address: 192.168.0.163
[C][wifi:356]:   BSSID: xx:xx:xx:8A:6B:7A
[C][wifi:357]:   Hostname: 'esp32-test'
[C][wifi:359]:   Signal strength: -62 dB ▂▄▆█
[C][wifi:363]:   Channel: 6
[C][wifi:364]:   Subnet: 255.255.255.0
[C][wifi:365]:   Gateway: 192.168.0.1
[C][wifi:366]:   DNS1: 192.168.0.10
[C][wifi:367]:   DNS2: 1.1.1.1
[C][ota:082]: Over-The-Air Updates:
[C][ota:083]:   Address: esp32-test.local:3232
[C][ota:086]:   Using Password.
[C][api:025]: Setting up Home Assistant API server...
[I][app:060]: setup() finished successfully!
[I][app:099]: ESPHome version 2021.10.3 compiled on Nov  8 2021, 18:36:27
[C][wifi:490]: WiFi:
[C][wifi:352]:   Local MAC: 24:0A:C4:81:E5:08
[C][wifi:353]:   SSID: '[my ssid]'
[C][wifi:354]:   IP Address: 192.168.0.163
[C][wifi:356]:   BSSID: xx:xx:xx:8A:6B:7A
[C][wifi:357]:   Hostname: 'esp32-test'
[C][wifi:359]:   Signal strength: -62 dB ▂▄▆█
[C][wifi:363]:   Channel: 6
[C][wifi:364]:   Subnet: 255.255.255.0
[C][wifi:365]:   Gateway: 192.168.0.1
[C][wifi:366]:   DNS1: 192.168.0.10
[C][wifi:367]:   DNS2: 1.1.1.1
[C][logger:233]: Logger:
[C][logger:234]:   Level: DEBUG
[C][logger:235]:   Log Baud Rate: 115200
[C][logger:236]:   Hardware UART: UART0
[C][esp32_ble_tracker:705]: BLE Tracker:
[C][esp32_ble_tracker:706]:   Scan Duration: 300 s
[C][esp32_ble_tracker:707]:   Scan Interval: 320.0 ms
[C][esp32_ble_tracker:708]:   Scan Window: 30.0 ms
[C][esp32_ble_tracker:709]:   Scan Type: ACTIVE
[C][ota:082]: Over-The-Air Updates:
[C][ota:083]:   Address: esp32-test.local:3232
[C][ota:086]:   Using Password.
[C][api:134]: API Server:
[C][api:135]:   Address: esp32-test.local:6053
[C][api:139]:   Using noise encryption: NO
[D][api:098]: Accepted 192.168.0.10
[D][api.connection:736]: ESPHome Logs 2021.10.3 (192.168.0.10): Connected successfully
[I][app:099]: ESPHome version 2021.10.3 compiled on Nov  8 2021, 18:36:27
[C][wifi:490]: WiFi:
[C][wifi:352]:   Local MAC: 24:0A:C4:81:E5:08
[C][wifi:353]:   SSID: '[my ssid]'
[C][wifi:354]:   IP Address: 192.168.0.163
[C][wifi:356]:   BSSID: xx:xx:xx:8A:6B:7A
[C][wifi:357]:   Hostname: 'esp32-test'
[C][wifi:359]:   Signal strength: -62 dB ▂▄▆█
[C][wifi:363]:   Channel: 6
[C][wifi:364]:   Subnet: 255.255.255.0
[C][wifi:365]:   Gateway: 192.168.0.1
[C][wifi:366]:   DNS1: 192.168.0.10
[C][wifi:367]:   DNS2: 1.1.1.1
[C][logger:233]: Logger:
[C][logger:234]:   Level: DEBUG
[C][logger:235]:   Log Baud Rate: 115200
[C][logger:236]:   Hardware UART: UART0
[C][esp32_ble_tracker:705]: BLE Tracker:
[C][esp32_ble_tracker:706]:   Scan Duration: 300 s
[C][esp32_ble_tracker:707]:   Scan Interval: 320.0 ms
[C][esp32_ble_tracker:708]:   Scan Window: 30.0 ms
[C][esp32_ble_tracker:709]:   Scan Type: ACTIVE
[C][ota:082]: Over-The-Air Updates:
[C][ota:083]:   Address: esp32-test.local:3232
[C][ota:086]:   Using Password.
[C][api:134]: API Server:
[C][api:135]:   Address: esp32-test.local:6053
[C][api:139]:   Using noise encryption: NO
[D][esp32_ble_tracker:719]: Found device xx:xx:xx:4E:D2:94 RSSI=-98
[D][esp32_ble_tracker:740]:   Address Type: RANDOM
[D][esp32_ble_tracker:744]:   TX Power: 2
[D][esp32_ble_tracker:719]: Found device xx:xx:xx:62:17:F3 RSSI=-81
[D][esp32_ble_tracker:740]:   Address Type: RANDOM
[D][esp32_ble_tracker:719]: Found device xx:xx:xx:F2:15:E8 RSSI=-72
[D][esp32_ble_tracker:740]:   Address Type: RANDOM
[W][api.connection:082]: ESPHome Logs 2021.10.3 (192.168.0.10): Connection closed
[D][esp32_ble_tracker:719]: Found device xx:xx:xx:F0:20:4F RSSI=-58
[D][esp32_ble_tracker:740]:   Address Type: RANDOM
[I][ota:102]: Boot seems successful, resetting boot loop counter.
[D][esp32.preferences:113]: Saving preferences to flash...
[D][esp32_ble_tracker:214]: Starting scan...
[D][esp32_ble_tracker:719]: Found device xx:xx:xx:F2:15:E8 RSSI=-82
[D][esp32_ble_tracker:740]:   Address Type: RANDOM
[D][esp32_ble_tracker:719]: Found device xx:xx:xx:0E:5E:F9 RSSI=-76
[D][esp32_ble_tracker:740]:   Address Type: RANDOM
[D][esp32_ble_tracker:719]: Found device xx:xx:xx:F0:20:4F RSSI=-61
[D][esp32_ble_tracker:740]:   Address Type: RANDOM
[D][esp32_ble_tracker:719]: Found device xx:xx:xx:62:17:F3 RSSI=-92
[D][esp32_ble_tracker:740]:   Address Type: RANDOM
```
```

### Additional information

When `esp32_ble_tracker` is enabled, these are all the mDNS messages I see after resetting the device:
![image](https://user-images.githubusercontent.com/761433/140795074-36bc3c93-2bc1-4735-aa78-b54bc536c163.png)

When it is disabled, the device seems to send the same packets on reset but also sends periodic packets now (repeating forever every ~16 seconds):
![image](https://user-images.githubusercontent.com/761433/140795187-edb1ebd7-77d5-4e0f-bf3f-7a983566583e.png)

Combined with the wifi issues, could it be something timer related that is not being executed?

cheisig, 2021-12-26:
Hi!
Similiar problem here.

Running:
ESPHOME Stable: 2021.12.1
Device: Sonoff Dual R3 v1.6

As soon as I enable esp32_ble_tracker I get a some strange side effects:
1.) Disconnect from API randomly, device gets unavailable in HA
2.) When trying to upload a new fimware, it gets aborted between 10 and 80 Percent, with connection broken - timeout
3.) WebServer upload is luckily working

Now I am heavily using the current sensor for my current based covers.
Not sure if the board is just overloaded with BLE tracking enabled.

As soon as I disable the BLE stuff, the ESP32 is fine again.

My config:

`#ESP32 current based cover (DUAL R3 v1.6 Sonoff)
#'coverofficesmall' - Open position reached. Took 11.9s.
#'coverofficesmall' - Close position reached. Took 11.5s.
substitutions:
  device_name: coverofficesmall
  open_duration: 11.9sec
  close_duration: 11.5sec
  max_duration: 12.5sec

esphome:
  name: ${device_name}

esp32:
  board: esp32dev
  framework:
    type: arduino

wifi:
  reboot_timeout: 0s
  domain: ".iot.heisig.one"
  networks:
  - ssid: !secret wifi_ssid
  - password: !secret wifi_pass
  - ssid: 'rescue'
  - password: '12345678'

# Enable logging
logger:

  # Enable Web server
web_server:
  port: 80

# Enable Home Assistant API/OTA
api:
  password: !secret api_pass2
  reboot_timeout: 0s
ota:
  safe_mode: true
  password: !secret ota_pass

# Enable Bluetooth scanning for this ESP32
#esp32_ble_tracker:

# Sync time with Home Assistant
time:
  - platform: homeassistant
    id: homeassistant_time

# cse7761 uart port
uart:
[... 189 lines cut ...]
  - platform: current_based
    name: ${device_name}
    id: rollers
    max_duration: ${max_duration}
    open_sensor: open_current
    open_moving_current_threshold: 0.45
    open_duration: ${open_duration}
    open_action:
      - switch.turn_on: open_relay
    close_sensor: close_current
    close_moving_current_threshold: 0.45
    close_duration: ${close_duration}
    close_action:
      - switch.turn_on: close_relay
    stop_action:
      - switch.turn_off: close_relay
      - switch.turn_off: open_relay
    obstacle_rollback: 30%
    start_sensing_delay: 1s
    malfunction_detection: true
    malfunction_action:
      then:
        - logger.log: "Malfunction detected. Relay welded."

`

fenderle, 2022-01-10:
ESP32 shares for BLE and WiFi the same antenna, meaning that WiFi and BLE cannot use the antenna at the same time. This of course has impact on WiFi and/or BLE usage since the access to the antenna has to be coordinated. The BLE and WiFi controllers in the ESP32 firmware try their best to make this work seamless but in reality there are some issues with this approach.

Not that this helps with your issue, but maybe the devs take a look into that direction as a cause of problems with WiFi and BLE.

ayufan, 2022-01-23:
I'm seeing the same problems: with BLE enabled esp32 does crash pretty often if used with Web.

For me it was pretty easy and constant problem when:
- have `esp32_ble_tracker` enabled
- scrape `/metrics` prometheus metrics by discovering all devices over mDNS

pplucky, 2022-01-25:
> Hi! Similiar problem here.
> 
> Running: ESPHOME Stable: 2021.12.1 Device: Sonoff Dual R3 v1.6
> 
> As soon as I enable esp32_ble_tracker I get a some strange side effects: 1.) Disconnect from API randomly, device gets unavailable in HA 2.) When trying to upload a new fimware, it gets aborted between 10 and 80 Percent, with connection broken - timeout 3.) WebServer upload is luckily working
> 
> Now I am heavily using the current sensor for my current based covers. Not sure if the board is just overloaded with BLE tracking enabled.
> 
> As soon as I disable the BLE stuff, the ESP32 is fine again.
> 
> My config:
> 
> ```yaml
> #ESP32 current based cover (DUAL R3 v1.6 Sonoff)
> #'coverofficesmall' - Open position reached. Took 11.9s.
> #'coverofficesmall' - Close position reached. Took 11.5s.
> substitutions:
>   device_name: coverofficesmall
>   open_duration: 11.9sec
>   close_duration: 11.5sec
>   max_duration: 12.5sec
> 
> esphome:
>   name: ${device_name}
> 
> esp32:
>   board: esp32dev
>   framework:
>     type: arduino
> 
> wifi:
>   reboot_timeout: 0s
>   domain: ".iot.heisig.one"
>   networks:
>   - ssid: !secret wifi_ssid
>   - password: !secret wifi_pass
>   - ssid: 'rescue'
>   - password: '12345678'
> 
> # Enable logging
> logger:
> 
>   # Enable Web server
> web_server:
>   port: 80
> 
> # Enable Home Assistant API/OTA
> api:
>   password: !secret api_pass2
>   reboot_timeout: 0s
> ota:
>   safe_mode: true
>   password: !secret ota_pass
> 
> # Enable Bluetooth scanning for this ESP32
> #esp32_ble_tracker:
> 
> # Sync time with Home Assistant
> time:
>   - platform: homeassistant
>     id: homeassistant_time
> 
> # cse7761 uart port
> uart:
>   tx_pin: GPIO25
>   rx_pin: GPIO26
>   baud_rate: 38400
>   parity: EVEN
> 
> 
[... 185 lines cut ...]
>     name: ${device_name}
>     id: rollers
>     max_duration: ${max_duration}
>     open_sensor: open_current
>     open_moving_current_threshold: 0.45
>     open_duration: ${open_duration}
>     open_action:
>       - switch.turn_on: open_relay
>     close_sensor: close_current
>     close_moving_current_threshold: 0.45
>     close_duration: ${close_duration}
>     close_action:
>       - switch.turn_on: close_relay
>     stop_action:
>       - switch.turn_off: close_relay
>       - switch.turn_off: open_relay
>     obstacle_rollback: 30%
>     start_sensing_delay: 1s
>     malfunction_detection: true
>     malfunction_action:
>       then:
>         - logger.log: "Malfunction detected. Relay welded."
> ```

Similar issue here happening after update to version 2022.1.0 (happened the same with 2021.12.x). Unfortunately I don't remember anymore what was the working version I had before, otherwise I would just downgrade...

cheisig, 2022-01-25:
For me it is working fine, since I disabled the web server.
No random restart since 7 days now.

Still with Version ESPHOME Stable: 2021.12.1

ayufan, 2022-05-12:
I start to think that the problem here is extensive memory (and flash, less likely) usage of BT and BLE stack. There's a very little memory left that results in a lot of fragmentation and random failures in memory allocation.

I think that if we would maybe use https://github.com/h2zero/NimBLE-Arduino for BLE tracking as an alternative, this very likely would solve this issue.