esphome/issues #2691: SDS011 UART not working

scottkohn, 2021-11-08:
### The problem

very similar issues to this thread:
https://github.com/esphome/issues/issues/1089

Have tried multiple UART pin and code configs.  Both SDS sensor and ESP8266 work separately but can't get them to work together.   

### Which version of ESPHome has the issue?

2021.9.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

supervisor 2021.10.8

### What platform are you using?

ESP8266

### Board

ESP8266 NodeMCU CP2102 ESP-12E

### Component causing the issue

UART

### Example YAML snippet

```yaml
esphome:
  name: "2"
  platform: ESP8266
  board: nodemcuv2

# Enable Home Assistant API
api:

ota:
  password: "XXXXXXX"

wifi:
  ssid: "255IOT"
  password: !secret 255IOT_ssid_pswd
  fast_connect: true

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "1 Fallback Hotspot"
    password: "XXXXXXX"

#For SDS011 on RX, TX pins
uart:
  #- id: uart_1
    rx_pin: GPIO3
    tx_pin: GPIO1
    baud_rate: 9600

# Enable logging
logger:
  #level: VERBOSE
  baud_rate: 0 
  
sensor:
  - platform: dht
    model: DHT22_type2
    pin: D2
    temperature:
      name: "Temperature_2"
    humidity:
      name: "Humidity_2"
    update_interval: 60s
  - platform: wifi_signal
    name: "WiFi Signal Sensor_2"
    update_interval: 60s
  #- platform: dallas
  #  address: 0x9B000004B8119928
  #  name: "Temperature #2 - Dallas"
  #- platform: dallas
  #  address: 0x99000004B829C028
  #  name: "Temperature #2.1 - Dallas"
  - platform: sds011
    rx_only: false
    update_interval: 0min
    pm_2_5:
      name: "Particulate Matter <2.5µm Concentration"
    pm_10_0:
      name: "Particulate Matter <10.0µm Concentration"
    
    #uart_id: uart_1
```

scottkohn, 2021-11-09:
KL

scottkohn, 2021-11-09:
YAML:

esphome:
  name: "2"
  platform: ESP8266
  board: nodemcuv2
  
# # Enable logging
logger:
  level: VERBOSE

# Enable Home Assistant API
api:

ota:
  password: "xxxxxxxxxxx"

wifi:
  ssid: "255IOT"
  password: !secret 255IOT_ssid_pswd
  fast_connect: true

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "1 Fallback Hotspot"
    password: "xxxxxxxxxxxxxxxxxx"

uart:
  rx_pin: D1
  tx_pin: D2
  baud_rate: 9600

sensor:
  - platform: sds011
    pm_2_5:
      name: "Particulate Matter <2.5µm Concentration"
    pm_10_0:
      name: "Particulate Matter <10.0µm Concentration"
    update_interval: 0min

LOG:

[19:48:45][I][wifi:515]: WiFi Connected!
[19:48:45][C][wifi:361]:   SSID: [redacted]
[19:48:45][C][wifi:362]:   IP Address: 192.168.2.178
[19:48:45][C][wifi:364]:   BSSID: [redacted]
[19:48:45][C][wifi:365]:   Hostname: '2'
[19:48:45][C][wifi:369]:   Signal strength: -55 dB ▂▄▆█
[19:48:45][C][wifi:373]:   Channel: 1
[19:48:45][C][wifi:374]:   Subnet: 255.255.255.0
[19:48:45][C][wifi:375]:   Gateway: 192.168.2.1
[19:48:45][C][wifi:376]:   DNS1: 192.168.2.1
[19:48:45][C][wifi:377]:   DNS2: (IP unset)
[19:48:45][D][wifi:524]: Disabling AP...
[19:48:45][C][ota:029]: Over-The-Air Updates:
[19:48:45][C][ota:030]:   Address: 2.local:8266
[19:48:45][C][ota:032]:   Using Password.
[19:48:45][C][api:023]: Setting up Home Assistant API server...
[19:48:45][I][app:060]: setup() finished successfully!
[19:48:45][I][app:102]: ESPHome version 2021.9.3 compiled on Nov  8 2021, 19:47:19
[19:48:45][C][wifi:501]: WiFi:
[19:48:45][C][wifi:361]:   SSID: [redacted]
[19:48:45][C][wifi:362]:   IP Address: 192.168.2.178
[19:48:45][C][wifi:364]:   BSSID: [redacted]
[19:48:45][C][wifi:365]:   Hostname: '2'
[19:48:45][C][wifi:369]:   Signal strength: -57 dB ▂▄▆█
[19:48:45][C][wifi:373]:   Channel: 1
[19:48:45][C][wifi:374]:   Subnet: 255.255.255.0
[19:48:45][C][wifi:375]:   Gateway: 192.168.2.1
[19:48:45][C][wifi:376]:   DNS1: 192.168.2.1
[19:48:45][C][wifi:377]:   DNS2: (IP unset)
[19:48:45][C][uart_esp8266:097]: UART Bus:
[19:48:45][C][uart_esp8266:099]:   TX Pin: GPIO4
[19:48:45][C][uart_esp8266:102]:   RX Pin: GPIO5
[19:48:45][C][uart_esp8266:103]:   RX Buffer Size: 256
[19:48:45][C][uart_esp8266:105]:   Baud Rate: 9600 baud
[19:48:45][C][uart_esp8266:106]:   Data Bits: 8
[19:48:45][C][uart_esp8266:107]:   Parity: NONE
[19:48:45][C][uart_esp8266:108]:   Stop bits: 1
[19:48:45][C][uart_esp8266:112]:   Using software serial
[19:48:45][C][logger:193]: Logger:
[19:48:45][C][logger:194]:   Level: VERBOSE
[19:48:45][C][logger:195]:   Log Baud Rate: 115200
[19:48:45][C][logger:196]:   Hardware UART: UART0
[19:48:45][C][sds011:069]: SDS011:
[19:48:45][C][sds011:070]:   Update Interval: 0 min
[19:48:45][C][sds011:071]:   RX-only mode: OFF
[19:48:45][C][sds011:072]:   PM2.5 'Particulate Matter <2.5µm Concentration'
[19:48:45][C][sds011:072]:     Device Class: 'pm25'
[19:48:45][C][sds011:072]:     State Class: 'measurement'
[19:48:45][C][sds011:072]:     Unit of Measurement: 'µg/m³'
[19:48:45][C][sds011:072]:     Accuracy Decimals: 1
[19:48:45][C][sds011:072]:     Icon: 'mdi:chemical-weapon'
[19:48:45][C][sds011:073]:   PM10.0 'Particulate Matter <10.0µm Concentration'
[19:48:46][C][sds011:073]:     Device Class: 'pm10'
[19:48:46][C][sds011:073]:     State Class: 'measurement'
[19:48:46][C][sds011:073]:     Unit of Measurement: 'µg/m³'
[19:48:46][C][sds011:073]:     Accuracy Decimals: 1
[19:48:46][C][sds011:073]:     Icon: 'mdi:chemical-weapon'
[19:48:46][C][ota:029]: Over-The-Air Updates:
[19:48:46][C][ota:030]:   Address: 2.local:8266
[19:48:46][C][ota:032]:   Using Password.
[19:48:46][C][api:135]: API Server:
[19:48:46][C][api:136]:   Address: 2.local:6053
[19:48:46][C][api:140]:   Using noise encryption: NO
[19:48:55]pm open,type:0 0
[19:49:00][D][api:095]: Accepted 192.168.2.121
[19:49:00][V][api.connection:711]: Hello from client: 'Home Assistant 2021.10.5 (192.168.2.121)'
[19:49:00][D][api.connection:727]: Home Assistant 2021.10.5 (192.168.2.121): Connected successfully

oxan, 2021-11-10:
* Please put your yaml/logs in code blocks next time.
* What exactly is the behaviour you see? Are there any log messages if you let it running for a while longer?

scottkohn, 2021-11-14:
Thanks for the feedback. 

I can edit the original issue post if you'd like with the YAML and LOG. 

I saw nothing from the log file after a few minutes.  

The SDS unit fan runs continuously and the red LED on the backside blinks.

scottkohn, 2021-11-17:
Left it running for 8 minutes after installing code and see the same as before but with one additional message:
```
[17:55:08][V][api.connection:720]: Hello from client: 'Home Assistant 2021.11.1 (192.168.2.121)'
[17:55:08][D][api.connection:736]: Home Assistant 2021.11.1 (192.168.2.121): Connected successfully
[17:55:15]pm open,type:0 0
[17:59:47][I][ota:102]: Boot seems successful, resetting boot loop counter.
```

Red LED blinks every ~1 second on SDS unit (V 2.18) - same as before

scottkohn, 2021-11-17:
i think i sorted it out.  needed to flip flop the RX and TX lines between the SDS and ESP8266

scottkohn, 2021-11-17:
The TX and RX pins in some guides online have various hardware mapping between ESPXXXX and SDS011.  What worked is the TX on SDS connected to the pin defined as RX in the ESPXXXX and RX on SDS to pin defined as TX in the ESPXXXX.

closed 2021-11-17 by scottkohn